f4exb/sdrangel #286: Help! I need a simple instructional

DRTAC7, 2019-02-05:
Can someone reply with a simple step-by-step instructional on how to install SDRAngel on Ubuntu 16.04 LTS? I have tried to cmake install and evrything but I cannot figure out how to do this correctly.

DRTAC7, 2019-02-05:
I just need to know what commands to enter in what order.

f4exb, 2019-02-06:
This would be better answered at:
https://groups.io/g/sdrangel

If you are into cmake it means you will want to build from source:
https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

Then to start up:
https://github.com/f4exb/sdrangel/wiki/Quick-start

f4exb, 2019-02-07:
Posted in https://groups.io/g/sdrangel/topic/i_need_simple_instructions/29685375?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,29685375

closed 2019-02-07 by f4exb