f4exb/sdrangel #291: presets/audio preference dont save/load correctly

elitedata, 2019-02-15:
saving a sink preset does not save/load correctly.
preference > audio > selected sound card input/output does not save on next program execution.
this is on win10 pro x64
sdrangel has full admin permissions for hardware and OS/file/registry access.
device being used as sink is hack-rf.

f4exb, 2019-02-16:
Please have a look at this: https://github.com/f4exb/sdrangel/blob/master/sdrgui/audio.md to understand how audio is managed.

elitedata, 2019-02-16:
i appreciate the quick response but i think perhaps you misinterpreted the problems i am describing.
perhaps it may be easier to make a video with narration to describe and show the current problems rather than to try and explain them here.
i have read how the audio is managed but despite choosing the selected audio device, the program always reverts back to "system default" when the program is executed again.
perhaps this is an issue saving/loading a registry key (if the program is using the registry to save the settings) - i have not investigated this as of yet.
the issues with saving/loading a preset is also tedious and difficult to explain.
it may be easier reproduced on another PC with 64 bit version of windows 10, with full read/write and hardware access permissions applied to the SDR-angel program.
by all means, i would like to provide input on any issues/problems encountered, this program is an absolute wonder for the purposes i use it for and i give you huge thanks for creating it.
also, i am unable to locate a 32 bit released version of this software needed to operate on windows XP since i have additional software that will ONLY operate on windows XP 32 bit, not windows 7 32 bit or windows 10 32 bit.
i do realize there was a released version as i read here, possibly v3.6.0 or v3.9.0 but im unsure where to get it - thank you "f4exb".

f4exb, 2019-02-17:
Delete

f4exb, 2019-02-17:
I am still not sure you understand how things work. It might seem counter intuitive but this ensures maximum flexibility by decoupling the audio settings from the selection of the audio device

I suppose we are talking about this dialog:
[image]
This is NOT used to select which device will be used for audio output. Instead the audio device used can be selected individually in the channel GUI usually by right clicking on the "mute" button. By default it will go to "System default device". The audio device SELECTION dialog looks like this:
[image]

This selection should be maintained in the presets. There is an issue if it is not but I have not encountered this case on Linux though where the preferences are file system based. Of course the device should be available in your system at that time. If you use virtual devices make sure it is properly created before.

The audio PREFERENCES dialog let you specify some extra parameters like how to send the audio over the network. Note that the screenshot is from a development version and may have extra features vs yours. These preferences are not linked to a particular preset and are saved in the global section of the preferences.

elitedata, 2019-02-22:
yes.
i understand what you are saying and i have a very good understanding of how this wonderful software works but the problem is this:
shouldnt the application "remember" the last selected device under "audio preferences" the next time you open the application ?
in my case, i can select the device i want to use for input/output, thats fine but when i close the application and open it again, it reverts back to "system default" which is whatever device i set as the "default" within the configuration of the sound settings in windows 10.
i would rather not prefer to change the windows 10 sound input/output device default to "get around" this minor issue but its a bit annoying to have to chose the input/output device on the application every time i execute it.
here is an example screen shot of what is happening after i have chosen "HD Capture" as the selected output device, then i close the application and then i open it again, it selects "system default" instead of "HD Capture".
FYI, it does not matter what device i select, the application will always revert back to "system default" once it is closed and opened again.
i hope this helps clarify the issue.
http://scpdny.com/sdrangel.jpg

elitedata, 2019-02-23:
thank you for the response.
i understand how the device selection operates.
but the issue at hand is actually the presets as indicated when i first posted.
i have created a video detailing the issue.
this is the Hack RF transmitting at 6925Khz and being received on a Kiwi SDR.
saving the preset, exporting it, closing the application, opening the application and the preset is not there and cannot be imported.
am i doing something incorrect ?
basically, i have to configure everything from the start each time i open the application.
https://youtu.be/3Xb51OLjzVw

f4exb, 2019-02-24:
Youtube tells me the video is unavailable

elitedata, 2019-02-24:
apologies.
should be available now.

f4exb, 2019-02-24:
This seems Windows specific and even then I have not seen such an issue before. It seems that the preferences are not saved in the registry (Qt uses the registry in Windows). Another strange thing is that the preset export file is not visible in the file explorer when you open the preset imports.

elitedata, 2019-02-24:
the preferences are being saved in "Computer\HKEY_CURRENT_USER"
[image]

elitedata, 2019-02-24:
i found a temp work-around fix.
you have to import the preset into the program before you add any sink and modulator.
this solves the problem but now you cannot delete the preset from the program.
this issue can be resolved by deleting the saved registry key (the program doesnt delete it).
at this point at least, i can finally import the preset which saves me time having to configure everything each time i execute the program.

elitedata, 2019-02-26:
to add:
i found a complete work-around solution for the program to save all audio input/output preferences and presets.
delete the "SDRangel" sub-key under the "f4exb" key in the registry while the program is open.
then choose your audio input/output preferences and then save your chosen preset preset, it will load the audio input/output preferences when executing the program and load the preset when the preset is "double-clicked".
i have not tested this with multiple presets, only a single preset.
it seems the program has some trouble reading/writing from the registry keys, it may be due to how the data within the key appears to be in binary format while the key itself is a string format (some ASCII characters may not be readable in string format)

closed 2019-03-31 by f4exb