f4exb/sdrangel #293: Frequencies greater than 10 GHz.

KeyOz, 2019-02-24:
It would be good if you could also set frequencies greater than 10 GHz.

f4exb, 2019-02-24:
Do you know of any SDR hardware that supports this? I do not see the point of having two leading zeros just for the sake of it.

KeyOz, 2019-02-24:
I use SDRAngel with an LNB with a 9.75 GHz LO for Es'Hail 2 / QO-100, which is broadcast at 10.4 GHz. A direct input of frequencies above 10 GHz would be useful.

f4exb, 2019-02-25:
Then use the transverter mode setup (the button with an "X"). Most of the plugins have one there may still be some missing if this is the case with yours then I can add it. Note that there are still no 10's or 100's GHz digits but they are useless since you will not span over 10's of GHz anyway.  In your case enter 750 MHz for the transverter LO (drop the GHz digit). Then say you want to receive at 10.491 GHz this will actually tune down to 10491 - 9750 = 741 MHz with the transverter conversion engaged and LO set at 750 MHz you will see 750 + 741 = 1491 MHz on the main dial. The important digits are the sub GHz ones (491) because you know that it is at 10 GHz something.

thasti, 2019-02-25:
Isn't the whole point of the transverter feature to make the frequency readout correspond to the correct RF frequency? Having to use some sub-10Ghz-frequency just because the software doesn't have enough digits seems like a weird workaround for me. I would also enjoy having this.

f4exb, 2019-02-25:
Just think a little... Let's say you want to tune the entire 10000 to 10500 MHz band then the dial display will show from 1000 to 1500 MHz. Can you tell me what that GHz digit is used for and why should it waste UI real estate to show just a zero or a one? Now there's a little catch if you cross GHz boundaries say for example between 9900 and 10500 MHz. In this case just figure out that 9 is replaced by a symbolic 1 and 10 by 2 and set the transverter LO to 1750 MHz instead of 750 MHz. No big deal. Then what's next if you tune to the 122 or 250 GHz bands?

I think this discussion can be closed now.

closed 2019-02-25 by f4exb

kareiva, 2022-02-12:
Most SDR receivers are able to overlap to 10GHz, but not sdrangel. I think it is not even worth proving how bad the UX would be if I ever would have wanted to use it the way @f4exb just described. If I am receiving a signal at 10.489GHz, that's what exactly I would like the end user software to show. Period. Not IF, not the LO, not any other kind of math crap that puts cognitive pressure over the end user.

kareiva, 2022-02-13:
Well let's see if I am not too dumb to fork and fix the code myself.

kareiva, 2022-02-13:
I certainly appreciate your right (as the product owner) to call your users dumb for no apparent reason. That concludes the whole discussion and you don't have to prove anything here. Thanks for the insights.

f4exb, 2022-02-13:
> ... to call your users dumb for no apparent reason.

Maybe this is a consequence of your own language:

> Most SDR receivers are able to overlap to 10GHz, but not sdrangel. I think it is not even worth proving how bad the UX would be if I ever would have wanted to use it the way @f4exb just described. If I am receiving a signal at 10.489GHz, that's what exactly I would like the end user software to show. Period. Not IF, not the LO, <b>not any other kind of math crap</b> that puts cognitive pressure over the end user.

I can accept a slightly derogatory language but do not expect a nice reply.