f4exb/sdrangel #294: Crash on REST-API query for sink device channel settings

thasti, 2019-02-24:
I am trying to access the settings of an SSB Modulator in a sink channel using the REST API. My setup has a bladeRF source as deviceSet 0, and a bladeRF sink as deviceSet 1. The RX device has an SSB demodulator on channel 0, the TX device has an SSB modulator on channel 0. 

I **can** query the deviceSet 1 channel configuration report successfully using `http://127.0.0.1:8091/sdrangel/deviceset/1/channels/report`, the crash only happens when trying to get the settings.

However, when trying to query `http://127.0.0.1:8091/sdrangel/deviceset/1/channel/0/settings` (the transmitter deviceSet), sdrangel crashes with a segfault and without additional information. The query data is not returned before that happens.

In contrast: When querying `http://127.0.0.1:8091/sdrangel/deviceset/0/channel/0/settings` (the receiver deviceSet), I successfully get the SSB demodulator settings as intended, so it seems the issue is somehow connected to the SSB modulator.

Anything I can do to aid tracking this down?

thasti, 2019-02-24:
What I can already add from initial debugging efforts: 
- This crash is not bladeRF specific, i.e. happens also using a FileSink.
- Other modulators (AM, FM, ...) are not affected by this bug but instead return the correct information

(I'm using version 4.4.4, forgot to mention that earlier).

f4exb, 2019-02-25:
Yes I just came across this bug incidentally today when basing the new FreeDV modulator plugin code on the SSB modulator. This will get fixed when FreeDV is published.

PR #295 (2019-02-25): fix SSB modulator configuration API query
This patch fixes #294. Seems to have been a copy&paste-issue.
closed 2019-02-25 by f4exb