f4exb/sdrangel #296: LSB mode in SSBModulator is not correctly returned via API

thasti, 2019-02-24:
When the SSB modulator is set to LSB, it still returns positive values for the cutoff and bandwidth parameters when queried via the REST API. Even though the 'usb' flag correctly indicates the mode, it seems like the API is broken here: Just PUT/PATCHing the same values that a GET yielded causes the modulator to switch to USB (even though usb=0 in the request).

To actually set the modulator to LSB mode, the user has to set usb=0 _and_ provide bandwidth/cutoff as negative values.

To fix this, the API callback should:
- _either_ check the USB flag (which it currently does not) and accept only positive values as supplied by the GET request
- _or_ indicate LSB mode by using negative bandwidth/cutoff values in the GET request as the SSBDemodulator does.

I lean towards the second approach. Then, the 'usb'-field could go away completely, the SSBDemodulator does not have one either.

What do you think?

f4exb, 2019-02-25:
Well the idea that was taken to design the REST API settings structure was to take the exact same fields with the leading "m_" removed. This makes it easy to check consistency and relate the field names in both structures. This is why you have the "usb" flag in the REST API. Moreover for PUT/PATCH handling it makes it easier and less error prone to populate the settings structure and just call the applySettings() method of the plugin code. Using negative values for the bandwidth limits is in fact an artifact of the GUI. 

So I'd rather stick to the first approach.

thasti, 2019-02-25:
Okay. While inconsistent with the approach of SSBDemod, that will certainly work. I'll post a fix shortly.

thasti, 2019-02-25:
Okay, the fix provided in #300 does the trick for me. It essentially confines the use of negative bandwidth/cutoff values to inside the GUI. Now, every time the settings are exchanged between GUI and modulator, the GUI makes sure to keep the data structure passed to the modulator consistent. What do you think?

PR #300 (2019-02-25): respect USB flag for SSBMod REST API calls
This commit fixes #296. The API now expects positive values for bandwidth and cutoff and respects the setting of the USB flag in the query. This makes the expected parameters consistent with the ones provided.
closed 2019-02-25 by f4exb