f4exb/sdrangel #297: Clicking anywhere inside the 'Channels' layout disables keyboard commands

thasti, 2019-02-24:
After enabling keyboard commands using the little keyboard icon (which becomes highlighted), I can use commands as intended. However, as soon as I click anywhere in the right 'Channels' column of sdrangel, the button loses highlighting and commands don't work until re-enabled.

The README states that this is supposed to happen when the application loses focus, which is not the case here. I would consider the 'channels' dialog part of the application.

f4exb, 2019-02-25:
It seems Qt does not think this way. There is not much to do to fix this in SDRangel code.

thasti, 2019-02-25:
What is the reason to disable keyboard commands on focus changes in the first place? I find that when getting rid of the focusChanged event completely, keyboard commands just work as intended (and also do so after doing something in the 'Channels'-pane. When the focus is in another window, the application is not getting keyboard inputs, so there are no weird effects when typing text into another application running an parallel.

See https://github.com/thasti/sdrangel/commit/c55b77068727b5a3345506c9f431561be731cc54 for my solution. I wonder if it would be an acceptable change to sdrangel? Possibly with an option to make this feature optional? If you would consider it acceptable, I would go ahead and implement it.

f4exb, 2019-02-26:
This seems to be OK, thanks.

PR #301 (2019-02-26): Don't intentionally disable keyboard commands on application focus change
This fixes #297. Keyboard commands stay enabled until explicitly disabled or program closed.
closed 2019-02-26 by f4exb