f4exb/sdrangel #298: SoapySDR -windows10- sdrangel 4.4.5

lucapgt, 2019-02-24:
I've installed soapySDR on windows 10. SoapySDR is installed and running on PlutoSdr, connected on usb port. I can't see soapySDR source on Sdr Angel. What is the mistake?

f4exb, 2019-02-25:
Just a question: why would you use Soapy for Pluto when you have native support|?

lucapgt, 2019-02-25:
Because in this way i can use Pluto remotely. On Pluto only connect USB WiFi Key and Power cable, and Pluto is in my LAN :-)

f4exb, 2019-02-25:
Can you describe the architecture in more detail? It seems to me this is an issue with SoapySDR remote. Make sure you follow SoapySDR instructions here https://github.com/pothosware/SoapyRemote/wiki and make sure you can connect to your remote by other means (SoapySDR utility). Then only if you cannot see it in the list of SoapySDR devices there could be a problem with SDRangel SoapySDR support.

lucapgt, 2019-02-26:
Yes, i followed the instructions. If i try to connect with cubicsdr works fine, sdrangel not list pluto on soapy.
For example now, connected on usb port: 

[image]
[image]

on cubicsdr
[image]

[image]

lucapgt, 2019-02-26:
last info from pc:
[image]

lucapgt, 2019-03-06:
Hi, any idea?

KC7NOA, 2019-03-17:
Probably a stupid question but did you compile SDR Angel after soapySDR remote was installed?

lucapgt, 2019-03-18:
I'm using Windows and soapy plugin seems to be ok.

closed 2019-04-28 by f4exb

synthead, 2021-09-19:
@lucapgt, how did you make this work?