f4exb/sdrangel #299: DVB-S2

lucapgt, 2019-02-24:
Hi, is there ant chance to add DVB-S2 video decoder in sdrangel?

f4exb, 2019-02-25:
Pretty much depends on what Lean SDR does.

lucapgt, 2019-02-25:
Now leandvb Can decode DVB-S2?

f4exb, 2019-02-26:
It seems there has been progress concerning DVB-S2 on lean SDR probably motivated by QO-100 transponder and so is this request I suppose. So it seems worth having a second look at lean SDR.

lucapgt, 2019-02-26:
Yes, qo-100 is my same motivation

taherian2008, 2019-04-07:
Hi. 
Most of DVB-S/S2 transponders are 36 MHz bandwidth . the maximum processing bandwidth of sdrangel is 10 MHz .  
i want to watch (and capture) the DVB-S transponder with 36 MHz BW on Hotbird 13E .
is it possible to upgrade the windows sdrangel exe version with 36 MHz bandwidth ?

PS : the fft max size in sdrangel is 4K . it is not sufficient to monitor a wide band signal such as 36 MHz transponder. if the fft max size increase to 4M the spectrum will be much better than now.

f4exb, 2019-04-07:
"the maximum processing bandwidth of sdrangel is 10 MHz .": Definitely not. It depends on your hardware. There are of course limits and not a lot of available devices will let you go up the 36 MS/s required for a 36 MHz bandwidth. Maybe the Lime USB and XTRX.

Bear in mind that this plugin is not intended to receive commercial broadcasts but rather amateur digital TV that has much lower bandwidth requirements. Less bandwidth means also less requirements on the processing power and USB bandwidth and is more likely to effectively work.

You are mixing FFT size and bandwidth. The higher the FFT size the finer the details you can see in the spectrum. I very much doubt this is necessary for wideband digital TV signals.

So none of your requirements are valid sorry.

taherian2008, 2019-04-08:
my hardware is LimeSDR USB and the maximum RF bandwidth of Limesdr is 61.4 MHz (Equal to Maximum Symbol Rate) .
before i use SDR-Radio V3 . by Lime-sdr it can capture a 36MHz without any problem. BUT it has not process DVB signals.
Definitely it take a lower time from the developer of sdrangel (f4exb) to increase the DVB processing bandwidth rather than me .   

"The higher the FFT size the finer the details you can see in the spectrum" , it's correct and useful for adjust the satcom antenna to measure the C/N .

f4exb, 2019-04-08:
The sample rate range is obtained via the LimeSuite interface (LMS_GetSampleRateRange). I don' t know of any better way to do it.

GkvJeep, 2019-04-08:
By experience.
Usb 2.0 max 20 MHz/sample  IQ with 16 data bits.
Usb 3.0 max 40 MHz/sample IQ with 16 data bits.

f4exb, 2019-04-08:
```
2019-04-08 23:39:25.942 (D) LimeSDRInput::getLORange: min: 100000.000000 max: 3800000000.000000
2019-04-08 23:39:25.942 (D) LimeSDRInput::getSRRange: min: 100000.000000 max: 61440000.000000
2019-04-08 23:39:25.942 (D) LimeSDRInput::getLPRange: min: 1400100.000000 max: 130000000.000000
```
[image]

f4exb, 2019-07-18:
Implemented in v4.11.1

closed 2019-07-18 by f4exb

j1m1l0k0, 2023-06-14:
does it work capturing KU band? satellite start one D2/C4 - 70.0W