f4exb/sdrangel #302: Windows GUI interface

IW1EPY, 2019-02-28:


IW1EPY, 2019-02-28:
Just beginning  with Pluto - sdrangel , there is a way to use the mouse wheel to fine  tune stations and to rotate all the small pot ? can the small pots stop at the limits?
Thanks in advance for the help

IW1EPY, 2019-02-28:
Sorry i am a beginner on this site

f4exb, 2019-02-28:
Yes to rotate the small pots you move the mouse pointer inside the pot and use the wheel. You an also move the thumbwheel dials this way.

closed 2019-03-02 by f4exb

satfan52, 2019-05-10:
I would like to point a click on the SSB signal of interest in the waterfall display and then use the mouse wheel immediately thereafter to fine tune the reception. I am new to sdrangel and only started to use it very recently, but unless I am mistaken this is not possible. What I noticed is that it is possible instead is to use the mouse wheel in combination with the rotary encoders but this is less convenient as it requires to move the mouse away  from the signal of interest  to the rotary encoders, then  to the fine tune frequency and eventualy go back to the waterfall display to check the next signal of interest (not always convenient if it needs to be done very frequently).  I noticed that the arrows work too to change the rotary encoders so not only the mousewheel, but again it would be more convenient if I could use my mouse wheel after having clicked on the signal of interest in the waterfall frequency display. Thanks for clarifying if it can be done or not with sdrangel. Regards. Peter

f4exb, 2019-05-11:
You can drag and drop the center line in the waterfall

f4exb, 2019-05-11:
Yes it is definitely possible (and just coded) to move the center line with the mouse wheel. Will be in next version and in dev soon.

satfan52, 2019-05-11:
Yes, thanks, I know. The point is that fine tuning the rx frequency is not
easy that way  that's is why the mouse wheel could come very handy to
tlfine tune on an SSB signal