f4exb/sdrangel #303: No broadcast FM / no signal

majodi, 2019-03-01:
No reception (almost) with sdrangel while on the same machine, same dongle on sdr# things are okay. How to find out what the cause can be for this?

So on a dual-boot machine I have Windows 10 and Ubuntu 18.04 running. When using sdr# on Windows and tune to the FM broadcast range (with several strong stations) I see and hear these stations. When I then re-boot into Ubuntu and (with the exact same HW setup) start sdrangel on the same frequency I get nothing. I only see the noise floor and I just can hear a very faint (noisy) signal of the station.

What could be the cause? What can I check?

Thnx

f4exb, 2019-03-02:
Hello,

two things:
  - First you should check the bandwidth: at 950 kS/s and decimation by 16 this makes a 59.375 kS/s sample rate in the baseband and hence 59.375 kHz available to the demod which is much too narrow for broadcast FM. You need at least 250 kHz and hence 250 kS/s I/Q sample rate in the baseband. Therefore at 950 kS/s from te RTL-SDR the decimation factor should not exceed 2. Note that the baseband sample rate appears on the top left corner of the RTL-SDR control below the start/stop button.
 - Secondly you may have to lower the gain control. Broadcast FM stations are usually received with strong signals but this only comes once the first point is solved. 

SDR# uses a fixed sample rate of 2MS/s and no front end decimation. SDRangel lets you control both the sample rate and the decimation factor but this comes at the expense of understanding a little bit how digital signal processing works. However there is an on-going idea (not a ticket yet) to add the possibility to switch from sample rate to bandwidth control at a given decimation factor.

majodi, 2019-03-02:
That makes sense, thank you. So many things to keep in mind. I will try https://groups.io/g/sdrangel for future "beginners" questions. Thnx again for the explanation.

closed 2019-03-02 by majodi