f4exb/sdrangel #305: Can't compile with codec2

FFY00, 2019-03-02:
I got this error after adding `codec2` as a dependency.

```
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp: In member function ‘void FreeDVDemod::FreeDVStats::collect(freedv*)’:
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:72:24: error: aggregate ‘MODEM_STATS stats’ has incomplete type and cannot be defined
     struct MODEM_STATS stats;
                        ^~~~~
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp: In member function ‘virtual bool FreeDVDemod::handleMessage(const Message&)’:
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:390:35: error: ‘unsync’ was not declared in this scope
         freedv_set_sync(m_freeDV, unsync);
                                   ^~~~~~
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:390:35: note: suggested alternative: ‘isSync’
         freedv_set_sync(m_freeDV, unsync);
                                   ^~~~~~
                                   isSync
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp: In member function ‘void FreeDVDemod::applyFreeDVMode(FreeDVDemodSettings::FreeDVMode)’:
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:611:9: error: ‘freedv_set_ext_vco’ was not declared in this scope
         freedv_set_ext_vco(m_freeDV, 0);
         ^~~~~~~~~~~~~~~~~~
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:611:9: note: suggested alternative: ‘freedv_set_tx_bpf’
         freedv_set_ext_vco(m_freeDV, 0);
         ^~~~~~~~~~~~~~~~~~
         freedv_set_tx_bpf
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:612:35: error: ‘manualsync’ was not declared in this scope
         freedv_set_sync(m_freeDV, manualsync);
                                   ^~~~~~~~~~
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:621:18: error: ‘freedv_get_modem_symbol_rate’ was not declared in this scope
         int Rs = freedv_get_modem_symbol_rate(m_freeDV);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/sdrangel/src/sdrangel-4.5.0/plugins/channelrx/demodfreedv/freedvdemod.cpp:621:18: note: suggested alternative: ‘freedv_get_modem_sample_rate’
         int Rs = freedv_get_modem_symbol_rate(m_freeDV);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                  freedv_get_modem_sample_rate
make[2]: *** [plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/build.make:67: plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/freedvdemod.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2740: plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/all] Error 2
```

I am using `codec2` 0.8.

f4exb, 2019-03-02:
Please check: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux#codec2freedv
It explicitly requires release 4067.

FFY00, 2019-03-03:
My bad, sorry for missing that. I'll keep freedv support diabled in the Archlinux package until codec2's upstream makes a new release.

f4exb, 2019-03-05:
I could eventually make an internal library to support FreeDV so it does not depend on the more recent updates of codec2-dev. It can build on stock Ubuntu 18.04 with the packaged libcodec2-dev (version 0.7) This is now available on dev branch and on master at the next release.

FFY00, 2019-03-05:
Thnak you :grin:

closed 2019-03-05 by FFY00