f4exb/sdrangel #307: LimeSDR NCO and frequency tracking

KC7NOA, 2019-03-09:
this  feature request i guess ...

i would like to see the Main frequency display track with changes of the NCO when the nco is active ....

if im trying to listen to a certain frequency and i want to change the NCO frequency to try to move some harmonic that is interfering --- and not need to change back the main frequency display to the frequency i was trying to clean up ...

i would like an option to keep the main frequency on a desired frequency -- say 10mhz wwv --- and change the NCO, can the main display track in such a way that 10mhz wwv is still in the demodulated window?

f4exb, 2019-03-11:
I am even wondering if this should not be the default behavior but let's start with an option.

KC7NOA, 2019-03-13:
Ok ... let me know when a test is avaliable ... i dont get internet too offten

f4exb, 2019-03-27:
This is giving me headaches. Today we have a simple situation where:
Fc = Flo + Fnco and
  - Acting on the main dial (the one displaying Fc) acts on Flo
  - Acting on the NCO dial acts on Fnco
  - And the above equation is always verified

I cannot thing of a simple way to implement compensation of a change in Fnco

f4exb, 2019-03-27:
Something that could work. The equation is transformed from:
Flo = Fc - Fnco (this is the same as Fc = Flo + Fnco)
to:
Flo = Fc - Fnco + Fm where Fm is a frequency compensation for the variations of the NCO. Initially Fm is set to 0 and when the NCO lock is engaged any time the NCO frequency is changed it is changed by the same amount. When the NCO lock is relased Fm is reset to 0. Fm values are capped by the limits of Flo.

f4exb, 2019-03-27:
It could work by storing the Flo and Fnco at time of lock say Flo_ref and Fnco_ref respectively
Then Flo is calculated as:
Flo = Flo_ref + Fnco_ref - Fnco
Then acting on the main dial acts on Flo_ref. Fnco is just the same. Flo is hard limited from 30 to 3800 MHz. Then Fc is whatever it is.

f4exb, 2019-04-05:
Sorry I don't see a way to implement this cleanly and safely. Let alone the fact that both Rx (or Tx) can have different NCO frequencies or may use or not the NCO.

closed 2019-04-05 by f4exb