f4exb/sdrangel #309: LoRa revival

f4exb, 2019-03-15:
This has been requested through direct mail and I also have been thinking of doing it some day. Let's make a proper ticket... 

The present LoRa demod plugin is a legacy from the old Sdrangelove base and was not really taken care of since (well it just compiles) so in fact it is not really functional.

This ticket is open to make a functional LoRa demodulator plugin and also a modulator.

Zapero, 2019-09-17:
This would be great!

f4exb, 2020-01-29:
LoRa modulator to be handled in a separate issue.

f4exb, 2020-01-29:
https://www.link-labs.com/blog/what-is-lora
https://github.com/BastilleResearch/gr-lora

f4exb, 2020-02-03:
More info here: https://www.ntu.edu.sg/home/limo/papers/TOSN-LoRa.pdf

f4exb, 2020-02-07:
The demodulation part shows that +/-1 bit error are common on FFT argmax detection. This paper explains how this is mitigated by Gray coding: https://myriadrf.org/news/lora-modem-limesdr/

f4exb, 2020-02-26:
Implemented in v5.2.0

f4exb, 2020-03-11:
LoRa demodulator plugin is now ChirpChat demodulator plugin. It has a LoRa compatible mode.

f4exb, 2020-11-08:
To be retrofitted to master as v6

f4exb, 2020-11-12:
Deployed in v6.0.0

closed 2020-11-12 by f4exb