f4exb/sdrangel #310: Possibility to enter baseband sample rate at fixed decimation

f4exb, 2019-03-15:
For the devices that support arbitrary sample rates add the possibility to control the baseband sample rate for a given decimation factor. This will be a GUI only feature using the same thumbweel dial as the sample rate making the "SR" label a toggle button making it possible to switch from one mode to another.

Plugins concerned: BladeRF all versions, HackRF, LimeSDR, PlutoSDR, RTL-SDR, XTRX.

KC7NOA, 2019-03-16:
Do you mean to code .... some thing to change the hardware decimation rate if the requested frequency is unattainable based on Sample rate and NCO frequency offset combination?

f4exb, 2019-03-16:
NCO? Does RTL-SDR have a NCO? I mean at a given decimation give the baseband sample rate instead of the device sample rate. Of course the limits have to be set accordingly.

f4exb, 2019-04-14:
Implemented in v4.5.4

closed 2019-04-14 by f4exb