f4exb/sdrangel #312: Reverse API users need better way of determining correct device index for calling REST API

texasyojimbo, 2019-03-21:
Currently, Reverse API calls use the reverse API index assigned by the user to the device when the reverse API trigger is activated.

This index is not necessarily the same as the index that is autogenerated when the device is created (the index that is needed to get other info about the device through the REST API, for example the frequency, device type, antenna setting, etc).

Had e-mail discussion with Edouard about this in the past week.

texasyojimbo, 2019-03-21:
From e-mail:

In fact the issue is that you don't have the particulars of the device (or device and channel if you would deal with channels) to know from which device slot.the request is.originating and thus this makes 2 way coordination difficult. 

I have to think a little.bit about it but may be adding some.keys to the Json structure.about originator information could solve the issue. This makes the API not exactly the same in one way or the other but adding keys should not have an impact if you were targeting another SDRangel instance.

f4exb, 2019-03-27:
Implemented in v4.5.2

closed 2019-03-27 by f4exb