f4exb/sdrangel #319: Low / No audio with HackRF

keyboarderror, 2019-03-30:
Hello. I am experiencing problems with very low audio from the transmit sink. The microphone in Linux is typically 10-20%, while the test tone generator gets a full 70%. All modes. Boosting audio does not help. FreeDV makes decodable noise but it's not intelligible. In Windows 10 there is no audio from anything including the test tone. It will produce carrier in appropriate modes most times. Windows is more quirky. I am currently running SDRAngel 4.5.2 in both operating systems but this has been going on for a while. It was not always this way. GNURadio transmit flowgraphs in Ubuntu work fine and have good audio so I don't think it is in the hardware. Running a HackRf under an Intel i7 on a USB3 port with Ubuntu 18.04 LTS / Windows 10 Build 1809. I've also tried compiling from source in Linux in addition to pre-compiled. No difference. Any help would be appreciated. Thanks!

f4exb, 2019-03-31:
Can you make a precise description of all the parameters you are using for a defined modulator (ex: AM)? Assuming Linux. I am not giving any support for Windows.
  - parameters of the modulator (including which device is selected by right clicking on the mic)
  - your audio levels with pulseaudio (using pavucontrol)
  - the audio device settings (Preferences -> Audio -> Audio Input tab)

keyboarderror, 2019-03-31:
Ok. Audio is the onboard sound fed via line input from a mixer and microphone. Audio input and output are system default, managed by pavucontrol. Using 12.5kHz audio bandwidth. Input level is slightly above base. I can increase audio input gain in SDRAngel from 1.0 to 2.0 and %100 modulation and get a bit more of a boost. Even at the point of input distortion, it is never more than %50 modulation. To a receiver it is too quiet. The test tone is consistently stronger. HackRF is at 10M samples and an Int of 1.

If a picture is a thousand words, here is a video.

https://youtu.be/2T-9qU4tNis

keyboarderror, 2019-03-31:
Update on Windows support. In a moment of futility testing I ran Zadig 2.4 selected List All Devices and the option was to downgrade WINUSB (v10.0.17763.1) to (v6.1.7600.16385). This does not seem to happen when I do it, but it goes though the process and after the driver is still shown as v10.0.17763.1 but after unplugging and replugging the HackRF I have NORMAL audio! This makes me happy. It might be a tip for anyone else experiencing problems connecting.

elitedata, 2019-03-31:
after watching your video, its been my experience that the bottom level indicator on SDR Angel is your percent of TX modulation output, im not entirely sure what the top level indicator is for but it does not appear to have any effect on TX modulation output.
if you would like to have consistent audio, i recommend using "StereoTool" broadcast audio processor and VB virtual audio cable.
here is a sample FLV video recorded and received locally from SDR Console using an RSP2, i was able to achieve these results with SDR Angel, StereoTool & VB virtual audio cable on a windows 10 64 bit system.
http://s000.tinyupload.com/index.php?file_id=00560382261822324578
the video is low quality due to attachment file size restrictions on tinyupload.
this is 100% AM modulation @ 32Khz bandwidth (16Khz audio)
i also use a different method to achieve better AM modulation results rather than using the AM module itself.
i use two modules, the NFM module with zero modulation to generate the highest possible carrier (i dont use the built-in RF amp) and the SSB module configured to DSB modulation to generate AM modulation (see screen capture attachment for example).
all modulation & bandwidth settings for the SSB module (DSB) on SDR Angel are "wide open" since i use StereoTool to control the modulation level and amount of modulated bandwidth (frequency response)
[image]
.

keyboarderror, 2019-04-01:
@Elitedata I appreciate the suggestion. I've been using the dynamic compressor in VLC to pass audio to a modified CQUAM AM Stereo flowgraph. (No stereo pilot, no Quadrature modulator, faster sample rate, lower audio bandwidth). It gives nearly broadcast quality mono AM. But my SSB flowgraph isn't nearly as good as SDRAngel's modulator. I will check out the Linux version of StereoTool. My success with SDRAngel Windows sound is now tempered by the fact that I can only activate transmit the very first time. Subsequent transmissions stream garbage on the last set frequency regardless of other settings. Only a full system reboot will allow another single transmit. And received transmission is ragged and full of errors. So not very useful.

f4exb, 2019-04-01:
The level indicator is a VU meter in percentage of the modulating analog signal full scale. The top moving bar is the average and this is what you see at 70% (green circle). The bottom bar in darker color is the peak and it has a tiny red peak hold (yellow circle). This should never exceed 100% (go into the red zone) else clipping or splatters will occur. In the picture below this is the AM modulator with the sine (test) signal that is calibrated to be slightly below 100% peak. Since it is a sine its average is about half sqrt(2) that is 70%. 

[image]

In the first video I see that when you speak loud the peak reaches 100% so that's correct.

There is also a input attenuation (gain <= 1) setting in the audio preferences / audio input panel that you can set by device. See: https://github.com/f4exb/sdrangel/blob/master/sdrgui/audio.md#26-input-volume The default is rather low (0.15) but was found to be fine in most cases. If this is not the cae for you you can increase it. Note that it will not work live if your modulator is running you will have to either change the input device and come back to the same or delete and re-create the modulator.

Another thing, you are setting the sample rate at 10 MS/s without decimation so there is no chance to see anything about the modulation of your signal in the spectrum window. The narrowband audio modulators work at 48 kS/s. Their upsamplers are not designed to handle such a big gap so things might work or not. Normally you would use the half band interpolator (on the device panel) to have a smaller sample rate in the baseband (32 or 64 would be fine).

So far I don't see anything wrong with the audio input.

keyboarderror, 2019-04-02:
That helped. Turning the audio input attenuator all the way down (up) to 1.0 and the modulator input gain up to 2.0 gets %80 modulation peaks. I can get yellows and red in the waterfall now. The received audio is much stronger. The high sample rate is to allow a Δ offset of 3 MHz or more from the center frequency for the SSB modulators. Any residual carrier can be removed by an external band pass filter prior to transmission. It isn't actually needed for AM/FM. The sample rate may also be why there can be a delay of several seconds between pushing the end transmit button and the device actually stopping after very long transmissions. I'll do some more testing. Thank you!

f4exb, 2019-04-02:
At this time there is no provision to move the center of transmission baseband away from the zero IF of the device. With the HackRF there is a remaining DC spike at zero IF although it is not as bad as on the Rx side where moving away from the center was there since the beginning. In AM the DC peak is hidden by the normal AM DC peak it may or may not work for FM although it should do for narrowband FM. With the next version 4.5.3 the baseband shift on Tx side will be available for the HackRF. Originally this is an attempt at doing a proper synchronization of frequencies between Rx and Tx that is not working presently if you have a baseband shift on Rx side.

keyboarderror, 2019-04-03:
Wow! Compiled 4.5.3 and HackRF output is dramatically better. The DC offset takes care of the spur. I've decreased the sample rate and interpolation accordingly. That solves the device control delay. Great work! 73

elitedata, 2019-04-03:
im just waiting for a windows 64 bit port of 4.5.3 please.

keyboarderror, 2019-04-03:
If you'd like a bit more feedback on the source/sink frequencies track ok, but the transmit/receive handoff doesn't quite work. One can stop the other, but it requires multiple clicks to toggle modes.
Also with a second machine built I can confirm the FreeDV audio still has problems but that may be completely unrelated. I seem to have strayed from the original issue.. We really appreciate all your efforts, Edouard.

f4exb, 2019-04-03:
There is no auto Rx/Tx switch. This can be done via REST API. See:
  - https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/ptt_active.py
  - https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/ptt.py

f4exb, 2019-04-03:
@elitedata : your obstinacy in using Windows is pathetic.

elitedata, 2019-04-03:
because the programs i use in conjunction with Hack RF and SDR Angel are only windows specific.

keyboarderror, 2019-04-04:
@elitedata: You might try using Voicemeeter and VBAN to pipe audio from Windows. There is VBAN support in Linux for the audio. Up to 196Khz.

elitedata, 2019-04-04:
@keyboarderror thanks for the suggestion but at this point, i do not have any issues operating SDR Angel and Hack RF on my windows 10 x64 based system.
my earlier post above, with the included screen captures and video file was a simple demonstration of the current configured operation which uses VB virtual audio sound cards and StereoTool audio processor in conjunction with SDR Angel and Hack RF to make a simple Hi-Fi AM broadcast.

keyboarderror, 2019-04-04:
@Elitedata It wasn't immediately clear to me that you weren't having similar problems. I'm not sure why you'd complain. If transmit is working fully for you I will look into it more. Turns out that cycling the receiver source in Windows SDRAngel resets the HackRF enough here to transmit again without the need to reboot. Something in the handshake. Well...

elitedata, 2019-04-04:
@keyboarderror complain about what ?
i was not complaining about anything.
i was providing a demonstration of my current operation using windows 10 64bit and a description of what the top and bottom level indicators likely meant (that f4exb cleared up).
i had a hunch the top indicator was for "average" RMS reading but this wasnt documented (anywhere i could find) so i was not going to assume it, i clearly knew what the bottom indicator meant which is most important anyway.
if it was my comment about waiting for the version 2.5.3 port to windows, well, thats because of f4exb's previous comment in another discussion about "not supporting windows releases" which led me to believe he may be ceasing further release ports for windows, which i hope he doesnt do.
the only "complaining" that i have commented about so far, is that i am considered to be "obstinate" and "pathetic" for using SDR Angel on windows.
anyway, i am finished commenting further about this subject since i appear to be the only "obstinate" and "pathetic" person using SDR Angel on windows.
have a good day :)

f4exb, 2019-04-04:
This wasn't documented anywhere: https://github.com/f4exb/sdrangel/blob/master/plugins/channeltx/modam/readme.md#8-level-meter-in-

Sometimes I am wondering why I am writing any documentation.

keyboarderror, 2019-04-04:
Ha! True. But the combination of changes and problems accumulating on my end led to a belief in issues beyond something as obvious looking through the manual! Apologies for such a simple issue in what is clearly a very complicated piece of software. The "no support" response does kind of block exploring or discarding parallel issues. I'm going to check Soapy in Windows support and Docker. I'll try not take up more of your time for a while unless it was for a hypothetical beer. Cheers.

closed 2019-04-04 by keyboarderror