f4exb/sdrangel #320: Custom RFBW in NFM/WFM

cocus, 2019-04-01:
Hi,
I've been playing with this SDR and I can tell it's really cool compared to the industry standard.
However, there's a simple thing missing. I listen to some FM radios on the 200MHz band and they don't have standard bandwidths, in this case one has 50KHz and the other one 30KHz. The RFBW combo box only allows to select pre-definde values, which none of them are useful in this situation.
Would it be possible to use a combo box with a text entry? or even better, a up/down control. The ultimate solution could be something like they do on Adobe Premiere.
If this has been already implemented please forgive me.

Thanks!

f4exb, 2019-04-02:
This is of course possible and makes some sense. From comments I see however that this concerns mostly the BFM and WFM demods. NFM has pretty much standard bandwidths and associated expected excursions and I am not too keen in changing that. The reference to Adobe Premiere is unclear for me. Anyway the application wide way to implement bandwidth control is with thumbwheel dials.

cocus, 2019-04-03:
Hi,
Yes, the NFM should have the presets only, but the WFM could be modified to have this new scheme.
About the Adobe Premiere (and probably other Adobe products), it's their way to implement the thumbwheel dials. They allow you to enter a number by clicking it, or if you left click and drag the mouse inside the control boundary. Scroll wheel also works. But that would be a cosmetic change, rather than a functional change.
By the way, is it possible for the thumbwheel control to stick into pre-set positions (i.e. the ones we currently use for WFM)?

f4exb, 2019-04-03:
I don't know if you have practiced the present thumbwheels very much but this is pretty much the same. You can also click and type directly. When a digit is selected you can use the arrows or the mouse wheel. You can also just mouse over and use the mouse wheel.

For predefined values the combo is still the most convenient and consistent with the rest of the UI so I would keep it like that and use thumbwheel controls for continuous values within a range.

cocus, 2019-04-03:
Oh, sorry, I was confused about what's a thumbwheel. I was thinking about knobs! My mistake. 
Regarding to the custom RFBW, for this usage, you'll have either a predefined values that you could use, or a custom bandwidth. What would be the best approach? I think the thumbwheel would do, but there's no way to use predefined values there.

f4exb, 2019-05-16:
Unless you need DSD+ functionality explicitely there is a receiver plugin that does D-Star, YSF, DMR, dPMR and NXDN: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/demoddsd

For piping into DSD+ the main issue you will encounter with NFM demod is the high pass filter to cancel CTCSS that cannot be switched off. WFM may not have such a thing which is why you resort to it. You can also have a look at the UDP sink and find a way to transfer the audio samples from UDP to DSD+: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/udpsink

f4exb, 2019-05-16:
Switchable audio high-pass on NFM will be handled in #343

Pedro9012, 2019-05-16:
Muchas gracias por tu respuesta , uso DSD+ con las señales deviles que el complemento de SDRANGEL no es capaz de decodificar , (WFM demodulador ,  RFBW 12,5,  AFBW 3 )  y va de maravilla.
Thank you very much for your answer, I use DSD + with the deviles signals that the SDRANGEL addon is not capable of decoding, (WFM demodulator, RFBW 12,5, AFBW 3) and it goes great.
Sorry to take the opportunity to consult you because from version 4.3.2 you can not transmit anything more than once and you have to go back to rx and again to tx, I have hackrf with firmware 2017 because with 2018 SDRANGEL I do not recognized (windows 10).
Perdona que aproveche para consultarte por que a partir de la version 4.3.2 ya no se puede transmitir nada mas que una vez y hay que volver a rx y otra vez a tx , tengo hackrf con firmware 2017  porque con el de 2018 SDRANGEL no lo reconocia (windows 10).

f4exb, 2019-05-17:
Which are the modes DSD+ can decode and SDRangel (DSDdemod) can't? P25 maybe?

Pedro9012, 2019-05-19:
hello, it is not that there is some way that one yes and another does not, it is because of quality signal (santiago 8, 9, 9 +, etc) sometimes it does not decode Sdrangel and with DSD + yes, but other times it is the opposite DSD + no It goes well and Sdrangel perfect.
A greeting.
hola , no es que alla algun modo que uno si y otro  no ,es por cuestion de calidad señal (santiago 8 ,9, 9+,etc)unas veces no decodifica Sdrangel y con DSD+ si , pero otras veces es al contrario DSD+ no va bien y Sdrangel perfecto.
Un SALUDO.

f4exb, 2019-06-09:
Implemented in 4.9.1

closed 2019-06-09 by f4exb