f4exb/sdrangel #321: Gain is not properly set for RTL-SDR when changing presets

cocus, 2019-04-03:
Version 4.5.2 running on Windows (I don't think that has anything to do), with an RTL-SDR as a sampling device.
Whenever I change presets, the gain of the selected preset is correctly shown, even the slider is set correctly, but you can clearly see in the spectrum that it's really low.
Sliding the gain control down a notch and then up a notch sets the gain to the one set in the preset.

As a side note, changing the tuner frequency manually doesn't affect the gain at all; so I think the issue might be related to the RTL-SDR when a preset is loaded.

f4exb, 2019-04-03:
I also noticed that and this should be a valid issue. It has been there for a while already and lies within the code (not an O/S related stuff as you noticed).

cocus, 2019-04-03:
@f4exb I've been looking thru the code and the debug output, and I can confirm that the deserialization does indeed work and sets the gain fine. My theory is that one of the other settings might override the gain value; and I couldn't find which one, but probably the gain value should be set last?

Probably moving the entire if() block from line https://github.com/f4exb/sdrangel/blob/ab49d4bf19e05f5a645c3e7d065a596a53bf60df/plugins/samplesource/rtlsdr/rtlsdrinput.cpp#L383
to where line 554 currently is.

Also, a small UI bug related to the RTL SDR plugin, the RFBW of the sampling device is not correctly set in the thumbwheel after selecting another preset, although, it gets deserialized and set in hardware correctly.

f4exb, 2019-04-03:
Yep, well done. Just tried that and it works!

f4exb, 2019-04-14:
Fixed in v4.5.4

closed 2019-04-14 by f4exb