f4exb/sdrangel #323: Source Selector in Windows 10; High Resolution Displays doesn't display devices

rgetz, 2019-04-10:
SDRangel 4.5.3; Win 10; 4k display. by default, I don't see the drop down list in the source selector, and clicking everywhere doesn't display the drop down box.

[image]

If I change properties -> compatibility ->  Override high DPI scaling:
[image]

I get something that works:

[image]

-Robin

f4exb, 2019-04-11:
"If I change properties -> compatibility -> Override high DPI scaling:"
  - Any more details on how to do that?
  - Do you have any idea on making it programmatically?

This is an annoying recurring issue (not restricted to Windows) and I would like to fix it once for all.

f4exb, 2019-04-11:
See #103

rgetz, 2019-04-15:
For Windows - there are two options (that I can find):
- Create a new file GdiScaling.manifest in your project. See : https://stackoverflow.com/questions/46428510/enhanced-system-dpi-scaling-with-vs2017
 - call IsValidDpiAwarenessContext and SetProcessDpiAwarenessContex See : https://stackoverflow.com/questions/49333034/programmatically-override-the-high-dpi-awareness

I don't think either of those fixes the actual root cause (which is some Qt DPI issue for the widget that is being used).

f4exb, 2020-05-20:
Suppsedly fixed with #522 (cannot test it)

closed 2020-05-20 by f4exb