f4exb/sdrangel #324: Cannot enable audio input Windows

Retartist, 2019-04-16:
When attempting to set an audio input, nothing happens and no sound is transmitted 
[image]

f4exb, 2019-04-16:
You have no input selected on your modulator. Please read documentation before opening issues.

closed 2019-04-16 by f4exb

Retartist, 2019-04-16:
I will admit that the photo isn't the best, but even when enabling the input on the modulator, nothing comes out. After further tweaking, I found out that the systems default audio input will work, but i cannot change that in the program. E.g. if the default device is the microphone, the microphone will work, BUT if i wanted to change it to the Virtual cable, the program will ignore my selection.

f4exb, 2019-04-17:
The source is selected by right clicking on the microphone: https://github.com/f4exb/sdrangel/blob/master/plugins/channeltx/modnfm/readme.md#104-audio-input-select-and-select-audio-input-device