f4exb/sdrangel #342: funcion equipo hf/u/v

Pedro9012, 2019-05-15:
Could you include a button for tx that commutates between rx / tx in the same frequency and mode without having to stop reception, change screen, activate tx, stop tx, change screen again, activate rx, and so on?
As of version 3.2, it is no longer possible to transmit twice in a row, when short tx is no longer functional until you return to rx active / deactivated.

¿se podria incluir un boton para tx que commutara entre rx/tx en la misma frecuencia y modo sin tener que  parar recepcion ,cambiar de pantalla,activar tx,parar tx,volver a cambiar de pantalla,activar rx , y asi sucesibamente?.
A partir de la version 3.2 ya no permite transmitir dos veces seguidas , cuando cortas tx ya no es funcional asta que no vuelves a rx activas /desactivas .

f4exb, 2019-05-16:
This has been already covered in a few occasions. I will NOT include such a thing. You have to use an external application (script) to do so. This script makes use of the REST API. Please see an examples here: 

  - https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/ptt_active.py
  - https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/ptt.py

closed 2019-05-16 by f4exb

Pedro9012, 2019-05-16:
Muchas gracias por tu atencion.
un saludo.