f4exb/sdrangel #343: A bunch of few small changes

f4exb, 2019-05-16:
I am packing here a list of small changes already developed (in dev branch) or to be done for next patch release:
  - Ability to use the mouse wheel to move the channel center line in the main spectrum window
  - Splash screen to wait for program startup when the log is not visible
  - Make the audio high-pass switchable on the NFM demod to allow piping audio in apps requiring DC (ex: DSD+, Multimon...)

satfan52, 2019-05-16:
Great!

f4exb, 2019-05-30:
Done in v4.8.1

closed 2019-05-30 by f4exb

satfan52, 2019-07-22:
Hi Edward, 
Is the Ability to use the mouse wheel to move the channel center line in the main spectrum window already activated in the Master repository ? I tried it a few days ago after doing a "git pull" and after recompiling and it did not seem to be available yet. Thanks for clarifying. Regards. Peter

f4exb, 2019-07-23:
Hello Peter,
yes it has been there for quite some time now. I use it routinely.
Brgds, Edouard.

satfan52, 2019-07-23:
Thanks, I propably fail to recompile the right way then. Will investigate
further.

satfan52, 2019-10-07:
Hi Edward, regarding the mouse wheel, it works but the steps are 10hz ;-( 
I am used to steps of 100 hz, it is much faster (10 times ;-) and my ears don't suffer much.
I checked your REST API settings and unless I am mistaken it is no there ?
I understand you may not want to increase it to 100 hz as the matter is somehow subjective, but if you tell me where I can manually edit it in the code, I will just change what needs to be  updated and recompile.
Many thanks ;-)

f4exb, 2019-10-07:
Ctr + wheel -> steps of 100 Hz
Shift + wheel -> steps of 1 kHz.
Cheers.

mathisono, 2019-10-07:
Can we get waterfall zoom on the scroll wheel too?

satfan52, 2019-10-08:
Yes please 😌, it is an excelent idea