f4exb/sdrangel #344: SoapySDR: can receive with the redpitaya but can-not transmit.

satfan52, 2019-05-16:
Hello,

I tried using the redpitaya "SDR transceiver" with "Sdrangel , "soapysdr" and the "SoapySDRredpitaya" module. I was only able to receive, impossible to transmit and Sdrangel does seems to report any errors in the console.

I am using the latest version of sdrangel (recently compiled in the Linux Ubuntu 18.04 environment). As I was able to use sdrangel sucessfully with my plutosdr (both with iio and soapysdr drivers) so I think that my Sdrangel and Soapysdr set-up are probably OK.
 
Since there is no discovery mechanism implemented yet in the SoapSDRredpitaya module, I had to hardcode the the static IP address of myredpitaya in the .cpp file. It works well in reception so it should logically work in transmission too according to the developper, but it is unfortunately not the case. 

I initially opened a ticket  on the SoapySDRredpitaya github respository but I concluded after some discussion with the developer that the Sdrangel repository is more suited to tackel this issue  considering that everything works fine in reception mode and only TX is the issue.

More information (including Sdrangel console info) in the initial ticket opened in the SoapySDRredpitaya github respository

https://github.com/pothosware/SoapyRedPitaya/issues/5

Thanks
Regards
Peter

f4exb, 2019-05-17:
Sorry I do not have a Red Pitaya to test. If it were the case I would have made a native plugin.

pavel-demin, 2019-05-17:
I see that no callback is called for `InterpolatorFloat` in `SoapySDROutputThread::run` and in `SoapySDROutputThread::callbackMO`:

https://github.com/f4exb/sdrangel/blob/master/plugins/samplesink/soapysdroutput/soapysdroutputthread.cpp#L170-L173

https://github.com/f4exb/sdrangel/blob/master/plugins/samplesink/soapysdroutput/soapysdroutputthread.cpp#L254-L257

Doesn't it mean that the buffer is never filled?

f4exb, 2019-05-17:
Yep well spotted. I am opening a specific issue for this (not RedPitaya specific).

pavel-demin, 2019-05-17:
Regarding the IP address problem. 

Would it be possible to add a `QLineEdit` field to the `soapysdroutput` and `soapysdrinput` UIs and to append the value of this field to the `kwargs` variable in `DeviceSoapySDR::openopenSoapySDRFromSequence`?

It would allow to configure the IP address and other parameters via the UIs.

f4exb, 2019-05-17:
Not sure... this is a device specific thing and should be handled in the SoapySDR interface.

pavel-demin, 2019-05-17:
SoapySDR is also used in other projects like [Pothos](http://www.pothosware.com) and [QSpectrumAnalyzer](https://github.com/xmikos/qspectrumanalyzer). Both these projects provide a field to input the arguments for the SoapySDR modules. I think that these two examples are enough to show that it's technically possible to implement.

The arguments is a part of the SoapySDR API and they are used by several SoapySDR  modules (SoapyBladeRF, SoapyRedPitaya, SoapyUHD, etc.).

f4exb, 2019-05-17:
This is probably where the "universalism" of SoapySDR falls short and why I was reluctant to implement it finally doing it under popular pressure. This is also why this issue is directed at native support of RedPitaya.

Anyway if you mean a line edit to type in a freeflow argument string then this is fine (and also to be covered in another issue).

satfan52, 2019-05-17:
In reception mode, it would also be useful, for the end user  to be able to control (one way of the other) the GAIN of the redpitaya. The reason I am pointing this out specifically  is because I noticed an obvious an very anoying problem with the reception of SSB signals with the redpitaya in the sdrangel and sopaysdr ecosystem. The audio is saturating and clipping on most of the ssb signals, sometimes its not too bothering and at other times it's just impossible to listen to the signal. Everything happens as if the redpitaya was "saturating" sdrangel  or as if the audio system of my PC was getting saturated, leading to these noisy anoying clipping effects. I noticed that the gain slider in the sdrangel GUI is currently greyed when the redpitaya is defined as a soapysdr source, so I can't reduce it to attempt to get rid of that problem.

pavel-demin, 2019-05-17:
> In reception mode, it would also be useful, for the end user to be able to control (one way of the other) the GAIN of the redpitaya.

Red Pitaya doesn't have control.

satfan52, 2019-05-17:
OK fine. I  guess this audio quality issue (saturation and clipping on SSB
signal with redpitaya defined as source ) should to be addressed in an
other more appropriate and relevant  way, but I think it should be
addressed that is my point.I will try to upload an audio sample here

satfan52, 2019-05-17:
Here is an audio sample recorded in the 40m band. Weak or modest SSB signals are Ok, but strong ones get clipped somehow. The stronger the signal the more clipped it is. 

[captured_audio_redpitaya_sdr_txcv_soapysdr_sdrangel.zip](https://github.com/f4exb/sdrangel/files/3193230/captured_audio_redpitaya_sdr_txcv_soapysdr_sdrangel.zip)

pavel-demin, 2019-05-18:
Since I'm not an SDRangle expert, the following could be completely meaningless. If it's so, then please ignore it.

Looking at the [demodssb  documentation](https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodssb/readme.md), I've found the 'Volume and AGC' section showing several volume and AGC controls.

I'd naively suggest that changing some of the volume and automatic gain settings could help to solve the saturation and clipping problem.

satfan52, 2019-05-18:
Well I played with the different SSB demodulator settings and yes I was able  to decrease clippings effects significantly. It's actually so basic that I didn't think about it at first. The trick that worked for me is to lower the volume knob of the ssb demodulator to 2.0 (it is set by default to 3.0) and to compensate by increasing the general volume of the PC.

f4exb, 2019-05-19:
The defaults have been set for weak signal work without AGC. It might not be practical in other use cases.

f4exb, 2019-05-22:
While the final idea is to get native support in the SoapySDR plugin the support of passing IP address via GUI cannot work. This is because this is a parameter that should be discovered at device scan time to be used by the opening method that precedes the GUI instantiation. Actually the scanning process for Soapy looks for different possible exposed key value pairs during enumeration and the "addr" key comes last so it can be the case that another one ("serial" or "device_id") is also present and takes precedence. More on that when I can experiment with the RedPitaya.

pavel-demin, 2019-05-23:
> While the final idea is to get native support in the SoapySDR plugin the support of passing IP address via GUI cannot work. 

Strange. It works in [Pothos](http://www.pothosware.com) and [QSpectrumAnalyzer](https://github.com/xmikos/qspectrumanalyzer). Here is [a link to an issue comment with a screenshot](https://github.com/pothosware/SoapySDR/issues/55#issuecomment-167004955) showing how the `addr` parameter can be set in Pothos GUI and here is [a link to an issue comment with a screenshot](https://github.com/pavel-demin/red-pitaya-notes/issues/695#issuecomment-381334014) showing how the `addr` parameter can be set in QSpectrumAnalyzer.

I think that it could be useful to have two options:
 - scanning with automatic discovery
 - manual configuration

While automatic discovery works well for USB devices, it doesn't for network devices. The discovery protocols like Bonjour/DNS-SD can't cross subnet boundaries. So, it could be useful to have a manual configuration option if one wants to connect to a network device from a different subnet.

f4exb, 2019-05-23:
SoapyRemote can do self discovery on IP addresses so it could be worth having a look. 

If self discovery cannot be done for SoapySDR/RedPitaya then it is certainly not an option to use the plugin UI because this UI is tied to a particular device so this device should already be discovered. This is why I have ruled out this option completely. The self discovery is at the heart of the process when SDRangel starts so it is pretty difficult to implement a "manual" discovery later and at the moment I cannot see an elegant way to do it. So for the moment you have to take it as a limitation.

satfan52, 2019-05-23:
But why am I able to receive with the redpitaya (after hardcoding the
address in the soapysdrredpitaya module and not transmit? This is so
counter intuitive! Looks like the module code is using __addr to receive
and addr to transmit? __addr is hard-coded but addr is supposed to be
returned and it does not happen for the reasons you explained above? Can
the module not use __addr for both TX and RX?

satfan52, 2019-05-23:
and if sdrangel discovers the soapysdrredpiraya  module (even when no
redpitaya hardware is present) and is able to use the redpitaya in
reception when the device is present, why can't sdrangel also use the same
ip in transmission?

Apologies, I am not a programmer, just a common sense guy

f4exb, 2019-05-23:
`soapysdrredpitaya` module is nothing known in SDRangel so I suppose you are referring to the SoapySDR support module for RedPitaya and I cannot speak for it. Anyway I really need hands on experience with it to understand what happens. In any case this does not change the fact that the details on how to access the device (i.e. the address) should be known prior to opening it hence at enumeration time. If this is not possible and has to be given by the user then one needs to find a nice way (or not too ugly) to do it. I have some ideas but I need to mature them a bit.

pavel-demin, 2019-05-23:
> why can't sdrangel also use the same ip in transmission?

This question is already answered in [these two](https://github.com/f4exb/sdrangel/issues/344#issuecomment-493432719) [comments](https://github.com/f4exb/sdrangel/issues/344#issuecomment-493433623) and in [this issue](https://github.com/f4exb/sdrangel/issues/345).

In the current version, the transmission doesn't work with the `SoapyRedPitaya` module because `SoapySDROutputThread` doesn't fill its output data buffer when the data format is `CF32`.

f4exb, 2019-06-10:
Soapy for Red Pitaya does not work and therefore is not an option:
```
SoapySDRUtil --args="driver=redpitaya" --rate=1250000 --channels=0 --direction=RX
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] SoapySDR::loadModule(/opt/install/SoapySDR/lib/SoapySDR/modules0.7/libHackRFSupport.so)
  dlopen() failed: libhackrf.so.0: cannot open shared object file: No such file or directory
[ERROR] SoapySDR::loadModule(/opt/install/SoapySDR/lib/SoapySDR/modules0.7/librtlsdrSupport.so)
  dlopen() failed: librtlsdr.so.0: cannot open shared object file: No such file or directory
[ERROR] SoapySDR::loadModule(/opt/install/SoapySDR/lib/SoapySDR/modules0.7/libsdrPlaySupport.so)
  dlopen() failed: libmirsdrapi-rsp.so.2.13: cannot open shared object file: No such file or directory
[ERROR] SoapySDR::loadModule(/opt/install/SoapySDR/lib/SoapySDR/modules0.7/libuhdSupport.so)
  dlopen() failed: libuhd.so.3.15.0: cannot open shared object file: No such file or directory
Stream format: CF32
Num channels: 1
Element size: 8 bytes
Begin RX rate test at 1.25 Msps
Starting stream loop, press Ctrl+C to exit...
Error in rate test: sendCommand failed: 0
```
I get the same error trying to start the device in SDRangel.

satfan52, 2019-06-10:
Hey Edward, it will not work without installing first the SoapySDR
RedPitaya module. After installing the module it should be OK. I was able
to use the "redpitaya SDR transceiver" with Sdrangel in reception mode. It
is also found without by SoapySDR Until after install the module

pavel-demin, 2019-06-10:
> Soapy for Red Pitaya does not work and therefore is not an option

I've recently tested the SoapyRedPitaya module with the latest PothosSDR version ([2019.03.24](https://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2019.03.24-vc14-x64.exe)). It works for both RX and TX without any problem.

> Exception thrown here: https://github.com/pothosware/SoapyRedPitaya/blob/3d576f83b3bde52104b2a88150516ca8c9a78c7a/SoapyRedPitaya.cpp#L522

I'd suggest that you check that 192.168.1.40 is the correct address of your Red Pitaya board and that SDR transceiver application is running on your Red Pitaya board.

pavel-demin, 2019-06-10:
BTW. I see that #345 is now closed and the fix is in 4.8.1. So, I suppose that SDRangel can now transmit using the `SoapyRedPitaya` module.

f4exb, 2019-06-10:
> ...and that SDR transceiver application is running 
Yes this is probably the issue. But I have no idea on how to run it moreover I cannot ssh to the RedPitaya which makes this "RedPity" adventure a total disaster. I think I will hide it somewhere deep in one of my drawers and forget about it.

pavel-demin, 2019-06-10:
> But I have no idea on how to run it moreover I cannot ssh to the RedPitaya which makes this "RedPity" adventure a total disaster.

Since the Red Pitaya board is designed as a universal tool with a network interface, it's a slightly more difficult to use than a single function USB device. It requires a working network configuration and it has a set of applications that should be started via the its web interface.

The network configuration and the web interface are described at

https://redpitaya.readthedocs.io/en/latest/quickStart/connect/connect.html

If the network problems persist, I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console as explained at

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

If the USB/serial console shows error messages during the boot, then most probably the problem is with the SD card and replacing the SD card often helps to solve this kind of problems.

Once you have access to the board's web interface, the SDR transceiver application can be installed via the application market place.

Otherwise, I'd suggest to try my SD card image that has all my applications already preinstalled:

http://pavel-demin.github.io/red-pitaya-notes/alpine/

pavel-demin, 2019-06-10:
BTW. What model of Red Pitaya board are you using? 10-bit, 14-bit or 16-bit?

f4exb, 2019-06-10:
I have the 16 bit model. I have seen the SD card image but the .zip does not look like an image that you flash to the SD using the `dd` command. It is the very collection of files and this does not make a bootable SD.

Regarding the SSH problem I found the solution but that's was bit tricky. It turns out that in the image I used which is the "official" one I suppose: https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html (there is only a beta available for the 16 bit version the stable link is broken) the keys in `/etc/ssh` are empty. It was obvious and I tried change the permissions before realizing that the file size was simply 0! All I needed to do is generate the keys logging via the USB console:
```
ssh-keygen -t rsa -f /etc/ssh/ssh_host_rsa_key
ssh-keygen -t dsa  -f /etc/ssh/ssh_host_dsa_key
```

pavel-demin, 2019-06-10:
> I have the 16 bit model.

The `SoapyRedPitaya` module only works with the 10-bit and 14-bit models. 

The right SD card image for the 16-bit board can be downloaded from

https://pavel-demin.github.io/stemlab-sdr-notes/alpine/

Currently, it's the only way to run my SDR applications on the 16-bit board.

> It is the very collection of files and this does not make a bootable SD.

Just copy the files from the .zip file to a FAT formatted SD card. No `dd` command is needed.

pavel-demin, 2019-06-10:
If you could add the HPSDR protocol to SDRAngel, then it'll allow to use SDRAngel with all the models of the Red Pitaya boards and other radios compatible with the HPSDR protocol such as Hermes, Hermes-Lite, etc.

BTW. The HPSDR protocol has a built-in auto discovery.

pavel-demin, 2019-06-10:
Here is an implementation of the HPSDR protocol:

https://github.com/g0orx/linhpsdr/blob/master/protocol1.c
https://github.com/g0orx/linhpsdr/blob/master/protocol1_discovery.c

f4exb, 2019-06-10:
> Just copy all the files and directories from the .zip file to a FAT formatted SD card. No dd command is needed.

It does not work and I still do not see how it would because you need some sort of special formatting in the MBR.

Else... I'll have a look at HPSDR as this could be an interesting addition.

pavel-demin, 2019-06-10:
> It does not work and I still do not see how it would because you need some sort of special formatting in the MBR.

The boot loader that comes with the ZYNQ chips is intelligent enough to read the FAT file system and find the required `boot.bin` file. You can find the commands that Xilinx recommends to run when preparing a micro SD card at the following link:

http://www.wiki.xilinx.com/Prepare+Boot+Medium

The following commands from these instructions create a FAT formatted partition and just copy `boot.bin` to this partition:
```
mkfs.vfat -F 32 -n boot /dev/sdX1

mkdir -p /mnt/boot
mount /dev/sdX1 /mnt/boot

cp boot.bin /mnt/boot/
```

So, copying `boot.bin` from a .zip file to a FAT formatted micro SD card is enough to make it bootable with the ZYNQ chips.

pavel-demin, 2019-06-10:
> I'll have a look at HPSDR as this could be an interesting addition.

For information. The HPSDR/Metis communication protocol is described in the following documents:

- [Metis - How it works](https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Metis/Documentation/Metis-%20How%20it%20works_V1.33.pdf)

- [HPSDR - USB Data Protocol](https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Documentation/USB_protocol_V1.58.doc)

f4exb, 2019-06-11:
I eventually got your software working on the RP preparing the SD card as mentioned in the Xilinx link (with a single FAT32 partition). I also verified the Hermes emulation was working with [openHPSDRJ](https://github.com/TAPR/OpenHPSDR-SVN/tree/master/N6LYT/openHPSDRJ) so I should be able to move on thanks!

satfan52, 2019-06-11:
Hey Edward 😊😂

Life is beatifull!

You do not need to ssh to the rp to start the sdr transceiver, things are
not as complicated! The RP has a web interface, all you have to do is type
the IP address of the RP in browser and from there you just click on the
"SDR transceiver". Instead of the IP address can also use the identifier
provided on the label attached to the ethernet port.

rp://something see your label

You need to be on the same subnet though otherwise it will not work (90
percent confidence)

If you really wanna ssh, just ssh the IP address of the RP and from there
you can navigate in the apps folder, there are start and stop script you
can launch. If you take the start script of the sdr transceiver and put it
at the root of the SD card then the sdr transceiver will automatically
start as soon as the rp is booted, turned on, so no need for all the
webstuff to launch the SDR transceiver! You just boot the RP and the SDR
transceiver is automatically launched!!!

satfan52, 2019-06-11:
The ssh password is "changeme"

pavel-demin, 2019-06-11:
> from there you just click on the "SDR transceiver"

The problem is that the Edouard's board is the latest 122.88-16 model. There is no "SDR transceiver" for this model.

satfan52, 2019-06-11:
that is really sad! :-(

Edward, I can borrow you my second 125-14 model so you can work on  it!  we can meet in Fredrieschaffen in two weeks if you want

satfan52, 2019-06-11:
of if you are in a hurry, i can ship it to you today.

satfan52, 2019-06-11:
I meant you can borrow me

f4exb, 2019-06-11:
For the native solution I'd rather go for trying to work with the HPSDR protocol that can work not only for the RP. 

For the Soapy thing I am going to open another ticket because in fact this is the more generic issue of giving the possibility to the user to specify "kwargs" at open time and this would cover RP and other possible cases. The audio parameters work a bit like that saving the audio parameters in the global presets so there are already some ideas. I do not necessarily need the hardware for such a simple thing (in theory). In fact the RP is listed by Soapy but without keyword parameters and this seems to be the issue since you would like to have `addr=xxx.xxx.xxx.xxx`. At open time it just says "I want to open the first device with `redpitaya` driver" and it does not appear to be enough.

f4exb, 2019-06-15:
Considering how the Metis frame is structured and even if not all FPGA images are truly MIMO I think it will be supported as a MIMO device and therefore in the future v.5

f4exb, 2019-09-23:
MIMO support (v5) is abandoned

satfan52, 2019-09-24:
Hello Edouart,

1) So no more native support for the 125-16 red-pitaya board then ? 
2) No  enhancements either to work around the famous detection issue with the redpitaya 122-14  board in the soapysdr environment  ? 
 
Thanks
Regards
Peter
Regards
Peter

satfan52, 2019-09-24:
Ok I just saw the soapysdr "qwark" enhancement went through (point 2 in my previous comment) went through. Many thanks, I will test very soon.

Regarding point 1, does that mean the new 125-16 board will only be supported via soapysdr (no native support like for the PlutoSDR for instance)

Many Thanks

f4exb, 2019-09-24:
OK, I was in a storm last night because BladeRF2 support as a MIMO device still does not work. The handling of FIFOs has to be redesigned from the ground up. I will reopen this issue without the "v5" label but do not expect anything soon

satfan52, 2019-09-24:
Dear Edouart,
It is a very good news for all redpitaya users that you are still open on the long run to attempt to provide native support for the  125-16 board. On the long run I would like to use SdrAngel for interfacing to both my plutosdr and my redpitaya and to use it as well in combination with wsjtx and similar programs. Unless I am mistaken I don't think any program can do that yet under Linux (maybe Windows but switching from Linux to Windows is not an option for me) 
Thank you again
Regards
Peter

f4exb, 2019-09-28:
I don't know exactly what you want to do. It is already possible to run more than one device in the same SDRangel instance. It is also possible to use WSJTX by piping the audio output of a SSB modulator into the audio input of WSJTX using a pulseaudio virtual audio device. The purpose of this ticket is to create a plugin that gets its I/Q samples from a hermes/metis connection (thus supporting Red Pitaya in the Metis mode and the like) and to do it respecting the synchronization between streams (what I called MIMO mode).

satfan52, 2019-09-28:
Yes, yesterday I managed to use SdraAngel with two different devices, my RP
125-14 board ("SDR transceiver" via "SoapySDR") and with my PlutoSDR
(native suppoart not SoapySDR). That worked well,  the only trouble is that
I was only able to use one of two receivers on the RP; in other words I
could monitor only one HF band at the time. Two receivers and two
transmitters would be available according to the doc of the "SDR
tranceiver" app :-(

What I am hoping is simply that this ticket will allow me to use the "hpsdr
transceiver" instead of the "SDR one" and the of course, I am also
logically hoping to be able to monitor several HF bands at the same
concurrently (as many as the hpsdr transceiver app allows it) instead of 1
nowadays with the "SDR transceiver/SoapySDR/SoapySDRredpitaya" trio.

What does not work yet (I think ?) is to be able to use multiple the 2
receivers  available on the  but I could only use 1 receiver

f4exb, 2019-10-02:
MIMO POC is definitely a fail. It is not possible to handle simultaneous threads in SDRangel. If Red Pitaya is to be supported natively it has to be with classical Rx and Tx streams (device sets).

satfan52, 2019-10-02:
Well, thanks for the clarification. I am sorry if I don't use the right
terminology to discuss about this. Indeed, I naievely thought that if the
redpitaya supports N receivers and 1 transmitter, then N different source
devices would appear in the dropdown source menu and one sink device would
appear in the dropdown sink menu.

devil1468h, 2020-06-13:
@satfan52 Hi, I am new to Sdrangel. I can't see the redpitaya in the sampling device. I am using sudo snap install sdrangel (ubuntu 16.04). I install the Soapysdr redpitaya as well. I also hardcode the the static IP address of my redpitaya in the .cpp file. Could you tell me how I can find redpitaya in the sampling device list.

satfan52, 2020-06-13:
Hello,
Well, Edouart is really the expert not me, hower are my two cents
(1) The method you describe is outdated and actually has never worked correctly. It only worked in reception and but not in transmission. 
(2) If you have installed SoapySDR and the SoapySDRredpitaya module the redpitaya should appear in your device list. if it does appear it is because your installation is not correct. Just run "SoapySdrUtil --find" from the console and  if your redpitaya is not detected it means there is an issue with your SoapySdr/RedpitayaSoapySDR installation. 
(3) When I did those tests a while ago I used the latest version of sdrangel (compiled from source not the one available in the Unbuntu repository) that includes a "menu" to enter the IP address and the Port Number (1001) of the redpitaya (so there is no need to modify the RedPitayaSoapySDR .cpp file to manually encode the IP address). Please refer to ticket 362 (https://github.com/f4exb/sdrangel/issues/362 ) that is closed 
(4) Kindly not that I do NOT routinely use sdrangel with my redpitaya using soapysdr, it used for experimental purpose only for making some tests in the soapysdr environment .  I am patiently waiting for this ticket (some kind of HPSDR compatibility supporting multiple receivers - up to 8) to be implemented so sdrangel supports the redpitaya WITHOUT the need to recourse to soapysdr.

I hope this helps

devil1468h, 2020-06-13:
Hi,

I really appreciate your reply. I just need redpitaya as the receiver for
my DATV test and I will consider your suggestion on HPSDR too. I failed to
build SDRangel from source and I will upgrade my Ubuntu to 18.04 to make
the new try.

Thanks again for your helps.

f4exb, 2020-06-14:
Thanks @satfan52 for the clarifications. Indeed the version you refer to in (3) is 4.10.1 and is ~9 months old. I don't think there has been a regression so the latest should also work. I strongly advise you to upgrade to 18.04 there is hardly any chance that you will be to compile with older releases. Note that on the other hand it should compile in 20.04 although this is not yet "official". 

You mention DATV. Please note that the DATV plugin is not fully operational and needs ~15 dB S/N or more to decode images. This makes it impractical for QO-100 for example.

Although this ticket is pretty old it is definitely on my list for this year. Being able to process Metis stream is key to open the way to more MIMO devices support. I am mainly thinking of a separate driver (TBD also) to support Kerberos SDR and possibly any generic array of clock synchronized sources and possibly sinks.

devil1468h, 2020-06-15:
@f4exb When I open  the sdrangel, soapysdr input and output show up in the log info. However, they dont show up in the sampling device list of sdrangel. Any possible reason?

2020-06-14 21:02:06.552 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fileinput
2020-06-14 21:02:06.552 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fcdpro
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fcdproplus
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.kiwisdrsource
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.localinput
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.remoteinput
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.soapysdrinput
2020-06-14 21:02:06.554 (D) DeviceSoapySDRScan::scan: Lib Version: v0.7.1-g5838bc91
2020-06-14 21:02:06.554 (D) DeviceSoapySDRScan::scan: API Version: v0.7.1
2020-06-14 21:02:06.554 (D) DeviceSoapySDRScan::scan: ABI Version: v0.7
2020-06-14 21:02:06.554 (D) DeviceSoapySDRScan::scan: Install root: /opt/install/SoapySDR
2020-06-14 21:02:06.554 (D) SoapySDROutputPlugin::enumOriginDevices: 0 SoapySDR devices
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.testsource
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.filesink
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.localoutput
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.remoteoutput
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.soapysdroutput
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateTxDevices: sdrangel.samplesink.testsink
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateMIMODevices: sdrangel.samplemimo.testmi
2020-06-14 21:02:06.554 (D) DeviceEnumerator::enumerateMIMODevices: sdrangel.samplemimo.testmosync
2020-06-14 21:02:06.554 (D) DeviceEnumerator::addNonDiscoverableDevices: start

f4exb, 2020-08-29:
Metis discovery example in Qt: https://github.com/alexlee188/ghpsdr3-alex/blob/master/trunk/src/HPSDRServer/discovery.cpp

f4exb, 2020-09-03:
Eventually I do not think this feasible due to weak documentation. It is not possible to guess how the stream is composed.

[11 more replies]

closed 2020-11-12 by f4exb