f4exb/sdrangel #347: Glitchy sound during self-test

Vort, 2019-05-19:
I was trying to perform simple self-test for this application.
And it managed to fail it somehow.

AM input was created by TestSource and demodulated by AM Demodulator.
Instead of clean tone resulting sound turned out to be full of hiccups.

During this test program was using ~4% of CPU resources, <1% when its window was minimized.
After adding 5-10% of load for visualizing of CPU usage, 10-15% of total available CPU resources was used during test.
Apparently, my PC have enough resources for this simple task.

Recordings of what was emitted by my earphones are here: [sdra.zip](https://github.com/f4exb/sdrangel/files/3195393/sdra.zip)
(.wav extension should be played without additional problems with Audacity)

Screenshot of SDRangel:
[image]

Version: 4.8.0 (Qt 5.12.1 64 bits)
OS: Windows 7 SP1 x64
CPU: Intel Core i5-4690
GPU: NVIDIA GeForce GTS 450

f4exb, 2019-05-20:
Is it "glitchy" in normal usage i.e. with a real source? no? so what is the issue? I am not going to spend time doing the ghost buster.

Vort, 2019-05-20:
Glitch also happens while using FileSource, especially when user clicks Play / Pause button: [sdra2.zip](https://github.com/f4exb/sdrangel/files/3196548/sdra2.zip)
[image]

f4exb, 2019-05-20:
I am afraid Windows is not up to the par when dealing with timers and events. The common thing between test source and file source is that both need a self pacing mechanism which is based on QTimer (which in turn is based on O/S driven timers...). Windows is nothing close to a real time O/S while Linux is. Even without the RT feature the 50 ms master timer is never far from more than a few ms. Each time I log in to my Windows machine I get frustrated and irritated. Basically it has the time to do anything but the thing you are requesting it to do and you just watch the paint drying in the meantime.  

I am going to check on the Linux version if there is anything software wise that would prevent the self pacing to work correctly but if this is not the case I will remove self pacing timer dependent plugins from the Windows deployment. These are:
  - Test input
  - File source
  - Remote input and output and related channel plugins

f4exb, 2019-05-20:
This is it:
  - in testsourcethread.cpp:
```c++
void TestSourceThread::tick()
{
    if (m_running)
    {
        qint64 throttlems = m_elapsedTimer.restart();

        if ((throttlems > 45) && (throttlems < 55) && (throttlems != m_throttlems))
        {
            QMutexLocker mutexLocker(&m_mutex);
            m_throttlems = throttlems;
            m_chunksize = 4 * ((m_samplerate * (m_throttlems+(m_throttleToggle ? 1 : 0))) / 1000);
            m_throttleToggle = !m_throttleToggle;
        }

        generate(m_chunksize);
    }
}
```
  - In filesourcethread.cpp:
```c++
void FileSourceThread::tick()
{
	if (m_running)
	{
        qint64 throttlems = m_elapsedTimer.restart();

        if (throttlems != m_throttlems)
        {
            m_throttlems = throttlems;
            m_chunksize = 2 * m_samplebytes * ((m_samplerate * (m_throttlems+(m_throttleToggle ? 1 : 0))) / 1000);
            m_throttleToggle = !m_throttleToggle;
            setBuffers(m_chunksize);
        }
...
```
In both cases the `tick()` method is called upon the regular beat of a timer at ~50ms. The actual elapsed time is measured by the `m_elapsed_timer` restart. Thus if there is some jitter we make sure more or less samples are taken the next time thus making sure the buffer does not become depleted not excessively large. In `testsourcethread.cpp` there is a supplementary check so that if the jitter exceeds +/-10% nothing is changed. I am pretty sure this works perfectly in Linux (maybe after a few seconds to balance the buffer) although I haven't checked recently but this code is fairly old.

If anyone has a better idea... else I will proceed as stated earlier.

Vort, 2019-05-20:
@f4exb, I see in Qt documentation, that several timer types exists: `PreciseTimer`, `CoarseTimer` and `VeryCoarseTimer`.
Default one is `CoarseTimer`. Maybe setting the type to `PreciseTimer` will help?

f4exb, 2019-05-20:
The Qt doc says `CoarseTimer` wake up can have 5% error, `VeryCoarseTimer` within 500ms and `PreciseTimer` down to the millisecond. It looks like we are more in the area of 500ms even with the `CoarseTimer`. so I doubt going for `PreciseTimer` would change anything. I will try to make up a small benchmark though but I am almost certain of the result since this is an issue I have encountered before.

f4exb, 2019-05-21:
On linux there is a difference between timer types indeed the 'precise' one is more... precise and is up to its specs within 1ms. I have put in place a histogram collection on the TestSource and after a while:
PreciseTimer:
```
2019-05-21 19:55:36.571 (D) TestSourceThread::tick: 49: 1028
2019-05-21 19:55:36.571 (D) TestSourceThread::tick: 50: 622
```
CoarseTimer (default) still pretty good:
```
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 45: 1
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 46: 1
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 47: 3
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 48: 7
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 49: 1135
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 50: 1096
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 51: 4
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 52: 2
2019-05-21 19:59:57.258 (D) TestSourceThread::tick: 55: 1
```

Vort, 2019-05-22:
Main problem with TestSource is fixed in 3f4c0a2e, thanks.
Play and Pause is still glitching, but that is the topic for another bugreport.

closed 2019-05-22 by Vort