f4exb/sdrangel #350: Windows zip file for latest release for sdrangel-4.8.0

zendogpc, 2019-05-21:
The latest release of sdrangel-4.8.0 for Windows doesn't contain a windows executable file? Somebody please advise.

f4exb, 2019-05-21:
How do you call `sdrangel.exe` ?

Vort, 2019-05-22:
It just have no icon. Look at #179.

f4exb, 2019-05-22:
Closed in favor od #179

closed 2019-05-22 by f4exb

zendogpc, 2019-05-22:
I have done a search through the extracted zip file and found no sdrangel.exe with or without an icon. I don't understand why someone has to be so against a operations system that aren't open source? I thought this would be a community to help people not put them down....

Vort, 2019-05-22:
@zendogpc looks like you have downloaded wrong file.
[This one](https://github.com/f4exb/sdrangel/archive/v4.8.0.zip) is source code and you need [this one](https://github.com/f4exb/sdrangel/releases/download/v4.8.0/sdrangel_v4.8.0.7z).
It can be extracted with 7-Zip or WinRAR.

zendogpc, 2019-05-23:
I appreciate the clarification and I have downloaded the correct file.
Thanks for the consideration.

Vort, 2019-05-23:
If file download location is correct, then it is probably some software on your PC is removing `sdrangel.exe` file.
Size of `sdrangel_v4.8.0.7z` should be `39137747` bytes, MD5 checksum `7a30f99d122f27f45355669a09ad6cf9`.
And here is the screenshot of executable file, as seen with 7-Zip 19.00 (x64):
[image]