f4exb/sdrangel #352: Bad quality AM demodulation for low SNR signals

Vort, 2019-05-27:
When signal level is low, AM Demodulator starts producing nasty noises.
Looks like some problems with filtering.
Here is the recording for the test (10 kHz RF BW): [michael.zip](https://github.com/f4exb/sdrangel/files/3223720/michael.zip)
SDRangel demodulation:
[image]
Dream demodulation (for comparison):
[image]

f4exb, 2019-05-27:
Thanks for the I/Q file this is quite helpful. Unfortunately it seems this is an artifact on the baseband signal and I am unable to identify its origin. With the channel analyzer you can see spikes in squared magnitude that are responsible of the audio clicks: 
[image]

f4exb, 2019-05-27:
Also because of the 16 bit only resolution and the low level of the signal the dynamic on the squared magnitude is very limited as shown in this detail of the squared magnitude trace. Also right before the peak it passes by zero. Possibly there is a value overload (quite unlikely though):
[image]

Vort, 2019-05-27:
Spectrum artifacts are here because I have "interpolated" 12 kHz signal to 48 kHz by just duplicating samples 4 times, since it looks like 12 kHz is not supported by AM Demodulator at all.
Maybe it is not correct to do it this way, but more accurate interpolation of signal, with Audacity, still produces noise after demodulation.

Vort, 2019-05-27:
> What hardware are you using btw?

I am trying to connect publicly available KiwiSDR receivers to SDRangel.
[Here is the code](https://github.com/Vort/KiwiAngel), it is not ready yet, but basic functions are there.

Vort, 2019-05-27:
Fresh example (with higher dynamic range):
China Radio, 9765 kHz, same 4x12kHz expansion: [china_radio.zip](https://github.com/f4exb/sdrangel/files/3224699/china_radio.zip)
And same spikes in spectrum:
[image]

f4exb, 2019-05-27:
>  since it looks like 12 kHz is not supported by AM Demodulator at all

Yes I think that's the issue. I thought you used a 48k sample rate. It is not that it is not working at all but there's a trick and this trick is to use a sound card with a sample rate lower than 12k. In Linux pulseaudio can do the job. You can create a virtual device or change the sample rate of the system device (ex: `alsa_output.pci-0000_00_1f_3.analog-stereo`) using the preferences>audio dialog and then select this device for the audio output of course (right click on the loudspeaker icon mute button).

In fact the demodulator can decimate down to the audio sample rate from the channel sample rate but not interpolate up to it. Note that I think this is implemented in AM demod but not working great. If you take another demod like SSB then it is not working at all.

f4exb, 2019-05-27:
I had a brief look at your code and spotted this:
```
		for (int i = 0; i < sampleCount; i++)
		{
			s.setReal(boost::endian::endian_reverse(messageSamples[i * 2]));
			s.setImag(boost::endian::endian_reverse(messageSamples[i * 2 + 1]));
			m_samplesBuf.push_back(s);
			m_samplesBuf.push_back(s);
			m_samplesBuf.push_back(s);
			m_samplesBuf.push_back(s);
		}
```
Is this your way to do interpolation? Then this is very wrong. First you need a low pass filter (usually a FIR) corresponding to the original sample rate (thus cutting above 12 kHz) then you insert a sample in the filter followed by 3 stuffing (complex) zeros (and not a copy of the sample) and after each insertion in the filter (thus at four times your original sample rate) you take the output of the filter.

For the thing to work properly with the rest of the software you should leave it at 12k and use the trick described above for audio. For FT8 reception I use a Perseus at 48k decimated by 4 so in the baseband the sample rate is 12k (thus exactly the same as yours without decimation). My SSB demod is piping audio in a virtual pulseaudio device at 8k which is at the input of WSJT-X. Works like a charm.

Vort, 2019-05-28:
I have made a separate report concerning interpolation: #354.
But this one is a different story, so it should stay open until more accurate analysis is performed.

f4exb, 2019-05-28:
I have tried your code with the suggested changes (12k sample rate) in SSB. It works fine and is fun!

Vort, 2019-05-29:
Since 12 kHz bandwidth is now supported, I have recovered original sample rate for the China Radio recording: [china_radio_12k.zip](https://github.com/f4exb/sdrangel/files/3230750/china_radio_12k.zip)
Result of demodulation is almost the same as before.

Vort, 2019-05-29:
Here is how this problem looks in the audio editor:
[image]

Vort, 2019-05-29:
Here is the quote from article _[Digital Envelope Detection: The Good, the Bad, and the Ugly](https://www.dsprelated.com/showarticle/938.php)_, which may help:
> this detector's advantage that no lowpass filtering is needed at the output of the square root operation. However, I have learned this only to be true for noise-free signals! In practical real-world applications the lowpass filter I've included in Figure 5 is necessary

So, most likely, one more lowpass filter is needed.
(I know that AM Demodulator have several operation modes, but basic one should work nevertheless)

f4exb, 2019-05-29:
You can switch on the bandpass filter at the right of "RF BW" this is a bandpass from 300 Hz to half the bandwidth that is set next (with the slider). Originally this is for communications on the air band. Incidentally it seems to reduce the high frequency spikes. This can be turned into a permanent lowpass if necessary.

I have read the article you mentioned and the author places the "figure 4" before the "figure 5" (that is used here) in the top ranking. In addition the "figure 4" seems more computationally efficient (no sqrt). Basically it uses a first order distance function instead of a second order. However the sum of squares is already available for signal power estimation.

Vort, 2019-05-29:
> Unfortunately it seems this is an artifact on the baseband signal

After closer look at I/Q channels, I see some sort of misalignment between them.
And Dream also produces glitches:
[image]

Vort, 2019-05-29:
So the problem most likely comes from hardware.
It can be fixed (actually, only partially) by additional lowpass filter.
But question is how often that happens with different hardware.
Which may lead to a conclusion if demodulator should take this corruption into account.

f4exb, 2019-05-29:
Looks like an imbalance rather than a skew between I and Q which would be surprising on any type of hardware that does synchronous I/Q sampling. There is a IQ imbalance correction block (you can see almost any other "hardware" type of receiver plugin like RTL-SDR). You may want to add it to the Kiwi interface also. However I doubt this has an impact on AM modulated signals since you base the demodulation on either the sum of squares or the sum of absolute values.

f4exb, 2019-05-29:
Low pass filter after demod implemented on dev. The bandpass switch simply switches between lowpass and bandpass.

Vort, 2019-05-29:
In [this](https://github.com/f4exb/sdrangel/commit/d2c31059dd4814cfc8365c76499e21fdeaf723fe#diff-a8892110cebfea306a074f0faca28b2fR464) line should be `settings` instead of `m_settings`.

Vort, 2019-05-29:
It is better to know what is the source of the glitch.
But filter fixes it anyway.
This solution can be refined in future.

closed 2019-05-29 by Vort

f4exb, 2019-05-29:
Copy and paste error fixed (dev) thanks for spotting.

f4exb, 2019-06-07:
Hello,
for the KiwiSDR plugin you may have trouble with the REST API interface. This is a gray area because the code generator from the .yaml files is a modified version that I haven't put online. Moreover it is easier if you are familiar with it. So don't worry I can do that part. This should not prevent you from requesting a PR.
Brgds.

Vort, 2019-06-07:
@f4exb it is ok if you want to support this plugin.
I don't know how to test remote control parts and more important is that I'm not sure why it is needed to remotely control remote controlled receiver.
So if you add missing parts - that will be good. If not - plugin will be still usable.
Another question - is what to write to GPL header. I will add my nickname to some of them, other ones are minimally modified from TestSource.
And since I do not own any KiwiSDR receiver, I may miss some important functionalty. But that can be addressed later.

Vort, 2019-06-07:
#361 is ready.

f4exb, 2019-06-07:
Thanks a lot! I already had hours of fun with the prototype. I suppose other people will... 

For the GPL header I will put `Vort` as the author to respect your privacy but this can be your real name if you want. Usually the version is the SDRangel version corresponding to the latest changes in the plugin so this will be 4.10.0 as I have set as an untold rule that a new plugin would automatically trigger a minor digit  bump.

Vort, 2019-06-07:
Nickname is there because of privacy, right.
Version change is ok, I agree with this approach.