f4exb/sdrangel #353: Faults detected version 4.8

Pedro9012, 2019-05-27:
Hackrf  firw. 2017+portapack , Windows 10.
Hi, I commented some faults detected in version 4.8 in case something can be done to solve them: 
only admits to transmit once by the play / stop button, to be able to transmit from new you have to go to RX play / stop and go back to TX .
 
the last device set program hangs and closes completely.
Otherwise it's luxury, good work.

   Hola comento unos fallos detectados en la version 4.8 por si se puede hacer algo para solucionarlos : solo admite transmitir una vez mediante el boton play/stop , para poder transmitir de nuebo hay que ir a RX play/stop y volver a TX .   
al remover los device set el programa se cuelga y se cierra.
Por lo demas va de lujo, buen trabajo.

f4exb, 2019-05-28:
HackRF is a half duplex device so you cannot run the Rx and the Tx at the same time. So you have to stop one and start the other. This works as designed. I have not notice any issue with this cycle.

Pedro9012, 2019-05-28:
hello, in version 4.3.2 it allows to transmit (play / stop) and without returning to the RX screen to transmit again (play / stop) as many times as you want, so if you are listening with another device (rtl-sdr combencional) you can maintain a fluid Qso. as of that version it is no longer possible, to transmit more than once you have to force through the RX screen to press (play / stop) and return to the tx screen every time. I hope you have explained me properly so that you understand the difference.
In the other version it also does not hang when you remove devices.
Receive a cordial greeting and many thanks for your time,

hola, en la version 4.3.2  permite transmitir (play/stop) y sin volver a la pantalla de RX volver a transmitir (play/stop) las veces que se quiera , por lo que si estas escuchando con otro dispositivo (rtl-sdr combencional) puedes mantener un Qso fluido .  a partir de dicha version ya no se puede , para transmitir mas de una vez hay que pasar a la fuerza por la pantalla de RX pulsar (play/stop) y volver a la de tx cada vez .  Espero haberme explicado adecuadamente para que entiendas la diferencia.
En las otra version tampoco se cuelga cuando remueves dispositivos .
Recive un cordial saludo y muchisimas gracias por tu tiempo ,

f4exb, 2019-05-28:
Yes this is the new functionality. You have to use it in combination with `ppt_active.py` (right now in `scriptsapi` directory in dev only) running in another terminal (it is a python script) to achieve PTT functionality. I have removed the Rx/Tx internal synchronization for HackRF as it was interfering with the new PTT program. Note that now you can also combine RTL-SDR Rx and HackRF Tx as the script is device agnostic. It was not possible before.

Can you please detail "it also does not hang when you remove devices"? What "devices" are you talking about?

Pedro9012, 2019-05-29:
Hello, thank you very much for your answer.
The hang of the program occurs: example RX screen with DSD, AM, plugin etc, we open add skin device set, select harckrf, open any chanel plugin (AM, NFM, etc ...), everything is still normal RX / TX, but When the last device set is removed, Sdrangel is closed and closed.
Resolution screen 1920 x 1080
If you do not understand it, comment and I'll try to communicate in some way even more detailed.

A greeting.

Hola, muchas gracias por tu respuesta.
El cuelgue del programa se produce : ejemplo   pantalla RX con DSD ,AM,etc plugin , abrimos add skin device set, seleccionamos harckrf , abrimos cualquier chanel plugin  (AM,NFM,etc...) , todo sigue normal RX / TX , pero al remove last device set se cuelga y se cierra Sdrangel completo .
Resolucion pantalla 1920 x 1080
Si no lo entiendes comenta y te lo intento comunicar de alguna forma aun mas detallada.

Un saludo.

Pedro9012, 2019-05-29:
Hi, I am testing V.8.1 and it is much worse than the previous one, it gets stuck and you have to restart the harckrf from the reset button simply by making a change of presets.
It has been spectacular to start the loading screen of the program.
Windows 10 VGA 1920 X 1080, HARCKRF + PORTAPACK, FIRMWARE 2017.
At the moment I still use V4.3.2 that does not reproduce these failures.
A greeting.

Hola , estoy probando la V.8.1  y me va mucho peor que la anterior , esta se queda pillada y hay que reiniciar el harckrf desde el boton de reset simplemente con hacer un cambio de presets.
Ha quedado espectacular al arrancar la pantalla de carga del programa.
Windows 10  VGA 1920 X 1080 , HARCKRF + PORTAPACK , FIRMWARE 2017.
De momento sigo usando V4.3.2 que no reproduce estos fallos.
Un saludo.

f4exb, 2019-05-29:
Please detail your scenario step by step because for now it is vague and cannot be reproduced.

Pedro9012, 2019-05-30:
Hello, can I send a video to some email address?

Hola , ¿ le puedo enviar un video a alguna direccion de email ?

f4exb, 2019-05-31:
Should be fine un the gmail one

Pedro9012, 2019-05-31:
Hola , despues de 3 horas intentando grabar video menor de 10 mgs no lo consigo , lo seguire intentando .
Un saludo.

Pedro9012, 2019-05-31:
Hello, after 3 hours trying to record video less than 10 mg I can not get it, we keep trying.
A greeting.

keyboarderror, 2019-06-04:
I could confirm.  The HackRF can transmit the first time your press the start/top button but will generate garbage on the second start/stop button press. Cycling the hackrf receive start/stop will allow transmit start/stop to work once again for one press. This is not intended functionality. The Linux version did not have this problem. But now in 4.9 Windows transmit won't work at all, And Linux won't even start. So...

f4exb, 2019-06-04:
So... you have to check your installation (Linux at least) because this is working perfectly. Will check later on Windows.

keyboarderror, 2019-06-04:
I have been able to reproduce the Windows transmit problem in 4.9 on another computer. Here's a video.

https://youtu.be/3JffDHOJ6EY

I also have a video of the Linux core dump at start that may be related to issue #359.

f4exb, 2019-06-04:
The core dump occurs when leaving the program or closing the Tx tab (remove last device). 

For the rest I think you do not understand what half duplex is. You cannot transmit and receive at the same time period. And If you attempt to do this unpredictable results can occur.

keyboarderror, 2019-06-04:
I think you didn't understand my video. The HackRF IS running half duplex. The receive and transmit cycles are started, stopped, and fully separated. The transmitted 1Khz signal tone you should be hearing is picked up by an RTLSDR in SDRConsole. The first transmit is clear, the second and all others are distorted until reset. The HackRF's receive audio is irrelevant for this demonstration.It works fine. It only serves to reset the interface. In Linux this is not needed. You can stop and start the transmit sink repeatedly and the 1KHz tone will be clear. Other modulated sound also distorts. It is not an audio problem. It is a problem in how the HackRF interface is being handled in Windows.

I will get make a separate report of the core dump I experience in Linux.

f4exb, 2019-06-05:
Confirmed the issue exists and in Windows only. There is no difference in the code. On Windows the `hackrf.dll` from Pothos is used. Maybe this causes the issue. In 4.3.0 a compiled version was used.

f4exb, 2019-06-05:
Unfortunately the problem persists even if the HackRF library is compiled from source. Therefore the HackRF output plugin will be discontinued in the next binary release for Windows.

Pedro9012, 2019-06-10:
Hola, Con la version 4.9.1 solocionado el problema de cierre de programa al eliminar device set.
Muchisimas GRACIAS-

Hi, with version 4.9.1 only the problem of closing the program when removing device set.
Many thanks-

Pedro9012, 2019-06-13:
V4.10.0    funciona de maravilla  .
Un saludo

V4.10.0 works great.
a greeting

keyboarderror, 2019-06-13:
Tested HackRF transmit works properly in windows as recently as 4.4.5

Pedro9012, 2019-06-14:
No , solo una vez y hay que volver a RX   play/stop ,  TX  una vez   volver RX  play/stop  etc...

f4exb, 2019-06-28:
It seems to work now with the new build v4.10.2

keyboarderror, 2019-06-28:
Working here too. Very much appreciated, Edouard.

f4exb, 2019-06-29:
Good! :smile:

closed 2019-06-29 by f4exb