f4exb/sdrangel #354: Interpolation in AM Demodulator is completely broken

Vort, 2019-05-28:
The goal is to demodulate 12 kHz BW file of AM transmission: [la_bamba.zip](https://github.com/f4exb/sdrangel/files/3227576/la_bamba.zip)
Right now result looks like this:
[image]
It is not needed even to hear that thing to know that it is broken (your ears will not like this).
I am not sure how it should be fixed (and how namy times this code is duplicated), but such quick and dirty hackfix allowed me to hear what looks like correct demodulation:
```patch
diff --git a/plugins/channelrx/demodam/amdemod.cpp b/plugins/channelrx/demodam/amdemod.cpp
index 99eeb32..448e996 100644
--- a/plugins/channelrx/demodam/amdemod.cpp
+++ b/plugins/channelrx/demodam/amdemod.cpp
@@ -130,14 +130,11 @@ void AMDemod::feed(const SampleVector::const_iterator& begin, const SampleVector
 
 		if (m_interpolatorDistance < 1.0f) // interpolate
 		{
-            processOneSample(ci);
-
-		    while (m_interpolator.interpolate(&m_interpolatorDistanceRemain, c, &ci))
+		    while (!m_interpolator.interpolate(&m_interpolatorDistanceRemain, c, &ci))
             {
                 processOneSample(ci);
+				m_interpolatorDistanceRemain += m_interpolatorDistance;
             }
-
-            m_interpolatorDistanceRemain += m_interpolatorDistance;
 		}
 		else // decimate
 		{ 
```
[image]
Please investigate this problem.

f4exb, 2019-05-28:
Yes it works thanks! 

I didn't care too much about it because pulseaudio can do it possibly more efficiently. But I agree this can be convenient and if it is there then it should work. It could be extended to other audio demods at least SSB and FM which are the most likely to be used with sample rates below the common 48k audio.

f4exb, 2019-05-28:
Done on SSB and NFB (dev branch) also.

Vort, 2019-05-29:
Interpolated waveform looks smooth, so it counts as fixed.

closed 2019-05-29 by Vort