f4exb/sdrangel #360: Illegal instruction (core dumped) in 4.9

keyboarderror, 2019-06-05:
I've started getting this in multiple Ubuntu 18.04 systems upgraded from 4.8 that previously worked. A problem with the Limesdr plugin? I'm only using the .deb version currently.

Terminal output:

$ export LD_LIBRARY_PATH=/usr/lib/sdrangel
$ sdrangel

2019-06-05 07:01:22.842 (I) SDRangel 4.9.0-6-g7a6f1b79c Qt 5.9.5 64b x86_64 Ubuntu 18.04.2 LTS DSP Rx:24b Tx:16b PID 9226
2019-06-05 07:01:22.842 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/jayson/.config/f4exb/SDRangel.conf
2019-06-05 07:01:28.054 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2019-06-05 07:01:28.054 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2019-06-05 07:01:28.055 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2019-06-05 07:01:28.056 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2019-06-05 07:01:28.097 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
2019-06-05 07:01:28.098 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsd.so
2019-06-05 07:01:28.099 (I) PluginManager::loadPluginsDir: loaded plugin libdemodfreedv.so
2019-06-05 07:01:28.100 (I) PluginManager::loadPluginsDir: loaded plugin libdemodlora.so
2019-06-05 07:01:28.101 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2019-06-05 07:01:28.101 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2019-06-05 07:01:28.102 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2019-06-05 07:01:28.102 (I) PluginManager::loadPluginsDir: loaded plugin libfreqtracker.so
2019-06-05 07:01:28.103 (I) PluginManager::loadPluginsDir: loaded plugin libinputairspy.so
2019-06-05 07:01:28.104 (I) PluginManager::loadPluginsDir: loaded plugin libinputairspyhf.so
2019-06-05 07:01:28.104 (I) PluginManager::loadPluginsDir: loaded plugin libinputbladerf1.so
2019-06-05 07:01:28.105 (I) PluginManager::loadPluginsDir: loaded plugin libinputbladerf2.so
2019-06-05 07:01:28.106 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdpro.so
2019-06-05 07:01:28.106 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdproplus.so
2019-06-05 07:01:28.107 (I) PluginManager::loadPluginsDir: loaded plugin libinputfilesource.so
2019-06-05 07:01:28.108 (I) PluginManager::loadPluginsDir: loaded plugin libinputhackrf.so
Illegal instruction (core dumped)

Syslog:

Jun  5 07:00:44 Computer systemd[4891]: Starting GNOME Terminal Server...
Jun  5 07:00:44 Computer dbus-daemon[4957]: [session uid=1000 pid=4957] Successfully activated service 'org.gnome.Terminal'
Jun  5 07:00:44 Computer systemd[4891]: Started GNOME Terminal Server.
Jun  5 07:01:03 Computer hud-service[5634]: #033[31mvoid DBusMenuImporter::slotGetLayoutFinished(QDBusPendingCallWatcher*)#033[0m: "No such interface 'com.canonical.dbusmenu' on object at path /org/ayatana/bamf/win$
Jun  5 07:01:03 Computer kernel: [ 2404.256520] traps: sdrangel[9189] trap invalid opcode ip:7ff9214e1741 sp:7ffddef19ed0 error:0 in libLimeSuite.so.19.01.0[7ff9214bb000+bb000]
Jun  5 07:01:04 Computer systemd[4891]: Starting Notification regarding a crash report...
Jun  5 07:01:04 Computer update-notifier-crash[9201]: sdrangel
Jun  5 07:01:04 Computer system-crash-no[9210]: GtkDialog mapped without a transient parent. This is discouraged.
Jun  5 07:01:11 Computer unity-panel-ser[5433]: Group Properties error: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
Jun  5 07:01:11 Computer unity-panel-ser[5433]: message repeated 5 times: [ Group Properties error: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying]
Jun  5 07:01:11 Computer unity-panel-ser[5433]: menus_destroyed: assertion 'IS_WINDOW_MENU(wm)' failed
Jun  5 07:01:11 Computer hud-service[5634]: QCoreApplication::postEvent: Unexpected null receiver
Jun  5 07:01:12 Computer zeitgeist-daemon[5664]: #033[31m[14:01:12.732032 WARNING]#033[0m zeitgeist-daemon.vala:127: Unable to parse version info!
Jun  5 07:01:13 Computer systemd[4891]: update-notifier-crash.service: Main process exited, code=exited, status=1/FAILURE
Jun  5 07:01:13 Computer systemd[4891]: update-notifier-crash.service: Failed with result 'exit-code'.
Jun  5 07:01:13 Computer systemd[4891]: Failed to start Notification regarding a crash report.
Jun  5 07:01:28 Computer hud-service[5634]: #033[31mvoid DBusMenuImporter::slotGetLayoutFinished(QDBusPendingCallWatcher*)#033[0m: "No such interface 'com.canonical.dbusmenu' on object at path /org/ayatana/bamf/win$
Jun  5 07:01:28 Computer unity-panel-ser[5433]: Already have a menu for window ID 71303183 with path /MenuBar/1 from :1.119, unregistering that one
Jun  5 07:01:28 Computer unity-panel-ser[5433]: gtk_widget_remove_accelerator: assertion 'GTK_IS_WIDGET (widget)' failed
Jun  5 07:01:28 Computer kernel: [ 2428.409437] traps: sdrangel[9226] trap invalid opcode ip:7fb6bf451741 sp:7ffdd05b2f60 error:0 in libLimeSuite.so.19.01.0[7fb6bf42b000+bb000]
Jun  5 07:01:28 Computer unity-panel-ser[5433]: gtk_widget_remove_accelerator: assertion 'GTK_IS_WIDGET (widget)' failed
Jun  5 07:01:28 Computer systemd[4891]: Starting Notification regarding a crash report...
Jun  5 07:01:28 Computer unity-panel-ser[5433]: Group Properties error: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
Jun  5 07:01:28 Computer unity-panel-ser[5433]: message repeated 5 times: [ Group Properties error: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying]
Jun  5 07:01:28 Computer unity-panel-ser[5433]: menus_destroyed: assertion 'IS_WINDOW_MENU(wm)' failed
Jun  5 07:01:28 Computer hud-service[5634]: QCoreApplication::postEvent: Unexpected null receiver
Jun  5 07:01:28 Computer update-notifier-crash[9245]: sdrangel
Jun  5 07:01:31 Computer hud-service[5634]: #033[31mvoid DBusMenuImporter::slotGetLayoutFinished(QDBusPendingCallWatcher*)#033[0m: "No such interface 'com.canonical.dbusmenu' on object at path /org/ayatana/bamf/win$
Jun  5 07:03:02 Computer update-notifier[6190]: GtkDialog mapped without a transient parent. This is discouraged.
Jun  5 07:03:02 Computer hud-service[5634]: #033[31mvoid DBusMenuImporter::slotGetLayoutFinished(QDBusPendingCallWatcher*)#033[0m: "No such interface 'com.canonical.dbusmenu' on object at path /org/ayatana/bamf/win$
Jun  5 07:03:49 Computer bamfdaemon[5171]: bamf_matcher_set_starting_desktop_file: assertion 'desktop_file != NULL' failed

Thanks,
Jayson

f4exb, 2019-06-05:
`Illegal instruction (core dumped)` seems to designate a CPU optimization in the compiled code that is not supported by your hardware. What is the output of `lscpu` in your system?

keyboarderror, 2019-06-06:
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              8
On-line CPU(s) list: 0-7
Thread(s) per core:  2
Core(s) per socket:  4
Socket(s):           1
NUMA node(s):        1
Vendor ID:           GenuineIntel
CPU family:          6
Model:               26
Model name:          Intel(R) Core(TM) i7 CPU         950  @ 3.07GHz
Stepping:            5
CPU MHz:             1631.804
CPU max MHz:         3060.0000
CPU min MHz:         1596.0000
BogoMIPS:            6102.63
Virtualization:      VT-x
L1d cache:           32K
L1i cache:           32K
L2 cache:            256K
L3 cache:            8192K
NUMA node0 CPU(s):   0-7
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 popcnt lahf_lm pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid dtherm ida flush_l1d

f4exb, 2019-06-06:
Well it does have ssse3`, `sse4_1` and even `sse4_2` but no `avx`, `avx2`. Normally the cpu flags should be limited to `sse4_1` but apparently somewhere it does not work. At least this gives some insight on where to look at.

f4exb, 2019-06-06:
Missed this one:
```
Jun 5 07:01:03 Computer kernel: [ 2404.256520] traps: sdrangel[9189] trap invalid opcode ip:7ff9214e1741 sp:7ffddef19ed0 error:0 in libLimeSuite.so.19.01.0[7ff9214bb000+bb000]
```
So yes this further narrows down to LimeSuite "external" compilaton in the new build system.

mnhauke, 2019-06-06:
I had similar issues with consumers of libLimeSuite in openSUSE and asked the LimeSDR guys if they could add a runtime detection for their SIMD optimizations to have a proper solution for this problem...
https://github.com/myriadrf/LimeSuite/issues/234

Since this request wasn't accepted by upstream you still have to compile libLimeSuite with " -DENABLE_SIMD_FLAGS="SSE3" to workaround those compatibility issues.
https://build.opensuse.org/package/view_file/hardware:sdr/limesuite/limesuite.spec?expand=1

f4exb, 2019-06-06:
Thanks for the tip I was wondering which option to use. In fact auto-detection in LimeSuite does not serve it right because it is going to detect the CPU features of the build server just like it does today so `-DENABLE_SIMD_FLAGS="SSE3` is actually what we are looking for. 

BTW this is how things are working with the new build system in SDRangel but `ra1nb0w` was smart enough to add a way to restrict the CPU flags to a specific level i.e.  `-DFORCE_SSSE3=ON` or `-DFORCE_SSE41=ON`. In 4.9.0 I have used the SSE41 flag but I may fallback to SSSE3 in the future to be even more generic.

f4exb, 2019-06-06:
Note: one can use this method to detect AVX (of all kinds) instructions: https://stackoverflow.com/questions/47878352/how-to-check-if-compiled-code-uses-sse-and-avx-instructions You need to use `gawk` in place of `awk` due to the huge size of the regexp.

f4exb, 2019-06-06:
Apparently `libcm256cc.so` also has AVX. This should be easy to fix since cm256cc is in my realm.

closed 2019-06-08 by keyboarderror

keyboarderror, 2019-06-08:
Appreciated, Edouard. All working normally again.