f4exb/sdrangel #362: SoapySDR: add possibility to specify a list of "kwargs" for a specific device

f4exb, 2019-06-11:
Following #344 this ticket is open to cover the more general issue of giving the user the possibility to specify the list of "kwargs" to be  used when opening a SoapySDR listed device. In some cases (like the RedPitaya) this is mandatory.

This can be inspired by what is in place for the audio parameters. The system gives a list of devices at start time then some parameters may be adjusted and saved per audio device identifier. When the program starts again if a device is effectively there it will used the corresponding saved parameters.

With Soapy this is similar. At start time the SoapySDR plugin lists the SoapySDR devices attached to the system. It is not always able to guess the appropriate parameters to open the device correctly. So similarly to audio the correct parameters can be given by the user and attached to the identification of the device which in the worst case is the driver name and sequence. 

The same system of synchronizing with existing devices at a given time can be implemented to do some cleanup if some devices are not routinely attached to the system anymore.

f4exb, 2019-06-12:
After some trial this has to be even more generic because you cannot afford adherence to SoapySDR in the main program. Therefore I will turn this into adding user string argument for a given hardware (hardware id) and sequence number. Then a dialog to configure like this can be open from the main window.

This user argument string related to hardware can be used by SoapySDR plugins when opening the corresponding device. Other plugins may use or not use it and it can be filled (size > 0) or not so this makes it quite flexible.

f4exb, 2019-06-15:
Now implemented in 4.10.1 

  - From Main window do Preferences -> Devices -> User arguments
  - A dialog opens. In the top list the scanned devices appear. Locate the SoapySDR line with the redpitaya driver and highlight the line
  - Push the down arrow button below the list. This creates a line below with the SoapySDR for redpitaya reference
  - Select this line and type in the 'addr=x.x.x.x` parameter
  - Click OK

Now when the corresponding redpitaya device is opened from the SoapySDR plugin this `addr` parameter will be used to open the Red Pitaya.

closed 2019-06-15 by f4exb

satfan52, 2019-09-25:
Hello Edouart,

I did the usual  "git Pull", "make uninstall", "make clean", "cmake .", "make" and "make install" sequence in my Linux Ubuntu 18.10 environment, but with my updated version of SdrAngel I do not see the "device" menu item in the preference menu. 

"SdrAngell --version" indicates version 4.8.0 

The redpitaya appears OK as a soapysdr device in the list of plugins (both source and sink), but launching the source (receiver) crashes SdrAngel

2019-09-25 08:00:19.384 (D) SoapySDRInput::start: (re)sart buddy thread
2019-09-25 08:00:19.384 (D) SoapySDRInput::start: started
2019-09-25 08:00:19.384 (D) DSPDeviceSourceEngine::gotoRunning: starting  SpectrumVis
2019-09-25 08:00:19.384 (D) DSPDeviceSourceEngine::gotoRunning: starting  FileRecord
2019-09-25 08:00:19.384 (D) SoapySDRInputThread::run: format: CF32 fullScale: 1.000000
2019-09-25 08:00:19.384 (D) DSPDeviceSourceEngine::gotoRunning: starting ThreadedSampleSink( DownChannelizer(SSBDemod) )
2019-09-25 08:00:19.385 (D) ThreadedBasebandSampleSink::start
2019-09-25 08:00:19.385 (D) DownChannelizer::start: thread:  0x55e543658400  m_inputSampleRate:  1250000  m_requestedOutputSampleRate:  0  m_requestedCenterFrequency:  0
2019-09-25 08:00:19.385 (D) SSBDemod::applyChannelSettings:  inputSampleRate:  1250000  inputFrequencyOffset:  0
2019-09-25 08:00:19.385 (D) NCOF::setFreq: freq: 0.000000 m_phaseIncrement: 0.000000
2019-09-25 08:00:19.385 (D) Interpolator::createPolyphaseLowPass: ntaps: 32
2019-09-25 08:00:19.385 (D) DSPDeviceSourceEngine::gotoRunning:input message queue pending:  0
2019-09-25 08:00:19.385 (D) DownChannelizer::applyConfiguration in= 1250000 , req= 48000 , out= 78125 , fc= 0
2019-09-25 08:00:19.385 (D) SSBDemod::handleMessage: MsgChannelizerNotification: m_sampleRate
2019-09-25 08:00:19.386 (D) SSBDemod::applyChannelSettings:  inputSampleRate:  1250000  inputFrequencyOffset:  0
2019-09-25 08:00:19.386 (D) SSBDemod::handleMessage: MsgChannelizerNotification: m_sampleRate
2019-09-25 08:00:19.386 (D) SSBDemod::applyChannelSettings:  inputSampleRate:  78125  inputFrequencyOffset:  0
2019-09-25 08:00:19.386 (D) NCOF::setFreq: freq: 0.000000 m_phaseIncrement: 0.000000
2019-09-25 08:00:19.386 (D) Interpolator::createPolyphaseLowPass: ntaps: 32
2019-09-25 08:00:19.386 (D) SoapySDRInputGui::handleInputMessages: message: DSPSignalNotification
2019-09-25 08:00:19.386 (D) SoapySDRInputGui::handleInputMessages: DSPSignalNotification: SampleRate:1250000, CenterFrequency:50000000
2019-09-25 08:00:19.386 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
terminate called after throwing an instance of 'std::runtime_error'
  what():  sendCommand failed: 0
Aborted (core dumped)

Regards
Peter
 
[image]

satfan52, 2019-09-25:
I guess if it is implemented in 4.10.1 it makes sense it does not work in 4.8.0, but I am not clean how to select the latest version so this feature is available

satfan52, 2019-09-26:
Ok I cloned a "fresh" copy of the repository and it now shows "4.11.1" instead of "4.8.0" so I guess I am back on track. It is curious that a simple "git pull" did not allow me to upgrade and thaty I had to do a fresh clone. Anyway, I will now test the functionality itself

satfan52, 2019-09-27:
Works like a charm in both transmit and receive mode with the SDR transceiver app of Pavel (125-14 board)
                         
I used the the following SoapySDR arguments : addr=192.168.0.100, port=1001
 (where 192.168.0.100 is the assigned IP address of my redpitaya board)

Many thanks!!!!