f4exb/sdrangel #363: Kiwisdr plugin not loading

N5FPP, 2019-06-13:
The kiwisdr plugin is not being loaded at startup and does not show in the plugin window.

I verified that it was built and installed in /usr/local/lib/sdrangel/plugins.

`
2019-06-12 22:39:51.300 (I) SDRangel 4.10.0 Qt 5.9.5 64b x86_64 Ubuntu 18.04.2 LTS DSP Rx:24b Tx:16b PID 687
2019-06-12 22:39:51.300 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/pascal/.config/f4exb/SDRangel.conf
2019-06-12 22:39:53.675 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2019-06-12 22:39:53.676 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2019-06-12 22:39:53.676 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2019-06-12 22:39:53.677 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2019-06-12 22:39:53.706 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
2019-06-12 22:39:53.706 (I) PluginManager::loadPluginsDir: loaded plugin libdemodlora.so
2019-06-12 22:39:53.706 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2019-06-12 22:39:53.707 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2019-06-12 22:39:53.707 (I) PluginManager::loadPluginsDir: loaded plugin libdemodudpsrc.so
2019-06-12 22:39:53.708 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2019-06-12 22:39:53.708 (I) PluginManager::loadPluginsDir: loaded plugin libudpsink.so
2019-06-12 22:39:53.709 (I) PluginManager::loadPluginsDir: loaded plugin libmodam.so
2019-06-12 22:39:53.812 (I) PluginManager::loadPluginsDir: loaded plugin libmodatv.so
2019-06-12 22:39:53.813 (I) PluginManager::loadPluginsDir: loaded plugin libmodnfm.so
2019-06-12 22:39:53.813 (I) PluginManager::loadPluginsDir: loaded plugin libmodssb.so
2019-06-12 22:39:53.814 (I) PluginManager::loadPluginsDir: loaded plugin libmodudpsink.so
2019-06-12 22:39:53.815 (I) PluginManager::loadPluginsDir: loaded plugin libmodwfm.so
2019-06-12 22:39:53.816 (I) PluginManager::loadPluginsDir: loaded plugin libudpsource.so
2019-06-12 22:39:53.817 (I) PluginManager::loadPluginsDir: loaded plugin liboutputbladerf1.so
2019-06-12 22:39:53.817 (I) PluginManager::loadPluginsDir: loaded plugin liboutputbladerf2.so
2019-06-12 22:39:53.818 (I) PluginManager::loadPluginsDir: loaded plugin liboutputfilesink.so
2019-06-12 22:39:53.819 (I) PluginManager::loadPluginsDir: loaded plugin liboutputhackrf.so
2019-06-12 22:39:53.820 (I) PluginManager::loadPluginsDir: loaded plugin liboutputlimesdr.so
2019-06-12 22:39:53.822 (I) PluginManager::loadPluginsDir: loaded plugin liboutputplutosdr.so
2019-06-12 22:39:53.823 (I) PluginManager::loadPluginsDir: loaded plugin liboutputsoapysdr.so
2019-06-12 22:39:53.824 (I) PluginManager::loadPluginsDir: loaded plugin liboutputsoapysdrsrv.so
2019-06-12 22:39:53.824 (I) PluginManager::loadPluginsDir: loaded plugin libinputbladerf1.so
2019-06-12 22:39:53.825 (I) PluginManager::loadPluginsDir: loaded plugin libinputbladerf2.so
2019-06-12 22:39:53.826 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdpro.so
2019-06-12 22:39:53.827 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdproplus.so
2019-06-12 22:39:53.828 (I) PluginManager::loadPluginsDir: loaded plugin libinputfilesource.so
2019-06-12 22:39:53.828 (I) PluginManager::loadPluginsDir: loaded plugin libinputhackrf.so
2019-06-12 22:39:53.830 (I) PluginManager::loadPluginsDir: loaded plugin libinputlimesdr.so
2019-06-12 22:39:53.831 (I) PluginManager::loadPluginsDir: loaded plugin libinputplutosdr.so
2019-06-12 22:39:53.832 (I) PluginManager::loadPluginsDir: loaded plugin libinputplutosdrsrv.so
2019-06-12 22:39:53.833 (I) PluginManager::loadPluginsDir: loaded plugin libinputrtlsdr.so
2019-06-12 22:39:53.834 (I) PluginManager::loadPluginsDir: loaded plugin libinputsoapysdr.so
2019-06-12 22:39:53.834 (I) PluginManager::loadPluginsDir: loaded plugin libinputsoapysdrsrv.so
2019-06-12 22:39:53.835 (I) PluginManager::loadPluginsDir: loaded plugin libinputtestsource.so
2019-06-12 22:39:54.563 (W) "No such interface 'org.freedesktop.DBus.Properties' on object at path /org/freedesktop/NetworkManager/ActiveConnection/2"
2019-06-12 22:39:54.580 (W) AudioOutput::start: using system default device alsa_output.pci-0000_00_1b.0.analog-surround-50
2019-06-12 22:39:54.580 (I) AudioOutput::start: audio format OK
2019-06-12 22:39:54.622 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
2019-06-12 22:42:27.073 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-06-12 22:42:27.074 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-06-12 22:42:27.074 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-06-12 22:42:27.074 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-06-12 22:42:27.089 (I) WebAPIServer::stop: stopped web API server at http://127.0.0.1:8091
SDRangel quit.
`

Is there anything I need to check or verify? The libinputkiwisdr.so is there and the permissions/ownership look right.

-Freeman, N5FPP

f4exb, 2019-06-13:
Seems it is not there at all. There was a problem during build.

f4exb, 2019-06-13:
I've tried it in a fresh VM and it works.
There is something funny: iy should install the libs in `/usr/lib/sdrangel` and the bin in `usr/bin`.

f4exb, 2019-06-15:
Mind that you need `libqt5websockets5-dev` for the KiwiSDR plugin. The compilation of the KiwiSDR plugin is skipped otherwise.

closed 2019-06-25 by f4exb