f4exb/sdrangel #364: Debian package should depend on libqt5multimedia5-plugins

theasp, 2019-06-15:
The Debian/Ubuntu package should depend on libqt5multimedia5-plugins, otherwise the audio devices won't be detected, so the device list shows only default, and there will be no audio.  After looking through previous issues I think this may be the cause of some of them,

f4exb, 2019-06-15:
Maybe you should look twice because it does. See: https://github.com/f4exb/sdrangel/blob/master/debian/control

theasp, 2019-06-15:
The control file lists it as a build dependency, not an actual dependency.

f4exb, 2019-06-15:
???

theasp, 2019-06-15:
https://github.com/f4exb/sdrangel/blob/32e0fb5dc473d12406b8f8ffda127711942fd725/debian/control#L38

The multimedia plugins won't be detected as a dependency unless you name them specifically.

theasp, 2019-06-15:
```
phillipsa-lt:~$ dpkg-query -Wf '${Depends}' sdrangel | sed -re 's/(, |$)/\n/g' | grep libqt5
libqt5core5a (>= 5.9.0~beta)
libqt5gui5 (>= 5.2.0)
libqt5multimedia5 (>= 5.6.0~beta)
libqt5network5 (>= 5.8.0)
libqt5opengl5 (>= 5.0.2)
libqt5websockets5 (>= 5.6.0)
libqt5widgets5 (>= 5.2.0~alpha1)

f4exb, 2019-06-15:
I have no time to deal with Debian packaging mysteries hence 4.10.1 is the latest version with packages provided in the assets section

theasp, 2019-06-16:
Classy...

f4exb, 2019-06-16:
This is open source, you have the source and this should be enough period.

theasp, 2019-06-16:
Actually I was commenting on your rude responses, not your lack of action.  I'd offer a PR to fix this, but now you can get stuffed.

f4exb, 2019-06-16:
Fair enough you can go away

closed 2019-06-16 by f4exb

theasp, 2019-07-03:
FWIW, your recent commit should solve this:
https://github.com/f4exb/sdrangel/commit/7b1232dbc2070658157668a2d85417e196fd398c#diff-d837a1c17de0268bcea321239ddbed47