f4exb/sdrangel #365: Segfault

bdn76, 2019-06-16:
`$ ./sdrangel
2019-06-16 17:21:39.011 (I) SDRangel 4.10.1 Qt 5.11.3 64b x86_64 Debian GNU/Linux 10 (buster) DSP Rx:24b Tx:16b PID 3207
2019-06-16 17:21:39.011 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/dima/.config/f4exb/SDRangel.conf
2019-06-16 17:21:39.011 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #0: jack
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #1: pulse
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #2: default
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #3: sysdefault:CARD=PCH
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #4: front:CARD=PCH,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #5: dmix:CARD=PCH,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #6: dmix:CARD=PCH,DEV=2
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #7: dsnoop:CARD=PCH,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #8: dsnoop:CARD=PCH,DEV=2
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #9: hw:CARD=PCH,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #10: hw:CARD=PCH,DEV=2
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #11: plughw:CARD=PCH,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #12: plughw:CARD=PCH,DEV=2
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #13: usbstream:CARD=PCH
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #14: sysdefault:CARD=USB
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #15: front:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #16: surround21:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #17: surround40:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #18: surround41:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #19: surround50:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #20: surround51:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #21: surround71:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #22: iec958:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #23: dmix:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #24: dsnoop:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #25: hw:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #26: plughw:CARD=USB,DEV=0
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #27: usbstream:CARD=USB
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #28: alsa_input.pci-0000_00_1f.3.analog-stereo
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #29: alsa_output.dmixer.monitor
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: input device #30: alsa_output.pci-0000_00_1f.3.analog-stereo.monitor
2019-06-16 17:21:39.331 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #0: jack
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #1: pulse
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #2: default
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #3: sysdefault:CARD=PCH
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #4: front:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #5: surround21:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #6: surround40:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #7: surround41:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #8: surround50:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #9: surround51:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #10: surround71:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #11: hdmi:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #12: hdmi:CARD=PCH,DEV=1
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #13: hdmi:CARD=PCH,DEV=2
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #14: hdmi:CARD=PCH,DEV=3
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #15: hdmi:CARD=PCH,DEV=4
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #16: dmix:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #17: dmix:CARD=PCH,DEV=3
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #18: dmix:CARD=PCH,DEV=7
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #19: dmix:CARD=PCH,DEV=8
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #20: dmix:CARD=PCH,DEV=9
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #21: dmix:CARD=PCH,DEV=10
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #22: dsnoop:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #23: dsnoop:CARD=PCH,DEV=3
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #24: dsnoop:CARD=PCH,DEV=7
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #25: dsnoop:CARD=PCH,DEV=8
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #26: dsnoop:CARD=PCH,DEV=9
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #27: dsnoop:CARD=PCH,DEV=10
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #28: hw:CARD=PCH,DEV=0
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #29: hw:CARD=PCH,DEV=3
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #30: hw:CARD=PCH,DEV=7
2019-06-16 17:21:39.951 (D) AudioDeviceManager::AudioDeviceManager: output device #31: hw:CARD=PCH,DEV=8
2019-06-16 17:21:39.952 (D) AudioDeviceManager::AudioDeviceManager: output device #32: hw:CARD=PCH,DEV=9
2019-06-16 17:21:39.952 (D) AudioDeviceManager::AudioDeviceManager: output device #33: hw:CARD=PCH,DEV=10
[... 159 lines cut ...]
2019-06-16 17:21:40.193 (D) FFTEngine::create: using FFTW engine
2019-06-16 17:21:40.623 (D) FFT: creating FFTW plan (n=1024,forward) took 430ms
2019-06-16 17:21:40.623 (D) GLSpectrum::connectTimer
2019-06-16 17:21:40.625 (D) DSPDeviceSourceEngine::addSink:  SpectrumVis
2019-06-16 17:21:40.625 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
Segmentation fault (core dumped)
`
gdb:
`Reading symbols from ./sdrangel...(no debugging symbols found)...done.
[New LWP 3207]
[New LWP 3210]
[New LWP 3214]
[New LWP 3208]
[New LWP 3209]
[New LWP 3212]
[New LWP 3211]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./sdrangel'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  __GI___libc_free (mem=0x2) at malloc.c:3093
3093	malloc.c: No such file or directory.
[Current thread is 1 (Thread 0x7fe0a07ec540 (LWP 3207))]
(gdb) 
`

f4exb, 2019-06-16:
Can you compile in debug mode so that we can have more information?

bdn76, 2019-06-16:
Of course. How i can enable debug mode?

f4exb, 2019-06-16:
This is now standard cmake. Use : `-DCMAKE_BUILD_TYPE=Debug` option when invoking `cmake`.

bdn76, 2019-06-16:
Kind of magick but now it's work fine ;)
[image]

closed 2019-06-16 by bdn76