f4exb/sdrangel #366: Device scan cache

f4exb, 2019-06-17:
Today each plugin will initiate a scan for the relevant devices for it even if they pertain to the same kind of device. Like the LimeSDR input plugin will scan for all LimeSDR device then later the LimeSDR output plugin will do the same thing again. However the gathered information is the same in both cases it is only the post process that is different. The device scan is by large the thing that takes the longest time at startup (just check the splash screen to get an idea...) so optimizing it would significantly reduce the initial program load time. Moreover when a third type of plugin (MIMO) will be put in place the same operation could take place 3 times. The criticality of this could justify that this is a v5 prerequisite.

The idea is to gather the connected devices information in some sort of cache present in the "core" of the program and can be accessed when the enumeration routine of the plugin is called.

gaspar1987, 2019-06-28:
hello j ai pas le device de rtl sdr comme faire le interniez sur sdrangel 4.10.2

f4exb, 2019-06-28:
Unreadable and unrelated to the subject

gaspar1987, 2019-06-28:
pas de rtl sdr sur vous logiciel  merci ?

f4exb, 2019-06-28:
Which system? Windows, Ubuntu 18.04, Ubuntu 19.04, Debian Buster ?

gaspar1987, 2019-06-28:
la cle rtl sdr marche pas sur la nouvelle version !

gaspar1987, 2019-06-28:
c est window s 8.1

f4exb, 2019-06-28:
Should be built as an external library but is missing here: https://github.com/f4exb/sdrangel/blob/master/external/CMakeLists.txt

f4exb, 2019-06-29:
Fixed in v4.10.3

closed 2019-06-29 by f4exb

gaspar1987, 2019-06-29:
slt de FRS 2606 qz merci F4exb .

f4exb, 2019-06-29:
Thanks for reporting :smile:

gaspar1987, 2019-06-29:
bonjour  je veux  le dsd  de crypto audio sur version nouvelle 4.10.4 sdrangel Windows 8.1 ?