f4exb/sdrangel #367: SDRAngel for old version

maciek84krk, 2019-06-17:
Hello!

Is any chance to have old binary version of sdrangel which was working with Debian Stretch?
New version want new libc, new libraries etc, and for me its change a half of all system, and
 i have very restricted system based on stretch which can't have newest libraries.
Can somebody give me old version for stretch in binary .deb?

I will very happy if yes... My mail maciek.tumidajski@gmail.com

My suggestion for author: Program is very very super, but don't remove old versions so 
 fast, many peoples can't change system to new in some days :)

Maciej

f4exb, 2019-06-17:
Stretch is too old and has dependency incompatibilities with the newest changes so it is definitely dropped.

f4exb, 2019-06-18:
This is essentially your problem. However if you are able to install Docker and I think this should be possible even on Stretch then you can build a Docker image which is based on Ubuntu 18.04 and therefore get free from all compatibility issues. See: https://github.com/f4exb/sdrangel-docker

maciek84krk, 2019-06-18:
Thank you f4exb, i think it's solving my problem :)

closed 2019-06-25 by f4exb