f4exb/sdrangel #372: Double free detected core dumped

Funcan, 2019-06-29:
Compiling from git; 563a4d589

Building on ubuntu 19.04

```
duncan@ion:~/src/sdrangel/build$ ./sdrangel
...
2019-06-30 00:29:33.319 (I) PluginManager::loadPluginsDir: loaded plugin libudpsink.so
2019-06-30 00:29:33.320 (I) PluginManager::loadPluginsDir: loaded plugin libudpsource.so
free(): double free detected in tcache 2
Aborted (core dumped)
```

More info as I figure it out, please do let me know if anything specific is useful

Funcan, 2019-06-29:
Backtrace of the free in question:

```
Thread 1 "sdrangel" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff6355ed7 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff6337535 in __GI_abort () at abort.c:79
#2  0x00007ffff639e726 in __libc_message
    (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff64c4952 "%s\n")
    at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff63a559a in malloc_printerr
    (str=str@entry=0x7ffff64c65e0 "free(): double free detected in tcache 2")
    at malloc.c:5352
#4  0x00007ffff63a76d5 in _int_free
    (av=0x7ffff64f6c40 <main_arena>, p=0x5555563a3250, have_lock=<optimized out>) at malloc.c:4209
#5  0x00007ffff7960c7b in DeviceUserArgs::findDeviceArgs(QString const&, int)
    () at /home/duncan/src/sdrangel/build/lib/libsdrbase.so
#6  0x00007ffff7960cb4 in DeviceUserArgs::findUserArgs(QString const&, int) ()
    at /home/duncan/src/sdrangel/build/lib/libsdrbase.so
#7  0x00007ffff7ea1bc7 in MainWindow::addSourceDevice(int) ()
    at /home/duncan/src/sdrangel/build/lib/libsdrgui.so
#8  0x00007ffff7ea6944 in MainWindow::MainWindow(qtwebapp::LoggerWithFile*, MainParser const&, QWidget*) () at /home/duncan/src/sdrangel/build/lib/libsdrgui.so
#9  0x0000555555556ca1 in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) ()
#10 0x0000555555556426 in main ()
```

Funcan, 2019-06-29:
v4.9.1 builds and starts fine

Funcan, 2019-06-30:
v4.10.1 has the same failure

f4exb, 2019-06-30:
Attempted to fix it on dev branch: 19305ee

Funcan, 2019-06-30:
ea1b3e90b41 seems to be the first commit with the issue, according to a bisect. I'll try your new branch

Funcan, 2019-06-30:
Ok, your new version builds and starts without an error. Thanks!

Funcan, 2019-06-30:
The pre-compiled sdrangel_4.10.3-1_amd64_disco.deb doesn't start for me either btw, same issue

f4exb, 2019-06-30:
The issue appeared with 4.10.1 and the device argument feature. I have changed the way the QLIst is handled using the naive approach instead of iterators and removed the operator()==. Not sure what exactly is wrong (I don't have the issue) but `DeviceUserArgs::findDeviceArgs` was a good indication thanks!

f4exb, 2019-07-05:
Fix done in v4.10.4

closed 2019-07-05 by f4exb