f4exb/sdrangel #373: Could not find any LimeSDR device  //  sdrangel-4.10.4-win64

netdoggy, 2019-07-03:
Could not find any LimeSDR device  on v 4.10.4
But finds and works in the previously installed version v 4.5.4

LimeUtil.exe --find
  * [LimeSDR-USB, media=USB 3.0, module=FX3, serial=0009081C05C11B21, index=0]

...
2019-07-03 17:33:05.934 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.hackrf
2019-07-03 17:33:05.940 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.kiwisdrsource
2019-07-03 17:33:05.940 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.limesdr
**2019-07-03 17:33:05.940 (D) LimeSDRInputPlugin::enumSampleSources: Could not find any LimeSDR device**
2019-07-03 17:33:05.940 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.localinput
2019-07-03 17:33:05.941 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.plutosdr
ERROR: Unable to find host: ▌ЄюЄ їюёЄ эхшчтхёЄхэ.
2019-07-03 17:33:08.236 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.rtlsdr
2019-07-03 17:33:08.254 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.remoteinput
2019-07-03 17:33:08.254 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.soapysdrinput
...

f4exb, 2019-07-06:
Linux:
```
2019-07-06 03:41:52.321 (D) DeviceLimeSDRParams::open: serial: LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=0009060B00473419 type: 2
Gateware version mismatch!
  Expected gateware version 2, revision 20
  But found version 2, revision 16
  Follow the FW and FPGA upgrade instructions:
  http://wiki.myriadrf.org/Lime_Suite#Flashing_images
  Or run update on the command line: LimeUtil --update

Reference clock 30.72 MHz
2019-07-06 03:41:52.668 (D) DeviceLimeSDRParams::open:  2  Rx channels for device  LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=0009060B00473419
2019-07-06 03:41:52.668 (D) DeviceLimeSDRParams::open:  2  Tx channels for device  LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=0009060B00473419
2019-07-06 03:41:52.668 (D) LimeSDRInputPlugin::enumSampleSources: device #1 channel 0: LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=0009060B00473419
2019-07-06 03:41:52.668 (D) LimeSDRInputPlugin::enumSampleSources: device #1 channel 1: LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=0009060B00473419
```

f4exb, 2019-07-06:
The LimeUtil.exe from the build shows this:
```
 ~/sdrangel/build/bin (dev)
$ ./LimeUtil.exe --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:      v19.01.0-release
  Build timestamp:      2019-06-27
  Interface version:    v2019.1.0
  Binary interface:     19.01-1

System resources:
  Installation root:    C:\Users\F4EXB\sdrangel\build
  User home directory:  C:\Users\F4EXB
  App data directory:   C:\Users\F4EXB\AppData\Roaming/LimeSuite
  Config directory:     C:\Users\F4EXB/.limesuite
  Image search paths:
     - C:\Users\F4EXB\AppData\Roaming/LimeSuite/images
     - C:\Users\F4EXB\sdrangel\build/share/LimeSuite/images

Supported connections:
   * FT601
   * PCIEXillybus
```
It seems it does not include Cypress FX3 support

f4exb, 2019-07-06:
Checked LimeSuite's `appveyor.yml` and apparently it needs the FX3SDK specifying it in `-DFX3_SDK_PATH` when compiling LimeSuite with MSVC. Provided this we now have:
```
4 ~/sdrangel/build/bin (dev)
$ ./LimeUtil.exe --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:      v19.01.0-release
  Build timestamp:      2019-07-06
  Interface version:    v2019.1.0
  Binary interface:     19.01-1

System resources:
  Installation root:    C:\Users\F4EXB\sdrangel\build
  User home directory:  C:\Users\F4EXB
  App data directory:   C:\Users\F4EXB\AppData\Roaming/LimeSuite
  Config directory:     C:\Users\F4EXB/.limesuite
  Image search paths:
     - C:\Users\F4EXB\AppData\Roaming/LimeSuite/images
     - C:\Users\F4EXB\sdrangel\build/share/LimeSuite/images

Supported connections:
   * FT601
   * FX3
   * PCIEXillybus
````
FX3 support is there

f4exb, 2019-07-06:
Alas this does not fix the issue.

netdoggy, 2019-07-06:
Alas, I am new to lime and do not know all the details of the lime API yet :)
If something needs to be checked on FX3, then I will be happy to check

netdoggy, 2019-07-07:
Hm I just installed the .5 build, and it looks like it solved the problem.

f4exb, 2019-07-07:
Good to know :smile: For some (other) reason it is not working on mine. So I assume the problem is solved.

closed 2019-07-07 by f4exb