f4exb/sdrangel #374: Unwanted consle window in 4.10.4

gm8arv, 2019-07-05:
I see a console window in 4.10.4 which wasn't in 4.10.0.  How can I hide this window, please?  I tried setting the console logging level to Info, but the unwanted console still appears, and the level is back to debugging.

f4exb, 2019-07-05:
"and the level is back to debugging." Sorry,,, no. Only until the logger is initialized then you have the requested level of messages.

Vort, 2019-07-05:
Windows GUI programs traditionally have no console window visible.

f4exb, 2019-07-05:
I understand you may want a toy app to get along with your toy O/S. For serious usage you do need console messages. 

I am not going to do anything about it personally but ready to receive a PR.

Vort, 2019-07-05:
> I am not going to do anything about it personally but ready to receive a PR.

However you was doing something to remove console hiding option.

> For serious usage you do need console messages.

Messages - yes, console - no.
Look how it is done in VLC:
[image]

f4exb, 2019-07-05:
> However you was doing something to remove console hiding option.
The build process for Windows is entirely new since 4.10.3 so maybe something got lost in the translation.

gm8arv, 2019-07-05:
I can understand that the change was unintentional and unexpected.  I agree that console/debugging information can be helpful, but I would prefer that it was normally disabled, and perhaps a command-line switch -console be added to enable the window.  Alternatively, perhaps a "-noconsole" flag?  Let me know if I can help.

Vort, 2019-07-05:
> The build process for Windows is entirely new since 4.10.3 so maybe something got lost in the translation.

I guess that [`/SUBSYSTEM:WINDOWS`](https://docs.microsoft.com/en-us/cpp/build/reference/subsystem-specify-subsystem?view=vs-2017) linker parameter was lost. Did not checked yet.

> but I would prefer that it was normally disabled, and perhaps a command-line switch -console be added to enable the window.

Console visibility is usually hardcoded inside exe file header.
Most likely, it is possible to control its appearing in runtime, but software do not do it usually.

Vort, 2019-07-05:
This change should fix the problem:
(also imported improvement for Qt detection from decff91427db150ee7043f708afe7d4d694e664c)
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index d4dc402..5a93a84 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -480,6 +480,10 @@ if (BUILD_GUI)
     sdrgui
     logging
     )
+
+    if (MSVC)
+        set(CMAKE_EXE_LINKER_FLAGS_RELEASE "${CMAKE_EXE_LINKER_FLAGS} /SUBSYSTEM:WINDOWS /ENTRY:mainCRTStartup")
+    endif()
 endif()
 
 ############ build sdrangel server ################
diff --git a/cmake/Modules/QtLocator.cmake b/cmake/Modules/QtLocator.cmake
index 1403bbb..34d97aa 100644
--- a/cmake/Modules/QtLocator.cmake
+++ b/cmake/Modules/QtLocator.cmake
@@ -2,7 +2,7 @@ SET(QT_MISSING True)
 # msvc only; mingw will need different logic
 IF(MSVC)
     # look for user-registry pointing to qtcreator
-    GET_FILENAME_COMPONENT(QT_BIN [HKEY_CURRENT_USER\\Software\\Classes\\Applications\\QtProject.QtCreator.cpp\\shell\\Open\\Command] PATH)
+    GET_FILENAME_COMPONENT(QT_BIN [HKEY_CURRENT_USER\\Software\\Classes\\Applications\\QtProject.QtCreator.pro\\shell\\Open\\Command] PATH)
 
     # get root path so we can search for 5.3, 5.4, 5.5, etc
     STRING(REPLACE "/Tools" ";" QT_BIN "${QT_BIN}")
@@ -22,7 +22,15 @@ IF(MSVC)
     # - qt uses (e.g.) "msvc2012"
     # - cmake uses (e.g.) "1800"
     # - see also https://cmake.org/cmake/help/v3.0/variable/MSVC_VERSION.html
-    MATH(EXPR QT_MSVC "2000 + (${MSVC_VERSION} - 200) / 100")
+
+    # check compiler version
+    if(MSVC_VERSION GREATER 1910 AND MSVC_VERSION LESS 1919)
+      set(QT_MSVC 2017)
+    elseif(MSVC_VERSION GREATER 1899 AND MSVC_VERSION LESS 1910)
+      set(QT_MSVC 2015)
+    else()
+      MATH(EXPR QT_MSVC "2000 + (${MSVC_VERSION} - 600) / 100")
+    endif()
 
     # check for 64-bit os
     # may need to be removed for older compilers as it wasn't always offered 
```

f4exb, 2019-07-05:
Tried it. Looks like it's doing the job 😄

netdoggy, 2019-07-07:
As for me the console was useful.
Maybe you should add or parameter to display the console. Or, judging by the fact, it's all about the build paramters, then you can probably build a build with 2 exes :)

Vort, 2019-07-07:
@netdoggy console window can be enabled with the help of the hex editor.
Just replace byte `02` with `03` at address `174`:
[image]

f4exb, 2019-07-07:
I am not going to do zillions of versions (even just two) and there is not going to be an option to show the console. So there will be no console started period. You have three ways to get the console messages:

  - use the trick just above for a permanent change
  - start the program from the bash prompt of the "Git for Windows" application https://git-scm.com/download/win (as a bonus you will have git)
  - Write the log to file See https://github.com/f4exb/sdrangel/blob/master/sdrgui/readme.md#12-preferences---logging

So I am going to close the issue.

closed 2019-07-07 by f4exb

gm8arv, 2019-07-07:
Perfect solution, thanks!  73, David

netdoggy, 2019-07-07:
Great, Thank you!