f4exb/sdrangel #376: SSB modulador version 10.    Hackrf   windows

Pedro9012, 2019-07-11:
Hola, a partir de esta version el modo ssb a hido a peor ,  o le pones con el volumen a 0,1 sin AGC ,
 o bien le pones con AGC y volumen a 9,9 ,  otros ajustes sale totalmente distorsionado.
Sdrangel v 4.10.02 , windows 10 pro , hackrf.
¿alguien mas tiene este problema?
Un SALUDO.

Hello, from this version the mode ssb to hido to worse, or you put it with the volume at 0.1 without AGC,
 or you put it with AGC and volume at 9.9, other settings comes out totally distorted.
Sdrangel v 4.10.02, windows 10 pro, hackrf.
Does anyone else have this problem?
A greeting.

f4exb, 2019-07-17:
It all depends on how you set the numerous controls of the CMP which is indeed an AGC. Anyway in v4.11.1 I am going to implement a real compressor based on the code of the sndfilter library which is a far better implementation as the first tests tend to prove.

f4exb, 2019-07-18:
As mentioned above v4.11.1 implements a new proper audio compressor with a single on/off button. This should fix issues related to compressor (not to your individual audio settings).

closed 2019-07-18 by f4exb

Pedro9012, 2019-07-20:
Ok, thank you very much for the information