f4exb/sdrangel #377: build error

KC7NOA, 2019-07-12:
I hate to post these .... i get a warning in the cmaketxt about and unresolved if / then statement ...

-- Asking qmake for QT_IMPORTS_DIR and got /usr/lib/x86_64-linux-gnu/qt5/imports
CMake Warning (dev) in cmake/cpack/CMakeLists.txt:
  A logical block opening on the line

    /home/kc7noa1/Downloads/sdrangel/cmake/cpack/CMakeLists.txt:15 (if)

  closes on the line

    /home/kc7noa1/Downloads/sdrangel/cmake/cpack/CMakeLists.txt:307 (endif)

  with mis-matching arguments.
This warning is for project developers.  Use -Wno-dev to suppress it.

but the error is

[ 78%] Built target demodam
[ 78%] Building CXX object plugins/channelrx/demodbfm/CMakeFiles/demodbfm.dir/rdsparser.cpp.o
[ 78%] Building CXX object plugins/channelrx/demodnfm/CMakeFiles/demodnfm.dir/demodnfm_autogen/mocs_compilation.cpp.o
/home/kc7noa1/Downloads/sdrangel/plugins/channelrx/demodnfm/nfmdemodgui.cpp: In member function ‘void NFMDemodGUI::displaySettings()’:
/home/kc7noa1/Downloads/sdrangel/plugins/channelrx/demodnfm/nfmdemodgui.cpp:407:9: error: ‘class Ui::NFMDemodGUI’ has no member named ‘highPassFilter’
     ui->highPassFilter->setChecked(m_settings.m_highPass);
         ^~~~~~~~~~~~~~
plugins/channelrx/demodnfm/CMakeFiles/demodnfm.dir/build.make:134: recipe for target 'plugins/channelrx/demodnfm/CMakeFiles/demodnfm.dir/nfmdemodgui.cpp.o' failed
make[2]: *** [plugins/channelrx/demodnfm/CMakeFiles/demodnfm.dir/nfmdemodgui.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 79%] Building CXX object plugins/channelrx/demodbfm/CMakeFiles/demodbfm.dir/rdstmc.cpp.o
Scanning dependencies of target demodssb
[ 79%] Building CXX object plugins/channelrx/demodssb/CMakeFiles/demodssb.dir/ssbdemod.cpp.o
[ 79%] Building CXX object plugins/channelrx/demodbfm/CMakeFiles/demodbfm.dir/bfmdemodgui.cpp.o
CMakeFiles/Makefile2:1544: recipe for target 'plugins/channelrx/demodnfm/CMakeFiles/demodnfm.dir/all' failed
make[1]: *** [plugins/channelrx/demodnfm/CMakeFiles/demodnfm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 79%] Building CXX object plugins/channelrx/demodssb/CMakeFiles/demodssb.dir/ssbdemodsettings.cpp.o
[ 79%] Building CXX object plugins/channelrx/demodbfm/CMakeFiles/demodbfm.dir/demodbfm_autogen/mocs_compilation.cpp.o
[ 80%] Building CXX object plugins/channelrx/demodssb/CMakeFiles/demodssb.dir/ssbplugin.cpp.o
[ 80%] Building CXX object plugins/channelrx/demodssb/CMakeFiles/demodssb.dir/ssbdemodgui.cpp.o
[ 80%] Building CXX object plugins/channelrx/demodssb/CMakeFiles/demodssb.dir/demodssb_autogen/mocs_compilation.cpp.o
[ 80%] Linking CXX shared library ../../../lib/plugins/libdemodssb.so
[ 80%] Linking CXX shared library ../../../lib/plugins/libdemodbfm.so
[ 80%] Built target demodssb
[ 80%] Built target demodbfm
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
kc7noa1@kc7noa1-Latitude-E5510:~/Downloads/sdrangel/build$

f4exb, 2019-07-13:
It is there:
https://github.com/f4exb/sdrangel/blob/5096816143a08b39cef227aea7b9b4788a175cb7/plugins/channelrx/demodnfm/nfmdemodgui.ui#L578
Please check your repository copy.

KC7NOA, 2019-07-17:
Ill not do a git pull .. ill just start a new folder ...

rgetz, 2019-07-18:
I get the same warning during cmake on the dev branch
```
CMake Warning (dev) in cmake/cpack/CMakeLists.txt:
  A logical block opening on the line

    /home/rgetz/github/sdrangel/cmake/cpack/CMakeLists.txt:15 (if)

  closes on the line

    /home/rgetz/github/sdrangel/cmake/cpack/CMakeLists.txt:307 (endif)

  with mis-matching arguments.
This warning is for project developers.  Use -Wno-dev to suppress it.
```

That's because:

```
rgetz@brain:~/github/sdrangel/build2$ sed -n 15p /home/rgetz/github/sdrangel/cmake/cpack/CMakeLists.txt
if(APPLE AND BUNDLE AND BUILD_GUI)
rgetz@brain:~/github/sdrangel/build2$ sed -n 307p /home/rgetz/github/sdrangel/cmake/cpack/CMakeLists.txt
endif(WIN32 OR MINGW)
```

but mine builds fine with the things I enabled.

f4exb, 2019-07-19:
I always use -Wno-dev so I don't see these. I think also that the closing endifs can simply be `endif()`.

rgetz, 2019-07-19:
yeah, this removes the warning, but I also see why it was done...

```
diff --git a/cmake/cpack/CMakeLists.txt b/cmake/cpack/CMakeLists.txt
index 1a8f3cf3c..f23ead1e4 100644
--- a/cmake/cpack/CMakeLists.txt
+++ b/cmake/cpack/CMakeLists.txt
@@ -304,7 +304,7 @@ elseif(WIN32 OR MINGW)
       "Delete '$SMPROGRAMS\\\\$START_MENU\\\\SDRangel.lnk'")
   endif(BUNDLE)
 
-endif(WIN32 OR MINGW)
+endif(APPLE AND BUNDLE AND BUILD_GUI)
 
 message(STATUS "CPack generators: ${CPACK_GENERATOR}")
 include(CPack)
```
the existing code does this (match the latest test):
```
15 :if(APPLE AND BUNDLE AND BUILD_GUI)
110:elseif(LINUX AND BUNDLE)
173:elseif(WIN32 OR MINGW)
307:endif(WIN32 OR MINGW)
```

cmake wants (a match with the original if)
```
15 :if(APPLE AND BUNDLE AND BUILD_GUI)
110:elseif(LINUX AND BUNDLE)
173:elseif(WIN32 OR MINGW)
307:endif(APPLE AND BUNDLE AND BUILD_GUI)
```

commit 1e655bae (2019-07-25): Corrected endif in cpack/CMakeLists.txt. Fixes #377
closed 2019-07-27 by f4exb