f4exb/sdrangel #379: without ~/.config/f4exb/SDRangel.conf I get weird behavior on dev branch

rgetz, 2019-07-18:
It looks like the dev branch doesn't make the .config/f4exb directory by default?

If I just build on a fresh machine, the directory isn't created, and I get some weird GUI issues (like the main spectrum analyzer window not wanting to change width, and snapping back as soon as the mouse button is released).

mkdir -p  $HOME/.config/f4exb

seems to fix the problem...

f4exb, 2019-07-19:
Strange.... Since this is manged by Qt I would expect Qt to take care about it as soon as things are being written to it. In Windows I think it uses the registry which of course is already there.

rgetz, 2019-07-19:
I think I understand (maybe)

```
rm -rf ~/.config/f4exb
strace -otrace.txt ./build/sdrangel
```

If you quit via the application by clicking on the  "X" icon (on the upper/right window):
```
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffebe2c6740) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffebe2c66e0) = -1 ENOENT (No such file or directory)
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffebe2c6740) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffebe2c66e0) = -1 ENOENT (No such file or directory)
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffebe2c66d0) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffebe2c6670) = -1 ENOENT (No such file or directory)
```

the application tries to open things, but can't - it's never there.

If you do the same (run, attach to a radio, and either quit with CTRL-Q or File -> Quit), strace will give you this:

```
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d156610) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d1565b0) = -1 ENOENT (No such file or directory)
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d156610) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d1565b0) = -1 ENOENT (No such file or directory)
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d1565a0) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d156540) = -1 ENOENT (No such file or directory)
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d155f30) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d155ed0) = -1 ENOENT (No such file or directory)
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf.lock", O_RDWR|O_CREAT|O_EXCL|O_CLOEXEC, 0666) = 46
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d156080) = -1 ENOENT (No such file or directory)
access("/home/rgetz/.config/f4exb/SDRangel.conf", F_OK) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7fff0d155fe0) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT|AT_SYMLINK_NOFOLLOW, STATX_ALL, 0x7fff0d155fd0) = -1 ENOENT (No such file or directory)
linkat(AT_FDCWD, "/proc/self/fd/47", AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_SYMLINK_FOLLOW) = 0
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, {stx_mask=STATX_ALL, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=965, ...}) = 0
statx(AT_FDCWD, "/home/rgetz/.config/f4exb/SDRangel.conf", AT_STATX_SYNC_AS_STAT, STATX_ALL, {stx_mask=STATX_ALL, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=965, ...}) = 0
access("/home/rgetz/.config/f4exb/SDRangel.conf", R_OK) = 0
access("/home/rgetz/.config/f4exb/SDRangel.conf", W_OK) = 0
access("/home/rgetz/.config/f4exb/SDRangel.conf", X_OK) = -1 EACCES (Permission denied)
chmod("/home/rgetz/.config/f4exb/SDRangel.conf", 0644) = 0
unlink("/home/rgetz/.config/f4exb/SDRangel.conf.lock") = 0
```

If opens and writes the file.

The way that the application closes is different - and it doesn't write the .conf file if you click the app "X" button.

f4exb, 2019-07-20:
Thanks for the investigation. I may have never run into this because I'm used to quit with Ctl-Q. Closing by "X" is probably handled differently in the code. Now I know where to look at.

rgetz, 2019-07-20:
I also noticed on console, closing with "X" 
```
2019-07-20 16:53:18.931 (I) WebAPIServer::stop: stopped web API server at http://127.0.0.1:8091
QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
SDRangel quit.
```

Control - Q:
```
2019-07-20 16:53:47.408 (I) WebAPIServer::stop: stopped web API server at http://127.0.0.1:8091
SDRangel quit.
```

So, yeah, it looks like it is different somewhere... 

Thanks for looking into it - I have seen the same on Windows, and many end users just click the "X" button to close an application there.

-Robin

gvanem, 2019-07-22:
> In Windows I think it uses the registry which of course is already there. 

@f4exb It would be nice for the Windows version to have a `%APPDATA%/SDRangel.conf` too. 
*BTW* where in the code is that read/written?

Now the Qt crap (the `@ByteArray`) under `HKEY_CURRENT_USER\Software\f4exb\SDRangel` has caused my  GUI to become unusable. I had to delete most of them to regain control. Not very nice IMHO.

*PS*. it would be nice if the main-window position be saved and reused on the next launch of SDRangel.exe.

rgetz, 2019-07-22:
I think this is all handled by the QtCore stuff - all sdrangel does is interface with the Qt Setting class https://doc.qt.io/qt-5/qsettings.html in sdrbase/settings/mainsettings.cpp

If you check out the filename function Qt provides - it doesn't provide it on Windows - it only writes to the registry.
https://doc.qt.io/qt-5/qsettings.html#fileName

f4exb, 2019-07-22:
The close event (happens when you close with 'X') is handled like this: https://github.com/f4exb/sdrangel/blob/master/sdrgui/mainwindow.cpp#L813

Where the exit action is handled like this:
https://github.com/f4exb/sdrangel/blob/master/sdrgui/mainwindow.cpp#L1978

This probably explains... I think it should be unified under a single sub-method. It is probably like this since ages.

f4exb, 2019-07-22:
handling action exit triggered and closing event are mutually exclusive. When the finalization code is in both the closing event handler and the exit action handler this is what happens:
```
2019-07-22 18:33:55.080 (D) MainWindow::finalizeBeforeClose
2019-07-22 18:33:55.080 (D) MainWindow::savePresetSettings: preset [default | no name]
2019-07-22 18:33:55.080 (D) DeviceUISet::saveRxChannelSettings: channel [sdrangel.channel.demoddatv] saved
2019-07-22 18:33:55.080 (D) DeviceAPI::saveSamplingDeviceSettings: serializing source sdrangel.samplesource.fileinput[0]: 
2019-07-22 18:33:55.081 (D) AudioDeviceManager::serialize
2019-07-22 18:33:55.081 (D) AudioDeviceManager::debugAudioInputInfos:  name:  "System default device"  sampleRate:  48000  volume:  0.15
2019-07-22 18:33:55.081 (D) AudioDeviceManager::debugAudioInputInfos:  name:  "default"  sampleRate:  0  volume:  0
2019-07-22 18:33:55.081 (D) AudioDeviceManager::debugAudioInputInfos:  name:  "pulse"  sampleRate:  48000  volume:  0.15
2019-07-22 18:33:55.081 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "System default device"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  true  udpChannelMode:  2  udpChannelCodec:  5  decimationFactor:  1
2019-07-22 18:33:55.081 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "alsa_output.pci-0000_00_05.0.analog-stereo"  sampleRate:  8000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  true  udpChannelMode:  0  udpChannelCodec:  2  decimationFactor:  1
2019-07-22 18:33:55.081 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "fcd48"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  false  udpChannelMode:  0  udpChannelCodec:  0  decimationFactor:  1
2019-07-22 18:33:55.095 (D) DSPDeviceSourceEngine::stopAcquistion
2019-07-22 18:33:55.095 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2019-07-22 18:33:55.095 (D) DSPDeviceSourceEngine::gotoIdle
2019-07-22 18:33:55.095 (D) DSPDeviceSourceEngine::removeSink:  SpectrumVis
2019-07-22 18:33:55.095 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2019-07-22 18:33:55.095 (D) DeviceUISet::freeAll: destroying channel [sdrangel.channel.demoddatv]
2019-07-22 18:33:55.095 (D) AudioDeviceManager::removeAudioSink: 0x55b7414a9be0
2019-07-22 18:33:55.096 (D) AudioOutput::stop
2019-07-22 18:33:55.115 (D) DSPDeviceSourceEngine::removeThreadedSink:  ThreadedBasebandSampleSink(DownChannelizer(DATVDemod))
2019-07-22 18:33:55.116 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveThreadedBasebandSampleSink
2019-07-22 18:33:55.116 (D) ThreadedBasebandSampleSink::stop
2019-07-22 18:33:55.116 (D) MessageQueue::~MessageQueue: message:  DSPConfigureChannelizer  was still in queue
2019-07-22 18:33:55.116 (D) MessageQueue::~MessageQueue: message:  DSPConfigureChannelizer  was still in queue
2019-07-22 18:33:55.118 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-22 18:33:55.123 (D) FileInput::stop
2019-07-22 18:33:55.127 (D) DSPDeviceSourceEngine::stop
2019-07-22 18:33:55.127 (D) DSPDeviceSourceEngine::gotoIdle
2019-07-22 18:33:55.127 (D) DSPDeviceSourceEngine::stop
2019-07-22 18:33:55.127 (D) DSPDeviceSourceEngine::gotoIdle
2019-07-22 18:33:55.127 (D) MainWindow::on_action_Exit_triggered
2019-07-22 18:33:55.127 (D) MainWindow::finalizeBeforeClose
2019-07-22 18:33:55.128 (D) MainWindow::savePresetSettings: preset [default | no name]
Segmentation fault (core dumped)
```
It seems clicking on Exit or Ctl-Q triggers the closing event before the exit action. Therefore only the closing event should be handled.

f4exb, 2019-07-22:
For the geometry in fact the layout is saved in the Preset object but not the geometry
https://doc.qt.io/qt-5/qmainwindow.html#saveState

rgetz, 2019-07-22:
Thanks for tracking down - looks like closeEvent has been blank since the initial git clone git://git.osmocom.org/sdrangelove.git

https://github.com/f4exb/sdrangel/blame/371bf9a5ec261f820e7457446540a9a828a37f4b/sdrbase/mainwindow.cpp#L467

and on_action_Exit_triggered() was introduced as part of the multi-device support re-work 2016
https://github.com/f4exb/sdrangel/commit/3e4a16330462042802d42246f7a977cfc30846c8

So, maybe I just just use CTRL-Q like everyone else obviously does :)

An additional 2 cents - although one may think that saving / restoring geometry is great - as someone who suspends laptops, de-docks from 2nd monitors, and then having applications start drawing to the 2nd (non-existent/not-connected) monitor, and me having to spend 2 min fighting with window managers - please don't do that, unless you want to handle all the weird corner cases... Not everyone runs on a static configuration.

f4exb, 2019-07-24:
Done in v4.11.2

closed 2019-07-24 by f4exb

rgetz, 2019-07-24:
confirmed that the problem doesn't seem to exist anymore - thanks

rgetz, 2019-07-29:
Since this change, I get alot of these:

```
2019-07-29 19:20:24.751 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-29 19:20:24.751 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-29 19:20:24.751 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-29 19:20:24.751 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-29 19:20:24.753 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-29 19:20:24.753 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-29 19:20:24.753 (W) Texture is not valid in the current context.
Texture has not been destroyed
2019-07-29 19:20:24.753 (W) Texture is not valid in the current context.
Texture has not been destroyed
```

only consistency is that it happens when closing with CTRL-Q or "X" windows button.