f4exb/sdrangel #380: sdriq file format

rgetz, 2019-07-20:
Currently SDRAngel uses a sdriq format, as described : 
https://github.com/f4exb/sdrangel/blob/dev/plugins/samplesink/filesink/readme.md

 S16LE I/Q samples. Thus there are 4 bytes per sample. I and Q values are 16 bit signed integers.

Are you interested in supporting other formats?

GNURadio Has it's file format : https://wiki.gnuradio.org/index.php/FAQ#What_is_the_file_format_of_a_file_sink.3F_How_can_I_read_files_produced_by_a_file_sink.3F

SigMF looks like it will replace it:
https://github.com/gnuradio/SigMF

MATLAB has a variety of file formats it can use. (that libmatio can parse)
https://www.mathworks.com/help/comm/ref/comm.basebandfilereader-system-object.html
https://www.mathworks.com/help/comm/ref/comm.basebandfilewriter-system-object.html

If we could make sure SDRangel could read/write other formats - it might help adoption. (and this is more of a question - if I go through the effort of adding one, and make the pull request, are you going to accept it? :)

Feel free to close this issue once you answer.

Thanks
-Robin

f4exb, 2019-07-21:
There already are two versions of the .sdriq format therefore this file converter was written: https://github.com/f4exb/sdrangel/tree/master/rescuesdriq

Among the formats you mention SigMF seems the more complex but also the most advanced and has the distinct advantage of making segmentation possible. It is also placed under the very liberal Creative Commons license. Even if you enrich the "rescuesdriq" program it will never be able to reconstruct possible changes in center frequency or sample rate because .sdriq meta data consists in a single header information. Therefore any change in center frequency and more importantly sample rate while recording can lead to inconsistent data. Because the segments are time stamped it also makes possible  to record data non-continuously. So this opens a lot of enhancements to the I/Q file processing.

I am not sure all sorts of formats should be supported by the core program. The "rescuesdriq" program could be beefed up to support many file conversions. I already faced the issue of wanting to process recordings that were done in complex floating point samples. So improving "rescuesdriq" has already reached my mind. "rescuesdriq" is written in Go language which might seem daunting to some. However Go language is not very complex (at least in this case) and is as versatile as Python regarding various O/S implementations and produces true binary executable files with generally very efficient code. Go was designed for devops which fits well the use case of file conversions.

So maybe the solution would be to drop .sdriq in favor of SigMF and ramp up "rescuesdriq" capabilities possibly renaming it to support many file format conversions including the then becoming old .sdriq

There is a little catch with SigMF in the sense that it seems to use full size 32 bit integers when in 32 bit mode where in SDRangel the samples are sized to 24 bits even if 32 bits integers are used. Not an issue but something to remember when scaling samples.

f4exb, 2019-07-21:
Because several recordings can be combined in a single SigMF archive this makes also possible (with some tweak to let the reader know what it is) to record MIMO data (synchronous streams). This will become interesting with the future implementation of true MIMO processing.

rgetz, 2019-07-22:
OK - let me poke some of the sigMF folks, and see what/when they were planning things.

As a side note - thoughts/opinions on including some example datafiles along with sdrangel?

f4exb, 2019-07-22:
The datafiles are usually big (if they are worth something) so I'd rather not store them along with the code.

rgetz, 2019-07-22:
Depends on what they are (obviously). We distribute "spectral" equivalents of various waveforms (LTE, QPSK, etc)- that allow I/Q testing of the Phy, and demod (for EVM testing), but they aren't really good for testing higher levels (packet level testing). The data is only 30-50ms long and just repeats.

The files we distribute as part of [IIO-oscilloscope](https://github.com/analogdevicesinc/iio-oscilloscope/tree/master/waveforms) are in the ~5Meg for each file (some much smaller) - so given that the sdrangel repo now (git clone) is 132M - I think it might be useful to have a few (CW, multi-tone, etc which is going to be pretty short/small); just to get end users going...

I haven't tested things - but I assume things support cyclic buffers?

MathWorks exposes things as ["transmit repeat"](https://www.mathworks.com/help/supportpkg/xilinxzynqbasedradio/ug/repeated-waveform-transmitter.html) _You can create any arbitrary waveform at the baseband in MATLAB® and transmit it repeatedly from the hardware. You are not limited by the Ethernet link or the processing speed of the host computer for signal transmission over the air._

It does this by using the _cyclic_ feature of iio. When calling [iio_device_create_buffer](https://analogdevicesinc.github.io/libiio/group__Buffer.html#ga6caadf077c112ae55a64276aa24ef832) just set the _cyclic_ bit - then you don't need to worry about USB transactions (but you don't get to change the waveform either - it's mean for high datarate, repeating waveforms) under 8Mbytes in length).

gaspar1987, 2019-09-10:
hello I have a cousis with enreigement plays repaired on new version sdrangel
[image]

gaspar1987, 2019-09-10:
steps not

gaspar1987, 2019-09-10:
[image]

f4exb, 2019-09-10:
@gaspar1987 Please comment appropriately. This has nothing to do with the subject.

dh5ym, 2020-03-17:
Just playing 16bit IQ files without the need of any header would be great. Entering the sample rate manually is a possibility.

f4exb, 2020-03-28:
Use this utility: https://github.com/f4exb/sdrangel/tree/master/rescuesdriq to build an arbitraty header.

dh5ym, 2020-03-28:
Thanks a lot! I already found it and with a small change it worked for me. However the header is not needed in most of the cases and converting every IQ file could be avoided if there would be a checkbox to ignore missing header information. However i am happy now. Thanks a lot for this great software!

f4exb, 2020-05-20:
This looks like to be the place for SigMF support: https://github.com/deepsig/libsigmf

f4exb, 2020-06-05:
For this functionality to be really useful a new input device plugin is to be created to support input form SigMF files. Hence the 'feature' tag. It is easier and cleaner to create a new plugin than trying to update the existing FileInput.

f4exb, 2020-06-18:
This takes part of the more general file recording refactoring project just created. It will be split in two parts:
  - The SigMF file input (device) plugin -> goes with v5.7.0
  - The SigMF file sink (channel) -> to be planned

Thus v5.7.0 for this ticket is removed.

f4exb, 2020-07-20:
The complete solution is available on v5.8.1 now

closed 2020-07-20 by f4exb