f4exb/sdrangel #382: RX - TX - fixed delta F

drom, 2019-07-23:
Thank you for the very useful program!
Is any way to lock TX (PlutoSDR) and RX (RTL Dongle) to have a fixed frequency offset while I an changing RX frequency?

f4exb, 2019-07-23:
You have to use an ancillary program that will run in the background and listen for changes in SDRangel to orchestrate both parts synchronization. A few examples of such programs written in Python can be found in the scriptsapi sectiion: https://github.com/f4exb/sdrangel/tree/master/scriptsapi More significantly the ptt_active.py that could be made to do what you expect with little change.

drom, 2019-07-24:
Looks good. Thank you for examples. I will try writing my plugin.

f4exb, 2019-07-24:
Since the question is answered I am closing the issue. You may open another ticket if you find something is missing in the API.

closed 2019-07-24 by f4exb

drom, 2019-07-25:
One aspect of Full-Transceiver mode operation is that RX and TX not just locked to a specific Delta frequency (that can be very large like 144MHz -- 435MHz) but this delta can be tuned by special [RIT](https://www.amateur-radio-wiki.net/index.php?title=Operating_procedures#RIT) (+-100KHz) knob while operating.