f4exb/sdrangel #383: DATV Demodulator Broken

KR0SIV, 2019-07-24:
I am nolonger able to demod DATV (DVB-S) I'm using the same setup I have in the past using both a pluto and LimeSDR Mini as the transmitter to no avail.

[image]

FEC 7/8 and 550ks/s

f4exb, 2019-07-24:
Issue seems related to FEC 7/8. Other FECs work (test with GNUradio dvbs_tx example):

[image]

Different symbol rates do not help.

f4exb, 2019-07-24:
Oh-oooh this is what I see in the log when switching to 7/8 and since it is not prefixed I think this comes from leandvb:
`Code rate not implemented; proceeding anyway`

I have redesigned the code rate limits according more or less to ETSI standard. I bet 7/8 was not supported before. Since leandvb does not support it it should simply be be removed from the list.

f4exb, 2019-07-24:
Actually issue is in my code because 7/8 should be supported. Here's the excerpt from leandvb dvb.h:
```c++
    // EN 300 421, section 4.4.3 Inner coding
    uint32_t pX, pY;
    switch (rate)
    {
    case FEC12:
        pX = 0x1; // 1
        pY = 0x1; // 1
        break;
    case FEC23:
    case FEC46:
        pX = 0xa; // 1010  (Handle as FEC4/6, no half-symbols)
        pY = 0xf; // 1111
        break;
    case FEC34:
        pX = 0x5; // 101
        pY = 0x6; // 110
        break;
    case FEC56:
        pX = 0x15; // 10101
        pY = 0x1a; // 11010
        break;
    case FEC78:
        pX = 0x45; // 1000101
        pY = 0x7a; // 1111010
        break;
    default:
        //fail("Code rate not implemented");
        // For testing DVB-S2 constellations.
        fprintf(stderr, "Code rate not implemented; proceeding anyway\n");
        pX = pY = 1;
    }
```

f4exb, 2019-07-24:
7/8 missing from the string to enum converter...

f4exb, 2019-07-24:
This is it! Fix is on dev branch.

f4exb, 2019-07-27:
Fixed in v4.11.3

closed 2019-07-27 by f4exb