f4exb/sdrangel #384: cmake failures when SerialDV not installed

rgetz, 2019-07-24:
on master & dev.

With the instructions at:
https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

I get:

```
CMake Error at cmake/Modules/FindSerialDV.cmake:29 (message):
  Could not find SerialDV
Call Stack (most recent call first):
  CMakeLists.txt:367 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/rgetz/github/sdrangel/build/CMakeFiles/CMakeOutput.log".
```

I don't have a AMBE3000 device - so I shouldn't need to install the library. correct?

?

f4exb, 2019-07-24:
Instructions are not up to date. Serialdv is now a requirement.

rgetz, 2019-07-24:
Ok - that sort of sucks, but I get it's a design decision you made (and since you are doing all the work - you can do that).

Thanks

closed 2019-07-24 by rgetz

f4exb, 2019-07-24:
Why does it suck? There are already many dependencies anyway to get a fully fledged version and btw the compile instructions says you must have it. Making it optional would require to make the AMBE interface optional which then would really suck.

magnum88, 2019-10-21:
in 
/opt/build/sdrangel/cmake/Modules
edit FindSerialDV.cmake
change line 13 by
  NAMES ${LIBSERIAL_NAMES}

and after:
sudo cp /opt/build/serialDV/dvcontroller.h /opt/install/serialdv/include/serialdv/
sudo cp /opt/build/serialDV/serialdv_export.h /opt/install/serialdv/include/serialdv/

and after

cd /opt/build
git clone https://github.com/f4exb/sdrangel.git
cd sdrangel
mkdir build; cd build
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DMIRISDR_DIR=/opt/install/libmirisdr \
-DAIRSPY_DIR=/opt/install/libairspy \
-DAIRSPYHF_DIR=/opt/install/libairspyhf \
-DBLADERF_DIR=/opt/install/libbladeRF \
-DHACKRF_DIR=/opt/install/libhackrf \
-DRTLSDR_DIR=/opt/install/librtlsdr \
-DLIMESUITE_DIR=/opt/install/LimeSuite \
-DIIO_DIR=/opt/install/libiio \
-DPERSEUS_DIR=/opt/install/libperseus \
-DXTRX_DIR=/opt/install/xtrx-images \
-DSOAPYSDR_DIR=/opt/install/SoapySDR \
-DCM256CC_DIR=/opt/install/cm256cc \
-DDSDCC_DIR=/opt/install/dsdcc \
-DSERIALDV_DIR=/opt/install/serialdv \
-DMBE_DIR=/opt/install/mbelib \
-DCODEC2_DIR=/opt/install/codec2 \
-DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
make -j4 install