f4exb/sdrangel #385: Morse key bindings

drom, 2019-07-24:
Is any way to bind specific keys on the keyboard to the `.` and `-` and use it as morse keyer?

f4exb, 2019-07-24:
No

f4exb, 2019-07-24:
To elaborate more on this: I think I went through this a long time ago and resorted to abandon it. Anyway this is not really practical. It would be more interesting to be able to use a real keyer. This also rises the issue of having a standard interface. I think eventually you would have to use a program that interfaces with your sound-card or using a virtual sound card to convert to an actual tone modulation and simply use it as input. Which means from the SDRangel point of view that you already have it.

drom, 2019-07-24:
I have a very simple keyer make of USB keyboard. It sends key-`Z` for `-` and key-`X` for `.`.
It is extremely simple to make and produces a very good resulting experience.
Soundcard is way more cumbersome IMHO.

f4exb, 2019-07-24:
There's an issue with tracking key events if I remember right but I can have a second look.

drom, 2019-07-24:
In fact, you already implemented 90% features, needed to implement automatic keyer: speed, tone, modulator, text to code, etc.
Here is an example of a simple script implementing the rest: https://observablehq.com/@drom/morse-code-keyer

drom, 2019-07-24:
I have upgraded my QO-100 satellite station setup using SDRangel and PlutoSDR today. Works great!
CW keyer feature would be a big plus!

drom, 2019-07-24:
If it works, I can publish my DIY keyboard to CW keyer hack ;)

f4exb, 2019-07-25:
The main issue with handling key events from a widget in Qt is that normally it has to have the focus. I remember running into the issue when putting in place the "commands" feature. I had to add a keyboard toggle button to activate/de-activate the binding even if the focus is set elsewhere. So if activating through a toggle is acceptable this can probably be done.

drom, 2019-07-27:
Does focus need to be on the SSB modulator window?
In a real HAM contest situation, the focus most probably is on the R0 waterfall.

f4exb, 2019-07-27:
It does not with the commands key bindings so I suppose the same can be done. However by applying the same design only one modulator can get the bindings at any time. I think this would not be an issue.

f4exb, 2019-07-29:
In fact there is an issue with the focus because only the main window can keep the bindings even if the focus is set elsewhere.

drom, 2019-07-29:
ok, will the CW keyer work if you bind keys to the main window?

f4exb, 2019-07-29:
The signal has to come from the main window then maybe it is possible to register any slot. I have to elaborate more on this to be more generic than just the CW keyer. I think the point is that there can be only one event filter installed on the main window and everything has to be dispatched from there.

f4exb, 2019-07-30:
Working version in dev however I have found impractical not to have proper audio feedback. The feedback from transmission is significantly delayed due to the necessary buffering on commodity hardware. This makes keying and relying on the audio from the transmitter next to impossible. An optional audio feedback could be put in place for the audio modulators. This has probably to be generic for all types of audio input.

drom, 2019-07-30:
Agree that proper, low latency audio feedback is essential.

f4exb, 2019-07-30:
The audio system has about half a second latency so audio monitoring from the software does not help anyway. So this is pretty useless if you don't use it with a proper keyer that has its own low latency audio feedback. Audio monitoring may still be useful in some other cases so I am going to keep it but the morse keyer from the keyboard is pretty much a fail.

drom, 2019-07-31:
I am getting very short latency audio feedback and realistic timing for the keyer withing my Web application.
I could send formed dit-dash events over HTTP/REST API. What do you think?

f4exb, 2019-07-31:
It is not working if you use a standard keyboard or just the keyboard of the PC if this is a laptop. I am curious on how you manage to trigger the dit-dash events from the API because there is no endpoint in the API to trigger these events.

drom, 2019-07-31:
We don't have API right now. But I could send dit-dash events if you add API.
Right now I getting feedback from my Web program directly.
Here is an example: https://observablehq.com/@drom/morse-code-keyer

drom, 2019-07-31:
Or as an alternative, I could fully form CW Audio stream and send it over HTTP API.
If we would have such API.

f4exb, 2019-07-31:
API is not meant for real time so I will intentionally limit it to control functions. There are many ways to bring in audio via audio devices possibly virtual or UDP with the UDP source.

Your keyer is what I called a "proper" keyer managing the audio feedback itself. This is not necessarily hardware and can be software like yours. For audio feedback within SDRangel I am using its AudioFifo class which seems to introduce latency by design. I Don't know if it is due to Qt or pulseaudio or anything else. It is used all over the place so I do not intend to touch it. 

If this meets your requirements let's keep it like that and move on to something else.

drom, 2019-07-31:
You are totally right!
I manage to connect my Web Keyer with SDRangel when it runs on the same machine via ALSA.
* WebApp -> `alsa_output.pci-0000_00_1f.3.analog-stereo`
* `alsa_output.pci-0000_00_1f.3.analog-stereo.monitor` -> SDRangel (T1 / SSB modulator)

f4exb, 2019-08-01:
FYI for fade in / fade out I use this function on the [0.0, 1.0] interval:  f(x) = 6.0x^5 - 15.0x^4 + 10.0x^3 that has first and second derivatives null at 0.0 and 1.0. See: https://github.com/f4exb/sdrangel/blob/master/sdrbase/util/stepfunctions.h

drom, 2019-08-01:
Interesting. Is it "Raised Cosine Filter" approximation of some sort?

drom, 2019-08-01:
Effect of Keying Waveform on CW Bandwidth, Jim Proctor, KE3HO
https://www.w8ji.com/cw%20bandwidth%20analysis.htm

f4exb, 2019-08-01:
No cosine there it is a simple polynomial (well in x⁵) and if you calculate its first and second derivative you realize both are zero at 0.0 and 1.0 therefore if you shape your step up or step down with this function you get a really smooth transition and therefore very clean spectrum.

f4exb, 2019-08-10:
Implemented in v4.11.5

closed 2019-08-10 by f4exb