f4exb/sdrangel #386: Compilations warnings

mnhauke, 2019-07-25:
There is a is a "-Wreturn-type" warning in the latest release (4.11.2) when compiling with gcc ...
```
[  107s] /home/abuild/rpmbuild/BUILD/sdrangel-4.11.2/sdrbase/webapi/webapirequestmapper.cpp: In member function 'bool WebAPIRequestMapper::validateAMBEDevices(SWGSDRangel::SWGAMBEDevices&, QJsonObject&)':
[  107s] /home/abuild/rpmbuild/BUILD/sdrangel-4.11.2/sdrbase/webapi/webapirequestmapper.cpp:2750:1: warning: control reaches end of non-void function [-Wreturn-type]
[  107s]  }
```

... which then results in an error by one of the RPM post-build checks on OBS while building the  openSUSE packages
```
[  398s] I: Program returns random data in a function
[  398s] E: sdrangel no-return-in-nonvoid-function /home/abuild/rpmbuild/BUILD/sdrangel-4.11.2/sdrbase/webapi/webapirequestmapper.cpp:2750
```

I could overide the CXXFLAGS (-Wno-return-type) to workaround this issue but it would be nice to get it fixed upstream.

f4exb, 2019-07-25:
It is a real issue the function should return true or false in all cases.

f4exb, 2019-07-27:
Fixed in v4.11.3

closed 2019-07-27 by f4exb