f4exb/sdrangel #387: Building on Linux

csylvain, 2019-07-25:
have all the pre-built dependency pieces in /opt/install .. had to provide "hint" (-Dxxx) for various dependencies (otherwise CMake would not find them). there appear to be /opt/install/... paths set for CMake but it does not seem to use them. even with the hints, SoapySDR appears not to be found.
```
$ cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv -DLIBDSDCC_INCLUDE_DIR=/opt/install/dsdcc -DSOAPYSDR_INCLUDE_DIR=/opt/install/SoapySDR -DLIBSOAPYSDR_INCLUDE_DIR=/opt/install/SoapySDR ..
.
.
-- Checking for module 'soapysdr>=0.4.0'
--   No package 'soapysdr' found
.
.
$ ls /opt/install
cm256cc  codec2  dsdcc  libairspy  librtlsdr  mbelib  serialdv  SoapySDR

f4exb, 2019-07-25:
I am not sure which instructions you are following but if you take the Wiki as a reference (you should): https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux then you end up with this (with all options):
`cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON -DMIRISDR_DIR=/opt/install/libmirisdr -DAIRSPY_DIR=/opt/install/libairspy -DAIRSPYHF_DIR=/opt/install/libairspyhf -DBLADERF_DIR=/opt/install/libbladeRF -DHACKRF_DIR=/opt/install/libhackrf -DRTLSDR_DIR=/opt/install/librtlsdr -DLIMESUITE_DIR=/opt/install/LimeSuite -DIIO_DIR=/opt/install/libiio -DCM256CC_DIR=/opt/install/cm256cc -DDSDCC_DIR=/opt/install/dsdcc -DSERIALDV_DIR=/opt/install/serialdv -DMBE_DIR=/opt/install/mbelib -DCODEC2_DIR=/opt/install/codec2 -DPERSEUS_DIR=/opt/install/libperseus -DXTRX_DIR=/opt/install/xtrx-images -DSOAPYSDR_DIR=/opt/install/SoapySDR -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..`

Which is not exactly the same as yours regarding SoapySDR.

`xxx_LIBRARIES` and `xxx_INCLUDE_DIR` were dropped a long time ago since cmake does not seem to check for their actual existence but it can do it if it has to compute the values from the root directory of the dependency. Therefore the `xxx_DIR` were introduced.

csylvain, 2019-07-25:
O I C .. i was (almost) following the instructions at
https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux and
managed to miss the very very long 'cmake' line. i'd been improvising based
on what i could see in the CMakeCache.txt in the build dir for everything
after the -DMIRISDR=... part. my bad.

may i suggest writing that line like this instead?

> cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON
> -DMIRISDR_DIR=/opt/install/libmirisdr  \

-DAIRSPY_DIR=/opt/install/libairspy -DAIRSPYHF_DIR=/opt/install/libairspyhf
> -DBLADERF_DIR=/opt/install/libbladeRF \

-DHACKRF_DIR=/opt/install/libhackrf -DRTLSDR_DIR=/opt/install/librtlsdr
> -DLIMESUITE_DIR=/opt/install/LimeSuite \

-DIIO_DIR=/opt/install/libiio -DCM256CC_DIR=/opt/install/cm256cc
> -DDSDCC_DIR=/opt/install/dsdcc \

-DSERIALDV_DIR=/opt/install/serialdv -DMBE_DIR=/opt/install/mbelib
> -DCODEC2_DIR=/opt/install/codec2 \

-DPERSEUS_DIR=/opt/install/libperseus -DXTRX_DIR=/opt/install/xtrx-images \

-DSOAPYSDR_DIR=/opt/install/SoapySDR
> -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
>

just as valid for the shell and makes sure no human eyeballs manage to miss
all the important definitions!

thanks!

csylvain, 2019-07-25:
still seeing this, though:

--   No package 'soapysdr' found
> CMake Error at cmake/Modules/FindSoapySDR.cmake:4 (file):
>   file failed to open for reading (No such file or directory):
>
>     /opt/install/SoapySDR/SoapySDR/Version.h
> Call Stack (most recent call first):
>   cmake/Modules/FindSoapySDR.cmake:42 (_SOAPY_SDR_GET_ABI_VERSION)
>   CMakeLists.txt:400 (find_package)
>
>
> -- Found SoapySDR: /opt/install/SoapySDR,
> /usr/lib/x86_64-linux-gnu/libSoapySDR.so
>

and getting

-- Asking qmake for QT_PLUGINS_DIR and got
/usr/lib/x86_64-linux-gnu/qt5/plugins
-- Asking qmake for QT_IMPORTS_DIR and got
/usr/lib/x86_64-linux-gnu/qt5/imports
-- CPack generators:
-- Configuring incomplete, errors occurred!
See also
"/home/csylvain/Software/sdrangel/build/CMakeFiles/CMakeOutput.log".
make: *** [Makefile:2780: cmake_check_build_system] Error 1

at the very end.

csylvain, 2019-07-25:
the SoapySDR error is caused by a 'include' being missing from the file
path for Version.h

$ ls /opt/install/SoapySDR/include/SoapySDR/
> Config.h     ConverterPrimitives.hpp  Device.hpp  Formats.h    Logger.hpp
>   Registry.hpp  Types.h    Version.hpp
> Config.hpp   ConverterRegistry.hpp    Errors.h    Formats.hpp  Modules.h
>  Time.h        Types.hpp
> Constants.h  Device.h                 Errors.hpp  Logger.h     Modules.hpp
>  Time.hpp      Version.h
>

csylvain, 2019-07-25:
patch for FindSoapySDR.cmake

diff --git a/cmake/Modules/FindSoapySDR.cmake
b/cmake/Modules/FindSoapySDR.cmake
index 77fe95ec9..226f402b8 100644
--- a/cmake/Modules/FindSoapySDR.cmake
+++ b/cmake/Modules/FindSoapySDR.cmake
@@ -1,7 +1,7 @@
 if(NOT SOAPYSDR_FOUND)

   function(_SOAPY_SDR_GET_ABI_VERSION VERSION SOAPY_SDR_INCLUDE_DIR)
-    file(READ "${SOAPY_SDR_INCLUDE_DIR}/SoapySDR/Version.h" version_h)
+    file(READ "${SOAPY_SDR_INCLUDE_DIR}/include/SoapySDR/Version.h"
version_h)
     string(REGEX MATCH "\\#define SOAPY_SDR_ABI_VERSION
\"([0-9]+\\.[0-9]+(-[A-Za-z0-9]+)?)\"" SOAPY_SDR_ABI_VERSION_MATCHES
"${version_h}")
     if(NOT SOAPY_SDR_ABI_VERSION_MATCHES)
       message(FATAL_ERROR "Failed to extract version number from
Version.h")

csylvain, 2019-07-25:
looking in the wrong place for the SerialDV include files:

[ 20%] Building CXX object
> sdrbase/CMakeFiles/sdrbase.dir/ambe/ambeengine.cpp.o
> In file included from
> /home/csylvain/Software/sdrangel/sdrbase/ambe/ambeengine.cpp:43:
> /home/csylvain/Software/sdrangel/sdrbase/ambe/ambeworker.h:27:10: fatal
> error: dvcontroller.h: No such file or directory
>  #include "dvcontroller.h"
>           ^~~~~~~~~~~~~~~~
> compilation terminated.
> make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:142:
> sdrbase/CMakeFiles/sdrbase.dir/ambe/ambeengine.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> In file included from
> /home/csylvain/Software/sdrangel/sdrbase/ambe/ambeworker.cpp:24:
> /home/csylvain/Software/sdrangel/sdrbase/ambe/ambeworker.h:27:10: fatal
> error: dvcontroller.h: No such file or directory
>  #include "dvcontroller.h"
>           ^~~~~~~~~~~~~~~~
> compilation terminated.
> make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:155:
> sdrbase/CMakeFiles/sdrbase.dir/ambe/ambeworker.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:533:
> sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
> make: *** [Makefile:152: all] Error 2
>

csylvain, 2019-07-25:
>
> -- Found LibMbe: /opt/install/mbelib/include,
> /opt/install/mbelib/lib/libmbe.so
> -- Found libserialdv: /opt/install/serialdv, /opt/install/serialdv/lib
>

libserialdv should read: /opt/install/serialdv/include/serialdv,
/opt/install/serialdv/lib/libserialdv.so

fixed it. diff attached.

On Thu, Jul 25, 2019 at 1:41 PM Christopher Sylvain <chris.sylvain@gmail.com>
wrote:

> looking in the wrong place for the SerialDV include files:
>
> [ 20%] Building CXX object
>
>
> [...]

csylvain, 2019-07-25:
nope. since source references a bare "dvcontroller.h" and not
"serialdv/dvcontroller.h", then the lines in FindSerialDV.cmake need to be:
NAMES dvcontroller.h
PATHS ${SERIALDV_DIR}/include/serialdv

still have the problem that the system SoapySDR is found and not the one in
/opt/install/SoapySDR

On Thu, Jul 25, 2019 at 2:24 PM Christopher Sylvain <chris.sylvain@gmail.com>
wrote:

> -- Found LibMbe: /opt/install/mbelib/include,
>
> libserialdv should read: /opt/install/serialdv/include/serialdv,
> /opt/install/serialdv/lib/libserialdv.so
>
> fixed it. diff attached.
>
> On Thu, Jul 25, [...]

csylvain, 2019-07-25:
had the problem that the system Codec2 is found and not the one in
/opt/install/codec2

made FindCodec2.cmake prefer the one in /opt/install instead of the system
one

diff --git a/cmake/Modules/FindCodec2.cmake b/cmake/Modules/FindCodec2.cmake
> index 8e31a0764..81a05a364 100644
> --- a/cmake/Modules/FindCodec2.cmake
> +++ b/cmake/Modules/FindCodec2.cmake
> @@ -3,8 +3,9 @@ PKG_CHECK_MODULES(PC_CODEC2 "codec2")
>
>  FIND_PATH(CODEC2_INCLUDE_DIR
>      NAMES codec2/codec2.h
> -    HINTS ${PC_CODEC2_INCLUDE_DIR}
> +    HINTS ${CODEC2_DIR}/include
>            ${CMAKE_INSTALL_PREFIX}/include
> +          ${PC_CODEC2_INCLUDE_DIR}
>      PATHS ${CODEC2_DIR}/include
>            /usr/local/include
>            /usr/include
> @@ -12,9 +13,11 @@ FIND_PATH(CODEC2_INCLUDE_DIR
>
>  FIND_LIBRARY(CODEC2_LIBRARIES
>      NAMES codec2 libcodec2
> -    HINTS ${PC_CODEC2_LIBDIR}
> +    HINTS ${CODEC2_DIR}/lib
> +          ${CODEC2_DIR}/lib64
>            ${CMAKE_INSTALL_PREFIX}/lib
>            ${CMAKE_INSTALL_PREFIX}/lib64
> +          ${PC_CODEC2_LIBDIR}
>      PATHS ${CODEC2_DIR}/lib
>            ${CODEC2_DIR}/lib64
>            /usr/local/lib
> @@ -33,4 +36,4 @@ endif()
>
>  INCLUDE(FindPackageHandleStandardArgs)
>  FIND_PACKAGE_HANDLE_STANDARD_ARGS(CODEC2 DEFAULT_MSG CODEC2_LIBRARIES
> CODEC2_INCLUDE_DIR)
> -MARK_AS_ADVANCED(CODEC2_LIBRARIES CODEC2_INCLUDE_DIR)
> \ No newline at end of file
> +MARK_AS_ADVANCED(CODEC2_LIBRARIES CODEC2_INCLUDE_DIR)
>

going to try the same trick in FindSoapySDR now

On Thu, Jul 25, 2019 at 2:28 PM Christopher Sylvain <chris.sylvain@gmail.com>
wrote:

> nope. since source references a bare "dvcontroller.h" and not
> "serialdv/dvcontroller.h", then the lines in FindSerialDV.cmake need to be:
> NAMES dvcontroller.h
> PATHS ${SERIALDV_DIR}/include/serialdv [...]

csylvain, 2019-07-25:
yes. this makes the FindSoapySDR prefer the one in /opt/install and not the
system

diff --git a/cmake/Modules/FindSoapySDR.cmake
> b/cmake/Modules/FindSoapySDR.cmake
> index 77fe95ec9..1bb274e26 100644
> --- a/cmake/Modules/FindSoapySDR.cmake
> +++ b/cmake/Modules/FindSoapySDR.cmake
> @@ -1,7 +1,7 @@
>  if(NOT SOAPYSDR_FOUND)
>
>    function(_SOAPY_SDR_GET_ABI_VERSION VERSION SOAPY_SDR_INCLUDE_DIR)
> -    file(READ "${SOAPY_SDR_INCLUDE_DIR}/SoapySDR/Version.h" version_h)
> +    file(READ "${SOAPYSDR_DIR}/include/SoapySDR/Version.h" version_h)
>      string(REGEX MATCH "\\#define SOAPY_SDR_ABI_VERSION
> \"([0-9]+\\.[0-9]+(-[A-Za-z0-9]+)?)\"" SOAPY_SDR_ABI_VERSION_MATCHES
> "${version_h}")
>      if(NOT SOAPY_SDR_ABI_VERSION_MATCHES)
>        message(FATAL_ERROR "Failed to extract version number from
> Version.h")
> @@ -15,6 +15,8 @@ if(NOT SOAPYSDR_FOUND)
>
>      find_path (SOAPYSDR_INCLUDE_DIR
>        NAMES SoapySDR/Version.h
> +      HINTS ${SOAPYSDR_DIR}/include
> +            ${CMAKE_INSTALL_PREFIX}/include
>        PATHS ${SOAPYSDR_DIR}/include
>        ${LIBSOAPYSDR_PKG_INCLUDE_DIRS}
>        /usr/include
> @@ -23,8 +25,10 @@ if(NOT SOAPYSDR_FOUND)
>
>      find_library (SOAPYSDR_LIBRARY
>        NAMES SoapySDR
> -      HINTS ${CMAKE_INSTALL_PREFIX}/lib
> -      ${CMAKE_INSTALL_PREFIX}/lib64
> +      HINTS ${SOAPYSDR_DIR}/lib
> +            ${SOAPYSDR_DIR}/lib64
> +            ${CMAKE_INSTALL_PREFIX}/lib
> +            ${CMAKE_INSTALL_PREFIX}/lib64
>        PATHS ${SOAPYSDR_DIR}/lib
>        ${SOAPYSDR_DIR}/lib64
>        ${LIBSOAPYSDR_PKG_LIBRARY_DIRS}
>

need to make the same change to FindLibRTLSDR.cmake (again: finding the
system one instead of preferring the one in /opt/install)

On Thu, Jul 25, 2019 at 2:46 PM Christopher Sylvain <chris.sylvain@gmail.com>
wrote:

> had the problem that the system Codec2 is found and not the one in
> /opt/install/codec2
>
> made FindCodec2.cmake prefer the one in /opt/install instead of the system
> one
>
> diff --git [...]

csylvain, 2019-07-25:
all the optional dependencies put in /opt/install need to supersede the
same packages installed in the system. system (distro) versions may very
well be older. since i've put the time and effort into putting the latest
into /opt/install -- they should always be preferred.

csylvain, 2019-07-25:
did not edit FindLibAIRSPYHF.cmake .. i've got an Airspy R2, so i've edited
FindLibAIRSPY.cmake. the other one should be edited likewise.

diff --git a/cmake/Modules/FindLibAIRSPY.cmake
> b/cmake/Modules/FindLibAIRSPY.cmake
> index 525d4f035..04ff271dc 100644
> --- a/cmake/Modules/FindLibAIRSPY.cmake
> +++ b/cmake/Modules/FindLibAIRSPY.cmake
> @@ -1,17 +1,21 @@
>  if(NOT LIBAIRSPY_FOUND)
>
>    pkg_check_modules (LIBAIRSPY_PKG libairspy)
> -
> -  find_path(LIBAIRSPY_INCLUDE_DIR
> +
> +  find_path(LIBAIRSPY_INCLUDE_DIR
>      NAMES libairspy/airspy.h
> +    HINTS ${AIRSPY_DIR}/include
> +          ${CMAKE_INSTALL_PREFIX}/include
>      PATHS ${AIRSPY_DIR}/include
>            ${LIBAIRSPY_PKG_INCLUDE_DIRS}
>            /usr/include
>            /usr/local/include
>    )
>
> -  find_library(LIBAIRSPY_LIBRARIES
> +  find_library(LIBAIRSPY_LIBRARIES
>      NAMES airspy
> +    HINTS ${AIRSPY_DIR}/lib
> +          ${CMAKE_INSTALL_PREFIX}/lib
>      PATHS ${AIRSPY_DIR}/lib
>            ${LIBAIRSPY_PKG_LIBRARY_DIRS}
>            /usr/lib
>

On Thu, Jul 25, 2019 at 3:00 PM Christopher Sylvain <chris.sylvain@gmail.com>
wrote:

> all the optional dependencies put in /opt/install need to supersede the
> same packages installed in the system. system (distro) versions may very
> well be older. since i've put the time and effort into putting the latest
> into /opt/install -- they should always be preferred.
>
>
>

f4exb, 2019-07-25:
This:
```
 -    file(READ "${SOAPY_SDR_INCLUDE_DIR}/SoapySDR/Version.h" version_h)
 +    file(READ "${SOAPYSDR_DIR}/include/SoapySDR/Version.h" version_h)
```
and the like replacing xxx_INCLUDE_DIR bu xxx_DIR would break the build with external libraries (Windows, Debian packages) so I am certainly not going to apply it.

On the other hand I take your suggestions to split lines in the long command in the Wiki and change the search order.

csylvain, 2019-07-25:
i only know what ultimately worked here. the Find routines were preferring
the system-wide installs of dependencies over those installed in
/opt/install. i think it best there be a preference for /opt/install.
whatever else needs to change, i am uncertain what they would be other than
perhaps to delay the Version.h check in the FindSoapySDR until after the
system-wide versus /opt/install selection is made.

 - 73 -

f4exb, 2019-07-25:
I am now almost sure you are using an older (> 7 months) version because the xxx_DIR always come first in the fond modules find_path PATHs

csylvain, 2019-07-25:
sigh. i did a 'git pull' the day before posting this new issue.
$ ./sdrangel -v
SDRangel 4.11.2
$ git log
commit c149d25b0c5484da16470bf99659437bf6f564f8 (HEAD -> master, tag:
v4.11.2, origin/master, origin/HEAD)
Author: f4exb <f4exb06@gmail.com>
Date:   Tue Jul 23 23:44:03 2019 +0200

    Updated version and changelogs

f4exb, 2019-07-25:
Then how do you explain this? https://github.com/f4exb/sdrangel/blob/master/cmake/Modules/FindLibAIRSPY.cmake#L7

csylvain, 2019-07-25:
now look at the diff i sent:

diff --git a/cmake/Modules/FindLibAIRSPY.cmake
> b/cmake/Modules/FindLibAIRSPY.cmake
> index 525d4f035..04ff271dc 100644
> --- a/cmake/Modules/FindLibAIRSPY.cmake
> +++ b/cmake/Modules/FindLibAIRSPY.cmake
> @@ -1,17 +1,21 @@
>  if(NOT LIBAIRSPY_FOUND) [...]

as you can see, i had that file. and noticed the system-wide libairspy was
being picked in the output of the cmake. then i added HINTS. now it
actually picks the stuff in /opt/install/libairspy instead
of /usr/lib/x86_64-linux-gnu/ and  /usr/include/libairspy or
/usr/local/include/libairspy ..

okay?

f4exb, 2019-07-25:
I see... but then why having HINTS at all? FindSerialDV.cmake does not have it for example.

csylvain, 2019-07-25:
i am not certain why. i know at this time without HINTS, cmake picks the
system-wide stuff. which in the case of codec2, failed completely because
the distro's codec2 did not have two functions called by sdrangel. but the
fresh new codec2 i put in /opt/install this morning does.

for SerialDV, i do not have a system-wide install of that. so i modeled the
FindSerialDV after the FindLibMbe which had a similar directory structure
for the include files.

in the sdrangel source of ambeengine, it references "dvcontroller.h",
where, by rights, it should reference "serialdv/dvcontroller.h" if you
follow how the serialdv include files get place into the filesystem tree.

i did play with the variable names, but without making the change to the
ambeengine source i just mention, it takes the following to successfully
compile ambeengine without bombing out of the make processing:

diff --git a/cmake/Modules/FindSerialDV.cmake
> b/cmake/Modules/FindSerialDV.cmake
> index c23e3d008..b3fab70c7 100644
> --- a/cmake/Modules/FindSerialDV.cmake
> +++ b/cmake/Modules/FindSerialDV.cmake
> @@ -3,14 +3,14 @@
>  find_path(LIBSERIALDV_INCLUDE_DIR
>    NAMES dvcontroller.h
>    PATHS ${SERIALDV_DIR}/include/serialdv
> -        /usr/include/serialdv
> -        /usr/local/include/serialdv
> +        /usr/include
> +        /usr/local/include
>  )
>
> -set(LIBSERIAL_NAMES ${LIBSERIAL_NAMES} serialdv libserialdv)
> +set(LIBSERIALDV_NAMES ${LIBSERIALDV_NAMES} serialdv libserialdv)
>
>  find_library(LIBSERIALDV_LIBRARY
> -  NAMES serialdv
> +  NAMES ${LIBSERIALDV_NAMES}
>    PATHS ${SERIALDV_DIR}/lib
>          /usr/lib
>          /usr/local/lib
> @@ -21,13 +21,13 @@ if (LIBSERIALDV_INCLUDE_DIR AND LIBSERIALDV_LIBRARY)
>  endif (LIBSERIALDV_INCLUDE_DIR AND LIBSERIALDV_LIBRARY)
>
>  if (LIBSERIALDV_FOUND)
> -    if (NOT SerialDV_FIND_QUIETLY)
> +    if (NOT LibSerialDV_FIND_QUIETLY)
>          message (STATUS "Found libserialdv: ${LIBSERIALDV_INCLUDE_DIR},
> ${LIBSERIALDV_LIBRARY}")
> -    endif (NOT SerialDV_FIND_QUIETLY)
> +    endif (NOT LibSerialDV_FIND_QUIETLY)
>  else (LIBSERIALDV_FOUND)
> -    if (SerialDV_FIND_REQUIRED)
> +    if (LibSerialDV_FIND_REQUIRED)
>          message (FATAL_ERROR "Could not find SerialDV")
> -    endif (SerialDV_FIND_REQUIRED)
> +    endif (LibSerialDV_FIND_REQUIRED)
>  endif (LIBSERIALDV_FOUND)
>
>  mark_as_advanced(LIBSERIALDV_INCLUDE_DIR LIBSERIALDV_LIBRARY)
>

in case that diff is a bit mangled, here's exactly my full
FindSerialDV.cmake file:

# Find serialDV
> find_path(LIBSERIALDV_INCLUDE_DIR
>   NAMES dvcontroller.h
>   PATHS ${SERIALDV_DIR}/include/serialdv
>         /usr/include
>         /usr/local/include
> )
> set(LIBSERIALDV_NAMES ${LIBSERIALDV_NAMES} serialdv libserialdv)
> find_library(LIBSERIALDV_LIBRARY
>   NAMES ${LIBSERIALDV_NAMES}
>   PATHS ${SERIALDV_DIR}/lib
>         /usr/lib
>         /usr/local/lib
> )
> if (LIBSERIALDV_INCLUDE_DIR AND LIBSERIALDV_LIBRARY)
>     set(LIBSERIALDV_FOUND TRUE)
> endif (LIBSERIALDV_INCLUDE_DIR AND LIBSERIALDV_LIBRARY)
> if (LIBSERIALDV_FOUND)
>     if (NOT LibSerialDV_FIND_QUIETLY)
>         message (STATUS "Found libserialdv: ${LIBSERIALDV_INCLUDE_DIR},
> ${LIBSERIALDV_LIBRARY}")
>     endif (NOT LibSerialDV_FIND_QUIETLY)
> else (LIBSERIALDV_FOUND)
>     if (LibSerialDV_FIND_REQUIRED)
>         message (FATAL_ERROR "Could not find SerialDV")
>     endif (LibSerialDV_FIND_REQUIRED)
> endif (LIBSERIALDV_FOUND)
>

mark_as_advanced(LIBSERIALDV_INCLUDE_DIR LIBSERIALDV_LIBRARY)

i suspect, as was the case for every dependency where i had a system-wide
version installed, that HINTS would be needed to cause cmake to pick the
'right' one.

f4exb, 2019-07-26:
I think HINTS is needed to override the default paths and PATHS cannot do that, PATHS should be there as a last resort and have only fixed values i.e. not computed elsewhere by cmake. At least this is what I understood from the link I posted above.

I have pushed on dev fixes to abide by this rule and that should fix most of your issues. I am still not clear with what happens with SoapySDR.

csylvain, 2019-07-26:
what i experienced is 'finding' fails if it doesn't do the Version.h check
successfully on its first attempt. i threw SOAPYSDR_DIR into the check so
it would be guaranteed to succeed on my system and then proceed to do the
rest in the FindSoapySDR cmake instructions.

yes, i agree the order of entries in PATHS does not appear to have much (if
any) effect on the search order.  HINTS appears to be the mechanism to
express a preference or order of precedence.

i will gladly test the 'new and improved' cmake instructions at a future
date.

f4exb, 2019-07-26:
What I suppose (note "I suppose" here...) is that if SOAPYSDR_INCLUDE_DIR is not right then the function _SOAPY_SDR_GET_ABI_VERSION is not successful and ends up in the fatal error trap there. With the HINTS coded right (I hope) you will get the correct SOAPYSDR_INCLUDE_DIR and hence the function can run successfully without modification.

csylvain, 2019-07-26:
sounds good!  will let you know if any more configuration wrangling will be
needed when it's ready for testing.

may i add a note of appreciation for the program now?  :-)

f4exb, 2019-07-31:
I suppose this is OK hence closing the issue

closed 2019-07-31 by f4exb

csylvain, 2019-07-31:
i haven't made any tests. did someone else?
hoping it's been smoothed out ..

csylvain, 2019-07-31:
tested and can report build proceeded smoothly. no wrangling needed. thanks!

$ git log
commit 11d12ab99283adb95eb5bdad4aaa50bc79d94165 (HEAD -> master, tag:
v4.11.5, origin/master, origin/dev, origin/HEAD)
Author: f4exb <f4exb06@gmail.com>
Date:   Wed Jul 31 03:19:56 2019 +0200

    CW keyer from keyboard: updated documentation changelogs and version

f4exb, 2019-08-02:
Thanks for reporting success :smile: