f4exb/sdrangel #388: RX-TX tracking with RIT (Receiver Incremental Tuning)

drom, 2019-07-26:
Related to #382 

In Full-Transceiver mode operation is that TX should track RX with a specific Delta frequency (that can be very large like 144MHz -- 435MHz) but this delta can be tuned by special [RIT](https://www.amateur-radio-wiki.net/index.php?title=Operating_procedures#RIT) (+-100KHz) knob while operating. That makes it hard to be a separate plugin because it requires knobs inside that same UI.

f4exb, 2019-07-26:
There are as many use cases like this as there are users so I will not make changes in the core program itself. It can drive and be driven by external entities through API and this is enough. Now if API is missing something or requires extension this can be done.

If I get your requirement right you would like to change the frequency delta as you go along. In fact it does not really matter how large it is you would just like to be able to change it on the fly. Then it is a matter of making the "monitor" program a bit more complex to accept variable frequency delta entered by the user (so having its own UI) and not just being launched with a few options and no further interaction.

If you want to implement something like this inside the core program and want to stay generic it means letting the user draw its own custom panel and interface (to the "monitor" program). An interesting idea but a large undertaking so I am not sure I am willing to spend the required (large) time on this.

drom, 2019-07-26:
@f4exb I totally understand your point about the big number of use cases, with contradicting goals.
I will try implementing the plugin with custom UI using your API.

closed 2019-07-26 by drom