f4exb/sdrangel #389: smack talk in readme.md

rgetz, 2019-07-26:
Just wondering:

From:
https://github.com/f4exb/sdrangel/blob/master/plugins/samplesink/plutosdroutput/readme.md

```
As you can see from the Wiki this is becoming a fairly popular SDR hardware platform. It does have interesting features but the library documentation and examples are poor when not misleading. Therefore while this implementation does work it should still be considered experimental.
```

Can you be more specific? (it's hard to make better examples, or make documentation better without concrete examples). 

Which documentation / examples are misleading?

-Robin

f4exb, 2019-07-26:
I'll just remove this sentence if you consider it offending. Libiio is rather (if not very) generic so I suppose I had trouble interpreting how it applied to Pluto apparently more on the Tx side of things.

f4exb, 2019-07-26:
Same thing on Rx.

rgetz, 2019-07-26:
Yes - libiio is a generic library for any IIO based device, not just Pluto.

If we can make things better - we can try. I don't mind adding some pluto specific examples to either libiio or libad9361-iio if you are interested in describing what you are looking for.

f4exb, 2019-07-26:
Well Pluto support was introduced nearly 2 years ago so it is difficult for me to remember exactly what had lead to frustration. Generally speaking libiio is quite different from the other support libraries that dedicate themselves to one particular hardware which does not mean it is any better or worse but probably surprising to the application developer so maybe this is something to bear in mind and it is natural to forget it when you have your hands into it all day and it becomes a routine. 

I will suppress the nasty comments that do not bring added value to the readmes' and are outdated in some sense. It is not worth arguing about it at all.

rgetz, 2019-07-26:
Thanks - we are here to help - if you are getting frustrated by things - just ask. I know it's not always clear - and as you said - from our perspective - when you work with things every day - everyone believes it's "obvious", and doesn't document things for those starting out... - it's hard on both sides.

IIO has been in the linux kernel since 2007 (in staging), and libiio has existed since 2014. Pluto was introduced 2017 - so it was actually pretty late to the party...

For some in-kernel history:
https://events.linuxfoundation.org/wp-content/uploads/2017/12/10-Years-of-the-Industrial-IO-Kernel-Subsystem-Jonathan-Cameron-Huawei.pdf

Nasty comments are OK (I don't mind) - just be more specific so we can improve things, or fix it. Saying "make more examples" doesn't really translate into an action item for anyone. It's easier for us to act on something when it is "I need an example which shows me how to do xxxxx task, it all seems missing from your documentation". Or "your example yyyy is sucky, since xxxxx". 

I agree docs are thin/missing, I'm not sure I agree that things are misleading. (until proven otherwise).

f4exb, 2019-07-27:
Fixed in v4,11.3

closed 2019-07-27 by f4exb